riscvarchive/riscv-v-spec #304: vslideup and vslidedown are not symmetric

solomatnikov, 2019-10-15:
vslidedown shifts in 0s

vslideup leaves elements < OFFSET unchanged

kasanovic, 2019-11-10:
I can see that making vslidedown leave values unchanged might be more useful in more cases.

kasanovic, 2019-11-11:
The vl can always be adjusted to give the symmetric behavior.  Returning zero of out-of-range indices is consistent with vrgather.

solomatnikov, 2019-11-11:
then, why not to do the same for vslideup?

aswaterman, 2019-11-11:
Zeroing the vacated elements for vslideup would add another special case.
Currently, vstart and OFFSET are handled the same as each other, but that
would no longer be the case if the vacated elements were zeroed.

I'm not bothered by the asymmetry, but if we had to make a change, I would
change vslidedown rather than changing vslideup.

solomatnikov, 2019-11-11:
zeroing for vslidedown is also a special case and is bad for performance because without renaming vslidedown takes the same number of cycles as if vl == VLMAX

kasanovic, 2019-11-15:
I'm OK with changing vslidedown to not zero past vl-offset.

kasanovic, 2019-11-15:
Consensus was to leave these instructions as is for now, and to work on code examples to see how they're used.

kasanovic, 2020-06-28:
I believe there is no motivation to change this.

closed 2020-06-28 by kasanovic