riscvarchive/riscv-v-spec #305: Load/Store Segment Instructions and strided/indexed operation

billhuffman, 2019-10-17:
I think the spec could use additional clarity on the memory locations accessed for load/store segment instructions - especially for strided and indexed accesses.

Lets say NFIELDS==3 and SEW=4 and we do a unit-stride load of element size.  I think that the first vector register contains words from addresses 0, 12, 24, 36, ... while the second contains words from addresses 4, 16, 28, 40, ... and the third contains words from addresses 8, 20, 32, 44, ....

If we do the same thing with a strided load, what happens if the stride is, say 20 bytes.  Do the three registers get words at 0, 20, 40, 60,... and 4, 24, 44, 64, .... and 8, 28, 48, 68,...?  I'm guessing so.  Or is it maybe 0, 60, 120, 180,... and 20, 80, 140, 200,... and 40, 100, 160, 220, ...   And if the stride is 4 bytes, is there overlap?

I have the same question for indexed accesses.  It seems like the i'th element of each resulting register must have come from contiguous places in memory for all of these cases. Is that right?  That would correspond to the first choice for the strided load above.

kasanovic, 2020-01-22:
yes, your first guess is correct.  I'm not sure how you arrived at other possibility.  Can you point out ambiguity in spec.

billhuffman, 2020-01-22:
Looking back and trying to see how I was thinking about it three months ago, I think the clarity issue revolves around the meaning of a "segment."  It seems to me that there is more description of what a "segment" is now than there was three months ago.  In any case, I think the description is fine at this point.

closed 2020-01-22 by billhuffman