riscvarchive/riscv-v-spec #306: Indexed Load/Store with index width different from data width

billhuffman, 2019-10-17:
There is a note in the spec under indexed loads and stores asking whether we need to provide for indexes wider than the data.  An 8-bit index limit for 8-bit data seems like a pretty serious limitation.  Even a 16-bit index will often be a limitation.  And yet the solutions I've thought about seem ugly as well.  Is there any progress on this?

aswaterman, 2019-10-17:
I'd just eat the cost and use 32-bit indexes when necessary, wasting the regfile capacity; then use narrowing operations to convert back to 8-bit if desired.   Given the much greater cost of the gather operation vs. the narrowing operation, I think it's not as bad as it sounds...

billhuffman, 2019-10-17:
That's one of the ugly solutions I've thought about.  And it does assume that gathers are slow.  We gather up to 32 elements per cycle today.

aswaterman, 2019-10-17:
Yeah, it's only a satisfactory solution for systems for which the scatter/gather occupancy is quite a bit greater than the occupancy of the subsequent narrowing instructions.

Really, we'd like to select the element width independently, but are hemmed in by limited encoding space.  Here are some alternatives that don't mess with the encoding scheme too much; they're all bad in their own ways:

- Employ 48-/64-bit instruction encodings so there's enough space to encode one or two element widths statically.  I imagine this will be the preferred answer in years to come, but is not digestible today.
- Only provide the `vlxe` and `vsxe` instructions, getting rid of `vlxX[u]` and `vsxX.  The resulting freed-up opcode space can be used to encode an index width different from the element width.  The downside is that an additional instruction is necessary when loading a narrow element into a wider register.  If the pessimized case is relatively uncommon, this option might be worth exploring.
- Define that indices are XLEN bits wide, unconditionally.  This means indices could occupy beaucoup registers.
- Add a `vxsew` field to the `vtype` CSR to control the scatter/gather index element size independently of SEW.

billhuffman, 2019-10-17:
Yes.  And one more bad solution would be to take over the nf bits of indexed accesses for an index size.  I suspect those bits may be quite useful.

There are actually two remaining mop encodings that could be used for some other form - maybe to indicate an index width in place of the static width encoding or the nf field.  Not sure it's wise to use those either.

I'm hoping someone has a good way, but I'm not really seeing one....

      Bill

On 10/16/19 10:21 PM, Andrew Waterman wrote:

Yeah, it's only a satisfactory solution for systems for which the scatter/gather occupancy is quite a bit greater than the occupancy of the subsequent narrowing instructions.

Really, we'd like to select the element width independently, but are hemmed in by limited encoding space. Here are some alternatives that don't mess with the encoding scheme too much; they're all bad in their own ways:

  *   Employ 48-/64-bit instruction encodings so there's enough space to encode one or two element widths statically. I imagine this will be the preferred answer in years to come, but is not digestible today.
  *   Only provide the vlxe and vsxe instructions, getting rid of vlxX[u] and `vsxX. The resulting freed-up opcode space can be used to encode an index width different from the element width. The downside is that an additional instruction is necessary when loading a narrow element into a wider register. If the pessimized case is relatively uncommon, this option might be worth exploring.
  *   Define that indices are XLEN bits wide, unconditionally. This means indices could occupy beaucoup registers.
  *   Add a vxsew field to the vtype CSR to control the scatter/gather index element size independently of SEW.

kasanovic, 2019-11-10:
Given lack of good solutions in 32-bit encoding space, the intent is to provide these in 64/48-bit encoding space.

closed 2019-11-10 by kasanovic