riscvarchive/riscv-v-spec #307: about "remaining elements  of vd are unchanged " for vcompress

wwang0, 2019-10-21:
Now for the tail elements , they are unchanged, how about the elements between the last compressed elements and vl for vcompress, the current doc shows they are set zero:
----
    Example use of vcompress instruction

        1 1 0 1 0 0 1 0 1   v0
        8 7 6 5 4 3 2 1 0   v1
                                vcompress.vm v2, v1, v0
        0 0 0 0 8 7 5 2 0   v2
----

wwang0, 2019-10-24:
@aswaterman , any comments for this one? Thanks!

aswaterman, 2019-10-29:
The current definition matches vslidedown, which also zeros elements between the end of the source vector and the destination vl.  I think we should only change vcompress if we also change vslidedown.  My preference is to leave the current spec as-is (zero the elements) because stopping early is less consistent with other vector instructions.

kasanovic, 2019-11-10:
Issue #304 points out that vslidedown is not symmetric with slideup, and maybe should leave tail elements unchanged.
I think the zeroing behavior here is left over from tail zeroing, and example vmcompress does not match text.
Leaving tail elements unchanged would match new model, and is probably more generally useful.

wwang0, 2019-11-11:
I prefer leaving them unchanged. Thanks.

kasanovic, 2019-11-11:
The spec actually says that remaining elements are unchanged, but the example is confusing as it shows zero elements in the unwritten tail but does not show the initial value of vd.    I think leaving unchanged makes sense, even if it slightly complicates renamed implementations.

kasanovic, 2019-12-07:
Left as unchanged, and spec updated.

closed 2019-12-07 by kasanovic