riscvarchive/riscv-v-spec #308: Clarification needed for mstatus.FS update (to Dirty) on updating to vxset/vxrm CSRs

chuanhua, 2019-10-22:
According to RISC-V privileged SPEC, the FS field encodes the status of the floating-point unit (including the CSR fcsr and floating-point data registers f0–f31).

If RVV is not supported, the definition is very clear. 

- The MSTATUS.FS is updated to “DIRTY” state if a scalar floating-point instruction update FCSR (including FFLAGS) and F0~F31.

- The MSTATUS.FS is updated to “DIRTY” state if a CSR instruction update FCSR (including FFLAGS and FRM).

But, the HW behavior is not clearly explained/defined in SPEC when RVV is supported.

When RVV is supported, CSR vxsat/vxrm are added. And, CSR fcsr holds mirrors of the vxsat and vxrm.

What is the expected HW behavior for case (3) and (4) in the following table? 

No. | Case | Update FS to “Dirty” State ?
-- | -- | --
1 | Scalar floating-point instruction | Yes (if it updates FCSR/FFLAGS or F0~F31)
2 | CSR instruction (write to FCSR/FFLAGS/FRM) | Yes
**3** | **CSR instruction (write to VXSAT/VXRM)** | **update FS for fixed-point CSR ?**
**4** | **Vector fixed-point instruction** | **update FS for fixed-point operation ?**
5 | Vector floating-point instruction | Yes (if it updates FCSR/FFLAGS or F0~F31)

Thanks!

aswaterman, 2019-10-22:
I wrote a proposal here: https://github.com/riscv/riscv-v-spec/pull/135/files

The upshot is, yes, vxsat/vxrm behave the same as fflags/frm/fcsr with respect to mstatus.FS.

aswaterman, 2019-10-22:
Not sure, I think the issue is that it hasn't been discussed in the task group meeting yet.  Hopefully that can happen next time.

kasanovic, 2019-11-24:
Clarified in commit a6f94e7

closed 2019-11-24 by kasanovic