riscvarchive/riscv-v-spec #312: Does vfmv.v.f and vfmerge.vfm checks for NaN-boxing when FLEN > SEW

EdwardC, 2019-10-29:
In riscv-v spec 11.1: 
3. For floating-point operations, the scalar can be taken from a scalar f register. If FLEN>SEW, the value in the f registers is checked for a valid NaN-boxed value

And there is a exception for vfmv.s.f, vfmv.f.s only copies LSBs and upper FLEN-SEW bits are ignored.

according to riscv ISA spec 12.2:
Apart from transfer operations described in the previous paragraph, all other floating-point opera-tions on narrower n-bit operations, n < FLEN, check if the input operands are correctly NaN-boxed.

Does vfmv.v.f and vfmege.vfm classified as transfer operations and not to check for NaN-boxing?

HanKuanChen, 2019-11-04:
I think there is a similar question #193 before.

kasanovic, 2019-11-10:
Yes, convention is that mv instructions in RISC-V ISA are transfer operations that do not interpret the bits.

closed 2019-11-10 by kasanovic

chuanhua, 2019-11-27:
Can the spec state this explicitly in the vfmv.v.f and vfmege.vfm instruction section? or in the vector arithmetic instructions format section that talks about NaN boxing for scalar f register? I have to search for this issue (issue #312) to know this.