riscvarchive/riscv-v-spec #314: Could we parameterize the nf field for load/store segment instructions

JerryShih, 2019-11-03:
@kasanovic @aswaterman 

Here is the closed issue for the same question: https://github.com/riscv/riscv-v-spec/issues/114

The reason is that:

> Decided to stay with current PoR, since immediate affects how many destination vector registers are written and different variants might be implemented differently in hardware.

But I still can't know why the nf field is affected by different hardware. All nf configurations should be supported, right?

The rvv use vsetvli to handle the tail elements. Could we have the similar idea for segment load/store?

Here is an example. I try to transpose a matrix.
First, group 8 rows together and process them in the same time
Then, group 4 rows together for the remaining rows and process them.
Finally, process 1 row sequentially for the last rows.

If we could parameterize the nf, we could simplify the code.
**vssseg\<nf\>{b,h,w,e}.v vs3, (rs1), rs2, vm**
=>
**vssseg{b,h,w,e}.v nf, vs3, (rs1), rs2, vm**

```
  # process 8 rows
process_8_row:
  srli iRow, iRowNum, 3
  beqz iRow, process_4_row
  li iTemp, 8
  rem iRowNum, iRowNum, iTemp
  ...
  vssseg8w.v vBuffer0, (pOut), iStride
  ...
  addi iRow, iRow, -1
  bltz iRow, loop_8_row
end_loop_8_row:

  # process 4 rows
process_4_row:
  srli iRow, iRowNum, 2
  beqz iRow, process_1_row
  li iTemp, 2
  rem iRowNum, iRowNum, iTemp
  ...
  vssseg4w.v vBuffer0, (pOut), iStride
  ...
  addi iRow, iRow, -1
  bltz iRow, loop_4_row
end_loop_4_row:

  # process 1 rows
process_1_row:
  beqz iRowNum, exit
  ...
  vssseg1w.v vBuffer0, (pOut), iStride # or use vssw.v
  ...
  addi iRowNum, iRowNum, -1
  bltz iRowNum, loop_1_row
end_loop_1_row:
```

=>
```
  # process 8 rows
process_8_row:
  srli iRow, iRowNum, 3
  beqz iRow, process_4_row
  li iTemp, 8
  rem iRowNum, iRowNum, iTemp
  ...
  vssseg8w.v vBuffer0, (pOut), iStride
  ...
  addi iRow, iRow, -1
  bltz iRow, loop_8_row
end_loop_8_row:

  # process the tail rows
process_tail_row:
  beqz iRowNum, exit
  ...
  # process the remaining rows. It might be a number from 7 to 1.
  vsssegw.v iRowNum, vBuffer0, (pOut), iStride <==========
  ...
end_loop_tail_row:
```
We could simplify the "process 4 rows" and "process 1 rows" block in the original code using the new segment instruction.

aswaterman, 2019-11-03:
Making the segment length a dynamic value will add another complication for register-renamed designs: the number of registers written would depend on a dynamic register value, and so you'd have to do physical register allocation after instruction issue for these instructions.

This is worse than the existing renaming complexity that LMUL implies, because in that case LMUL is provided as an immediate argument to vsetvli, rather than in a register.  (I expect the R-type vsetvl instruction will require a pipeline flush in these machines, whereas the I-type vsetvli will be fast.)

JerryShih, 2019-11-04:
@aswaterman @kasanovic 
I agree that will have more works to do in the issue stage, but we need to have the **SIMD-like** code to handle the tail elements if we don't parameterize the nf field. The rvv is trying to get rid of this problem, right?

aswaterman, 2019-11-04:
It’s true that the VL scheme only eliminates fringe code in one dimension,
but this proposal only helps the 2D case for some simple examples like
transpose. Usually when segmented accesses are used, you are performing
some computation on the specific registers that were loaded, so you would
still need multiple code blocks and/or branches around those instructions.

A more holistic approach to 2D vectorization is desirable, but is still an
area of active research so should be punted to future work.

HanKuanChen, 2019-11-04:
You can try to implement a jump table to solve this problem, even though this is cumbersome.
```
li eight, 8
process_row:
  ...
  la x6, store_seg
  rem iRow, iRowNum, eight
  slli iRow, iRow, 2
  add x6, x6, iRow
  jalr x6, 0(x6)
  ...
  sub iRowNum, iRowNum, iRow
  ...
end_row:
  ...

store_seg:
  vssseg8w.v vBuffer0, (pOut), iStride
  jalr x0, 0(x6)
  vssw.v vBuffer0, (pOut), iStride
  jalr x0, 0(x6)
  vssseg2w.v vBuffer0, (pOut), iStride
  jalr x0, 0(x6)
  vssseg3w.v vBuffer0, (pOut), iStride
  jalr x0, 0(x6)
  vssseg4w.v vBuffer0, (pOut), iStride
  jalr x0, 0(x6)
  vssseg5w.v vBuffer0, (pOut), iStride
  jalr x0, 0(x6)
  vssseg6w.v vBuffer0, (pOut), iStride
  jalr x0, 0(x6)
  vssseg7w.v vBuffer0, (pOut), iStride
  jalr x0, 0(x6)
```

kasanovic, 2019-11-04:
This does not actually add an additional complication to vector register renaming, as most implementations will have to wait for dynamic vl value before renaming individual vector registers in a group with LMUL>1.  However, without more ISA support, as @aswaterman points out, it would be too clumsy to use in general.

If you scan back through email archive you'll find a sketch of my plans for handling 2D shapes using an additional two vector length registers to handle 2D stripmining, though this was for an earlier version of the architecture spec with heavier weight configuration state.

This is a significant increment in complexity, so group decided to leave as part of future extension while we finished the 1D variant.  I think concept can be adapted to new base vector spec but I do not recommend doing this until after base is done.

solomatnikov, 2019-11-04:
Making the segment length a dynamic value will complicate not just renaming but also dependency checks: RAW, etc. These checks are already quite complicated even in simple microarchitecture.

kasanovic, 2019-11-10:
General 2D stripmining will be tackled in future extension.

closed 2019-11-10 by kasanovic