riscvarchive/riscv-v-spec #318: Add the float version of slidedown/up instructions.

JerryShih, 2019-11-09:
Ref: https://github.com/riscv/riscv-v-spec/issues/236#issuecomment-507997226

I had used the slidedown/up instructions in a loop, the are some extra costs for float-to-int conversion.
If there is no extra work to have the float version of slidedown/up, please make both float and int version of slidedown/up.

@aswaterman @kasanovic

JerryShih, 2019-11-11:
Here is my use case.
I try to compute a serial of fp32 values, and push them into a vector register.
If the loop number becomes bigger, the cost of "fmv.x.s" will be a problem.

```
loop:
    .... // a complex formula for fp32_vlaue
    fmv.x.s int_value, fp32_value
    vslide1down.vx v0, v0, int_value
    ....
end_loop:
```

kasanovic, 2019-11-11:
Irrespective of whether we add the float version, the code example given would generally run much faster and lower energy if you were to do scalar stores into a memory buffer, then do a single vector load to bring into vector register file.  Every vslide1down has to read and write the entire vector register VLMAX times, i.e., O(VLMAX^2) operands moved, whereas using a memory buffer only writes and reads each element once from RAM and once into a vector register, i.e., O(VLMAX) operands moved.

kasanovic, 2019-12-07:
Asking TG opinion.

David-Horner, 2020-02-07:
The most significant use case is when XLEN < FLEN.  (RV32 and D float).
The transfer to integer to vector register is not possible.

However, I would prefer scalar vslide1down.vs vslide1up.vs to vslide1down.vf
The float to scalar can be fused with the vslide1.

JerryShih, 2020-02-26:
> Asking TG opinion.

@kasanovic 
Any update for float version of slidedown/up?

closed 2020-04-03 by kasanovic

aswaterman, 2020-04-04:
@JerryShih added.

PR #402 (2020-04-03): Add vfslide1up/down instructions
Resolves #318
commit d0f1fabb (2020-04-03): Add vfslide1up/down instructions

Resolves #318