riscvarchive/riscv-v-spec #327: vmsleu.vi, vmsltu.vi and vmsgeu.vi immediate ranges

poinu, 2019-11-14:
In section 12.7 we find the following note:

> The immediate forms of `vmslt{u}.vi` are not provided as the
immediate value can be decreased by 1 and the `vmsle{u}.vi` variants
used instead.  The `vmsle.vi` range is -16 to 15, resulting in an
effective `vmslt.vi` range of -15 to 16.  The `vmsleu.vi` range is 0 to
15 (and `(~0)-15` to `~0`), giving an effective `vmsltu.vi` range of 1 to 16
(Note, `vmsltu.vi` with immediate 0 is not useful as it is always
false). Similarly, `vmsge{u}.vi` is not provided and the comparison is
implemented using `vmsgt{u}.vi` with the immediate decremented by one.
The resulting effective `vmsge.vi` range is -15 to 16, and the
resulting effective `vmsgeu.vi` range is 1 to 16 (Note, `vmsgeu.vi` with
immediate 0 is not useful as it is always true).

In it, the range for the `vmsleu.vi` instruction is defined as `[0, 15]` ⋃ `[(~0)-15, ~0]`. Given that the immediate field is 5 bits long (thus `~0 == 31`), I wonder if there is any reason why we can't express this range as `[0, 31]`.

When it comes to the range of the `vmsltu.vi` and `vmsgeu.vi` pseudoinstructions, I wonder if it should be `[1, 32]` instead of `[1, 16]`. Finally, the ranges for `vmsgt{u}` are not explicitly stated (even though they match those of `vmsle{u}.vi`).

I would suggest a rewording on the lines of:

```diff
 NOTE: The immediate forms of `vmslt{u}.vi` are not provided as the
 immediate value can be decreased by 1 and the `vmsle{u}.vi` variants
 used instead.  The `vmsle.vi` range is -16 to 15, resulting in an
 effective `vmslt.vi` range of -15 to 16.  The `vmsleu.vi` range is 0 to
-15 (and `(~0)-15` to `~0`), giving an effective `vmsltu.vi` range of 1 to 16
-(Note, `vmsltu.vi` with immediate 0 is not useful as it is always
-false). Similarly, `vmsge{u}.vi` is not provided and the comparison is
+31, giving an effective `vmsltu.vi` range of 1 to 32 (Note,
+`vmsltu.vi` with immediate 0 is not useful as it is always false).
+Similarly, `vmsge{u}.vi` are not provided and the comparison is
 implemented using `vmsgt{u}.vi` with the immediate decremented by one.
-The resulting effective `vmsge.vi` range is -15 to 16, and the
-resulting effective `vmsgeu.vi` range is 1 to 16 (Note, `vmsgeu.vi` with
+The `vmsgt.vi` range is -16 to 15, resulting in an effective
+`vmsge.vi` range of -15 to 16, while the `vmsgtu.vi` range is 0 to 31,
+giving an effective `vmsgeu.vi` range of 1 to 32 (Note, `vmsgeu.vi` with
 immediate 0 is not useful as it is always true).
```

Thanks!

jim-wilson, 2019-11-14:
All immediates are sign extended.  So the range for unsigned compares is (unsigned)-16 to (unsigned)15.  Special casing unsigned compare immediate fields would complicate the instruction decode and/or execution, which is undesirable from a hardware point of view.  Unless maybe someone can show proof that shifting the range is worth the extra hardware cost, but that would require a software study which is hard to do when you don't have any software yet.

But note that shift instructions, and shift-like instructions like vslide and I think some fixed-point operations, truncate the immediate at the shifter and hence are effectively unsigned, but the hardware can still treat them as signed immediate fields that get truncated at the shifter, and hence there is no extra hardware cost.

poinu, 2019-11-15:
Thanks for the explanation Jim. I understand your point, but the changes made in a16ef4c are a bit confusing to me:

In section 11.1, in the Table 12 in section 11.1 shows  `imm[4:0]` (as opposed to `simm[4:0]`), while the following text in the same section suggests the immediate could be zero-extended to SEW bits:

> Vector-scalar operations can have three possible forms, but in all
> cases take one vector of operands from a vector register group
> specified by `vs2` and a second scalar source operand from one of
> three alternative sources.
> 
> 1) For integer operations, the scalar can be a 5-bit immediate encoded
> in the `rs1` field.  The value is sign- **or zero-extended** to SEW bits.

Maybe we need to reword that part instead.

So, if I understood correctly, and the ranges are properly summarized in the following table,

| Instr. | Range |
|-|-|
| `vmsle.vi` | `[-16, 15]` |
| `vmsleu.vi` | `[0, 15]` ⋃ `[2^SEW-16, 2^SEW-1]` |
| (`vmslt.vi`) | `[-15, 16]` |
| (`vmsltu.vi`) | `[1, 16]` ⋃ `[2^SEW-15, 2^SEW]` |
| `vmsgt.vi` | `[-16, 15]` |
| `vmsgtu.vi` | `[0, 15]` ⋃ `[2^SEW-16, 2^SEW-1]` |
| (`vmsge.vi`) | `[-15, 16]` |
| (`vmsgeu.vi`) | `[1, 16]` ⋃ `[2^SEW-15, 2^SEW]` |

we could extend the paragraph so they are a bit more explicit, for example:

```diff
 NOTE: The immediate forms of `vmslt{u}.vi` are not provided as the
 immediate value can be decreased by 1 and the `vmsle{u}.vi` variants
 used instead.  The `vmsle.vi` range is -16 to 15, resulting in an
 effective `vmslt.vi` range of -15 to 16.  The `vmsleu.vi` range is 0 to
 15 (and `(~0)-15` to `~0`), giving an effective `vmsltu.vi` range of 1 to 16
+(and `(~0)-14` to `(~0)+1`)
 (Note, `vmsltu.vi` with immediate 0 is not useful as it is always
-false). Similarly, `vmsge{u}.vi` is not provided and the comparison is
+false). Similarly, `vmsge{u}.vi` are not provided and the comparison is
 implemented using `vmsgt{u}.vi` with the immediate decremented by one.
-The resulting effective `vmsge.vi` range is -15 to 16, and the
-resulting effective `vmsgeu.vi` range is 1 to 16 (Note, `vmsgeu.vi` with
+The `vmsgt.vi` range is -16 to 15, resulting in an effective
+`vmsge.vi` range of -15 to 16, while the `vmsgtu.vi` range is 0 to 16
+(and `(~0)-15` to `~0`), giving an effective `vmsgeu.vi` range of 1 to
+16 (and `(~0)-14` to `(~0)+1`) (Note, `vmsgeu.vi` with immediate 0 is
+not useful as it is always true).
```

Or perhaps it would be even clearer if we rewrote `~0` as `2^SEW - 1`:

```diff
 NOTE: The immediate forms of `vmslt{u}.vi` are not provided as the
 immediate value can be decreased by 1 and the `vmsle{u}.vi` variants
 used instead.  The `vmsle.vi` range is -16 to 15, resulting in an
 effective `vmslt.vi` range of -15 to 16.  The `vmsleu.vi` range is 0 to
-15 (and `(~0)-15` to `~0`), giving an effective `vmsltu.vi` range of 1 to 16
+15 (and `2^SEW-16` to `2^SEW-1`), giving an effective `vmsltu.vi` range
+of 1 to 16 (and `2^SEW-15` to `2^SEW`)
 (Note, `vmsltu.vi` with immediate 0 is not useful as it is always
-false). Similarly, `vmsge{u}.vi` is not provided and the comparison is
+false). Similarly, `vmsge{u}.vi` are not provided and the comparison is
 implemented using `vmsgt{u}.vi` with the immediate decremented by one.
-The resulting effective `vmsge.vi` range is -15 to 16, and the
-resulting effective `vmsgeu.vi` range is 1 to 16 (Note, `vmsgeu.vi` with
+The `vmsgt.vi` range is -16 to 15, resulting in an effective
+`vmsge.vi` range of -15 to 16, while the `vmsgtu.vi` range is 0 to 16
+(and `2^SEW-16` to `2^SEW-1`), giving an effective `vmsgeu.vi` range
+of 1 to 16 (and `2^SEW-15` to `2^SEW`) (Note, `vmsgeu.vi` with
+immediate 0 is not useful as it is always true).
```

Does this make any sense?

Thanks!

jim-wilson, 2019-11-15:
On Fri, Nov 15, 2019 at 5:57 AM Ferran Pallarès <notifications@github.com>
wrote:

> Thanks for the explanation Jim. I understand your point, but the changes
> made in a16ef4c
> <https://github.com/riscv/riscv-v-spec/commit/a16ef4c1fca01259dd090847f3d2a68b39a634a4>
> are a bit confusing [...]

As I mentioned, shift and shift-like instructions truncate the immediate
for the shift, and hence are effectively unsigned.  This is why the
paragraph says that immediate can be unsigned, but they can only be
unsigned for shift and shift-like instructions.

Jim

poinu, 2019-11-15:
Yes, I understood that the immediate can only be unsigned for shift and shift-like instructions from your first response. In my message I only intended to give some feedback and discuss whether we should state this in a more obvious way.

Cheers

commit 1814e24a (2020-07-01): Clarified immediate ranges in unsigned comparisons.
Closes #327.
closed 2020-07-01 by kasanovic