riscvarchive/riscv-v-spec #328: instruction synchronization with vxsat

ubc-guy, 2019-11-15:
Right now, checking vxsat requires synchronization (with a decoupled vector unit) to ensure that all prior issued vector instructions have completed. This could lead to slowdown of fixed-point code that checks it often, eg after every fixed-point instruction. The alternative, of waiting until the very end, may be too late, particularly after a long sequence of instructions. As a result, scalar code is forced to check vxsat in intervals (using counters & branches).

Can I recommend two types of reads of vxsat: a synchronizing one (forcing vector instruction completion) and an asychronous one (which just reports its current status, but which may be out of sync with completed vector instructions).

This would allow loops to frequently check vxsat asynchronously, and jump to a different section of code after a problem is detected, without worrying about slowdown effects.

As well, programmers must manually set vxsat=0 (it is not reset by hardware). Thus, a paranoid programmer will set vxsat=0 before each instruction, and check it after each instruction, forcing two synchronizing events. Perhaps we can have two vxsat flags: the current one (reset by programmer), and another one say vxsata (reset automatically either by each fixed-point vector instruction or by each vector instruction) -- this would eliminate the need to write to vxsat and avoid a synchronization of the decoupled vector unit.

kasanovic, 2019-11-15:
We have been trying hard to avoid implementation-specific behavior at unprivileged level.  Allowing code to observe differences in timing inside one thread is also worrying.  Providing a trap on vxsat would be a cleaner alternative than polling, but would complicate complex pipelines and we have not added for IEEE FP exceptions for that reason.

I think in  latter part, you're suggesting a non-sticky vxsat, which is basically adding  a condition code.  The zeroing write to vxsat before an instruction does not cause a loss-of-decoupling event, but does consume instruction space and bandwidth.

ubc-guy, 2019-11-15:
I'm not sure what implementation-specific behaviour you are referring to?

The current spec forces some type of synchronization upon read of vxsat, or at least that is my intepretation (a read of vxsat must reflect the status of all previous vector fixed-point instructions as seen in program-order). The duration of time to achieve this synchronization is already reveals implementation information. (Of course, there are different ways to achieve this synchronization, with the simplest being a full pipeline flush, and more complex ones being insertion of vxsat into the vector instruction stream as a fence for vector instruction reordering). No matter the implementation, the result will have an impact on the performance of the scalar instruction stream.

I'm suggesting adding another "viewpoint" of the same flag, where no synchronization is enforced, no instruction fence is performed, just a raw "live" view of the same flag. This is indeed polling, and it could be done with very low overhead. The "synchronizing" view would still be available as well.

Yes, the non-sticky vxsat is a condition code. If this is defined as a second flag, eg vxsata, then it is computed concurrently with vxsat, then there is no additional instruction space required. Instruction bandwidth is only needed to write 0 to vxsat (already needed), and this would actually alleviate the need to do that. Reading vxsata would not require any additional instruction bandwidth than reading vxsat.

kasanovic, 2019-11-16:
An asynchronous polling read of vxsat would cause implementation-specific behavior.  The value return is either 0 or 1 depending on timing, which is a functional divergence not just a timing difference.

kasanovic, 2020-06-28:
I am closing this.  I do not believe we want implementation-dependent timing exposed to give architecturally different results.  Please reopen to discuss in task group.

closed 2020-06-28 by kasanovic