riscvarchive/riscv-v-spec #329: Documentation of SLEN

ubc-guy, 2019-11-15:
I would like to suggest rewriting section 4 so the effects of SLEN are more apparent to programmers and architects. I'm only recommending a change of presentation for clarity, not a change in the spec.

Currently, section 4 is organized by different values of LMUL. The value of LMUL is dynamically changed by a program, so programmers must jump from section to section to understand data layout.

In contrast, SLEN is a static architectural parameter: once set, it is fixed for the implementation. By rearranging the subsections in section 4, based upon different examples of SLEN, it will be more clear to both programmers and architects: jump to the relevant section for a specific SLEN, and see how data should be laid out. Within each subsection, the different values of LMUL (and SEW) can be enumerated as examples.

kasanovic, 2019-11-15:
Can you make a PR?

ubc-guy, 2019-11-15:
I can work on that but it will take time. I would like to get an ok that this is a good reorg to do before undertaking the effort.

kasanovic, 2019-11-15:
It sounds OK to me, though I think having one place where different SLEN are shown next to each other is important to see the effect of SLEN.

ubc-guy, 2019-11-15:
Agreed. I can leave in an extra subsection to show impact of selecting different SLEN values (although it would be redundant from perspective of a spec, it would have merit as a Note or other annotation).

aswaterman, 2019-12-07:
When we improve the SLEN text, we might also document this code sequence that computes SLEN: https://github.com/riscv/riscv-v-spec/issues/233

kasanovic, 2020-04-30:
Text was rewritten with change in mapping.

closed 2020-04-30 by kasanovic

David-Horner, 2020-05-11:
Not only was mapping changed, the meaning of SLEN changed drastically.
There has been a level of confusion as a result. Perhaps we can refer to the new SLEN as hSLEN or SLENH for horizontal striping vs the original striping to support vertical register stacking?