riscvarchive/riscv-v-spec #331: AVL encodings (-1 and "don't change")

ubc-guy, 2019-11-15:
(1) Section 6.1 says AVL is unsigned, while a NOTE also says AVL of -1 is used to set vl to VLMAX. Should the -1 in the NOTE be changed to 2^XLEN-1 ?

(2) Using x0 as rs1 for AVL means "don't change VL". Will this cause a problem with some compilers? After constant propagation, it may determine that vl=0 is a valid value and attempt to use x0, but this will not have the normally intended effect. Hence, the compiler will have to have special-case code to detect this and move 0 into another register (something which isn't typically done).

Instead, maybe the vtype fields in rs2 could be expanded to include a flag (bit 7?) that has the effect of ignoring rs1 if set when executing vsetvl. For vlsetvi, an extra bit may be needed from the ISA encoding, which is less than ideal.

aswaterman, 2019-11-15:
Agreed re: 1 (alternatively `~0`)

Speaking as one of the GCC maintainers, (2) is no problem.  It's already an accepted pattern in RISC-V that rs1=x0 can have special meaning, as is the case for CSRRS/CSRRC and SFENCE.VMA.  GCC and LLVM already have to cope with some x-registers having special purpose, e.g., don't use x5 for tail calls to avoid messing up return-address stacks.

ubc-guy, 2019-11-16:
thanks for the clarification Andrew. I'll accept what you say as truth and don't need a reply ... but I just wanted to clarify my issue with (2) is not about generating a deliberate opcode (using x0) to achieve the intended special purpose.... it's about the accidental generation of what seems like it could be a plausible opcode. x0 is also a bit different than the special instructions and other registers (x0 == 0 has been ingrained in our mindset for a long time). still, i'll assume what you have said holds true and don't expect a response. (compiler-writers beware! here be dragons; pun intended).

aswaterman, 2019-11-16:
Haha.  Yes, it certainly merits a warning and careful auditing.

kasanovic, 2019-11-16:
if compiler knows vl=0, it should elide the code.  Also, don't tell compiler that there is a vl=immediate form of vsetvl in first place

rkruppe, 2019-11-16:
It's not about immediate forms. Before register allocation (using virtual registers in SSA form) you might have something like

```
%virtreg:GPR = <constant>
vsetvl ..., %virtreg, ...
```

which would in general be lowered to materializing the constant into one of x1-x31 and assign `%virtreg` to that register. But when the constant is 0, it's preferred to just assign `%virtreg` to x0 and not need any materialization code.

We can't rely on high level optimizations such as eliding all following vector code for vl=0 to side-step the problem, especially since the constant might only become apparent very late in the compilation pipeline. But it's easy to avoid the problem by having a register class for "GPR but not x0" and using it for this operand of vsetvl (treating the version with x0 as a different instruction internally). As Andrew said, similar tricks are already necessary for other instructions such as CSR operations or some C instructions.

aswaterman, 2019-11-16:
I'm closing this issue because I think there's rough agreement on the latter point.

I tried using 2^XLEN-1 instead of -1, but thought the spec became less clear as a result.

ubc-guy, 2019-11-18:
I think there is a way to get the -1 / VLMAX setting and the "keep VL" setting with slightly rearranged ISA encoding for setvl / setvli:

  rd  rs1    AVL
   0   0     vl     used to maintain current value
  !0   0     ~0     used to get largest VLMAX
   0  !0   x[rs1]   normal use, doesn't keep vl value in GPR
  !0  !0   x[rs1]   normal use

The current spec merges rows 1 and 2 of this table, and only has the functionality of row 1.

The change separates the rd=x0 case from the rd !=x0 case. When VL is kept at the same value, there is probably no need to return it in rd, hence the use of x0 in row 1. If the current value of vl is needed, the CSR can be read. When VLMAX is requested, a VL must be returned, hence the use of any register but x0 in rd.

closed 2019-11-19 by kasanovic

kasanovic, 2019-11-19:
This seems like a simple enhancement so I added to draft spec.

commit b6c48c3b (2019-11-19): Added new form of vsetvli/vsetvl to  allow setting to VLMAX.
Closes #331.