riscvarchive/riscv-v-spec #333: vstart should not be visible from userspace

jnk0le, 2019-11-17:
As of cd0cc6c `vstart` is no longer a "nice but sometimes slow" feature, so I think it is reasonable to leave it exclusive for interrupts/context switches.

> vstart should not be used by application programmers.

Now means that, in worst case, we are about to see "standardization" of some particular implementations.

aswaterman, 2019-11-17:
OTOH, making vstart privileged would interact poorly with user-level
interrupts, should that feature eventually become relevant.

jnk0le, 2019-11-18:
The N extension adds enough CSRs to screw with, that vstart can be aliased similarly only when N extension is active.

kasanovic, 2019-11-24:
Hiding vstart from unprivileged code would make it difficult to provide user-level threading libraries, even without N extension.  (I'll note SPARC had related issue with register window overflow/underflow).

closed 2019-11-24 by kasanovic