riscvarchive/riscv-v-spec #335: Allow VLEN to not be a power-of-2?

kasanovic, 2019-11-24:
Raised on isa-dev by Filip Moc

Hi,

current RVV spec constrains VLEN to be power of 2. I can imagine for some chips
e.g. 512b VLEN could be too much and 256b VLEN could be unnecessarily small
while 384b VLEN could fit just fine.

It seems to me that hardware implementations could benefit a lot from relaxing
VLEN requirements to just be any multiple of SLEN while I don't see any obvious
disadvantage from software point of view.

I don't see any significant conflict with the rest of the spec if VLEN would be
redefined as any multiple of SLEN.

SLEN seems to me that it could be somehow relaxed too.

Maybe I'm missing something? Acording to git this power of 2 restriction is
there from the beginning of new v-spec and I'm not sure whether this was
already discussed.
I tried to find some discussion about this but I didn't find anything.

So why is VLEN constrained to power of 2?

Filip

kasanovic, 2019-11-24:
Bruce Hoult response from mailing list:

 I can't immediately see any fundamental reason for it to be necessary.

When VLMUL is greater than 1, if the VLEN is not a power of 2 then
figuring out where a given element number will be located (which
vector register, and offset within the register) may require a real
divide and remainder instead of a shift and mask. If the vector
registers are stored contiguously in some memory then that operation
may never be necessary (the element is just at eltSize*eltNum + the
start of the first register. But in that case computing the position
of the start of each register will require a real multiply instead of
a shift.

If you're prepared to pay that kind of price (which is not large) then
maybe there is no reason not to allow it.

kasanovic, 2019-12-06:
To me, the major concern is if portable software incurs an extra penalty in considering the non-power-of-2 case, as hardware implementor can always choose what VLEN they use.  So far, I haven't really found an example where non-power-of-2 VLEN would complicate portable software, given that it has to handle variable VLEN, but will put this to group to decide.

vbx-glemieux, 2019-12-06:
If VLEN is a power of 2, it is almost certain the underlying machine is
built to efficiently transfer VLEN-aligned data structures. Software may
choose to use this alignment to ensure high performance.

If VLEN is not a power of 2, it may be more difficult to choose a suitable
alignment.

kasanovic, 2019-12-07:
I'd assume any sensible design would make sure VLEN-size operations were efficiently supported.
For non-power-of-2 designs, a common case would be to have VLEN a multiple of datapath width, e.g., if VLEN=384 but datapath width is 128b.

rkruppe, 2019-12-07:
There is one (somewhat far-fetched but judge for yourselves) thing I could imagine exploiting VLEN being a power of two for: computing addresses of stack slots for vectors. If I'm not mistaken, the current POR is to compute `first_vector_slot + i * vlenb` to get the address of the `i`th VLEN-sized stack slot. If VLEN is a power of two, this can instead be computed as `first_vector_slot + i << log2(vlenb)`.

On many machines the cost of the multiply is probably neglegible compared to a VLEN-sized load/store, but it might be a neat win on some small embedded systems that lack a fast (XLEN-wide) multiplier and/or have very small VLEN if instructions/CSRs are added that remove the need to compute `log2(vlenb)` at runtime. But perhaps those cases are specific enough that they can be addressed by a platform spec and extra extensions instead of the base V extension.

kasanovic, 2020-04-30:
I believe the addition of fractional LMUL means we cannot support non-power-of-2 VLEN.
I'm closing this issue - please reopen if you have a proposal for how non-power-of-2 VLEN would work now.

commit 07d9116e (2020-04-30): Addition of fractional LMUL means we cannot support  non-power-of-two VLEN.
Closes #335.
closed 2020-04-30 by kasanovic