riscvarchive/riscv-v-spec #336: Describe vstart operation for whole register move instructions.

kasanovic, 2019-11-25:


David-Horner, 2020-01-20:
Given that vstart=0 is not mandates for these whole register move instructions:

A fundamental question is what units will vstart represent.
And a further is the sequencing of the vstart values.

vstart units:
Currently whole register move instructions are aligned to memory on ELEN/8.
Historically over an ISAs evolution memory alignment requirements are relaxed.
Specification in bytes would better support vstart in such a an alignment relaxed situation (as the Implementation would presumably write cache line segments at a timem and thus partial ELEN elements).
vstart is already large enough to support the byte granularity as nf does not exceed max LMUL of 8.

There is some potential hardware simplification  if vstart is expressed in ELEN length chunks. The circuitry would match current SEW=ELEM load/store, however, a simple right shift truncation of vstart would create the internal counter used by that hardware (given current ELEN requirement that is is a power of 2 which also might be relaxed in the future).

I therefore propose vstart be defined in byte units.

vstart sequencing:
Although not technically require, in that the register contents are mapped to contiguous bytes in memory, vstart should correspond to that mapping and not any internal considerations such as SLEN. This requirement simplifies verification in that all implementations follow the same criteria and is consistent with the approach that the instructions ignore "configuration" information, specifically that in vl and vtype registers.

Lastly, I see no reason to vary other aspects of vstart, such as an implementation raising an illegal instruction exception on an inconsistent  vstart value. 

I'm ready to make a PR for this when there is agreement on these 2 criteria.

David-Horner, 2020-02-15:
I suggested in the description of the PR 378 that whole register load/store mapping is an unresolved issue, However, I cannot find an issue that matches.
So, I will raise one.

hanna-kruppe, 2020-02-15:
I believe it's desirable to make whole-register moves behave exactly like ordinary `vmv.v.v` under a particular (fixed) value of `vtype` and `vl`. This presumably reduces implementation (and verification) complexity, while also being consistent with the plans for a 64-bit instruction encoding that encodes SEW and LMUL as immediates in each instruction instead of taking them from a CSR. It's also a useful way of thinking about them for programmers and compiler people.

From a byte-centric perspective, this would mean that the bytes in the register group should be numbered [the same way `SEW = 8` elements are numbered under `LMUL = <nf>`](https://github.com/riscv/riscv-v-spec/blob/889e0060a65236661d9e557af6d2cfd7c85c1049/v-spec.adoc#mapping-of-vector-elements-to-vector-register-state). Other choices of SEW (e.g., SEW = ELEN) would be just as meaningful since memory is not involved, but I assume the choice is ultimately arbitrary.

David-Horner, 2020-02-15:
@hanna-kruppe - thank you for your response.

> I believe it's desirable to make whole-register moves behave exactly like ordinary vmv.v.v under a particular (fixed) value of vtype and vl

I agree. But it is also desirable to have instructions that operate on like objects (whole register) behave the same way.  POR for whole register doesn't match. But issue #379 hopes to change that to have fully consistent striped  sequencing across all register group instructions.
So I greatly support your recommendation to unify behaviour.

>  Other choices of SEW (e.g., SEW = ELEN) would be just as meaningful since memory is not involved, but I assume the choice is ultimately arbitrary.

There is a corner case where ELEN > SLEN that is avoided by using small ELEN. Consistency with load/store argues for ELEN  = 08.

hanna-kruppe, 2020-02-15:
Yes, I agree that whole register load/stores should behave as proposed in #379 for the same reasons.

Fair point re: ELEN > SLEN.

commit 9f75ab81 (2020-05-03): Moved comment to normative text to clarify interaction of vstart with whole vector register load/store instructions.
Closes #336.
closed 2020-05-03 by kasanovic