riscvarchive/riscv-v-spec #338: Is the slide amount in "vslideup.vx vd, vs2, rs1, vm" sign-extended to SEW bits?

JamesKenneyImperas, 2019-11-27:
Normally, x register operands to vector integer operations are sign-extended to SEW bits. Is this also the case for vslideup.vx and vslidedown.vx instructions?

If so, this would restrict the usefulness of these instructions when SEW=8: only shift values 0 to 127 would be possible, which is likely to be much smaller than the potential number of elements in a vector register group.

Please can you clarify the intended behavior for these instructions in the specification.

aswaterman, 2019-11-27:
We recently clarified this for `vrgather.vx` (using the more useful definition that does not truncate).  I'll make the corresponding clarification for `vslidedown.vx` and `vslideup.vx`.

closed 2019-11-27 by aswaterman

chuanhua, 2019-11-28:
In the vrgather section, there is a note saying that

> When SEW=8, only vector elements 0-255 can be referenced. 

Since you have decided to extend the index range for `vrgather.vx`, to reduce confusion, could you further clarify that the note description is only for vrgather.vv instruction? Otherwise, a reader may think that the note applies to `vrgather.vx` instruction as well.

Thanks.

aswaterman, 2019-11-28:
I've updated the note to refer to `vrgather.vv` specifically.

commit dc4358c5 (2019-11-27): vslidedown.vx and vslideup.vx do not truncate x[rs1] to SEW bits

Closes #338