riscvarchive/riscv-v-spec #339: Explicitly state NaN-box behavior for vfmerge.vfm and vfmv.v.f

JamesKenneyImperas, 2019-11-27:
As chuanhua commented on issue #312, please can you explicitly state in the specification that vfmerge.vfm and vfmv.v.f do not require NaN-boxed inputs. This is not mentioned anywhere and contravenes the statement in section 11.1, item 3 of the vector-scalar operand form list.

(I have promoted his comment to an issue so that it is not lost.)

aswaterman, 2019-11-27:
After some discussion with @kasanovic, I am not sure that the resolution of #312 is the right thing to do.  My current thinking is that `vfmerge.vfm`, `vfmv.v.f`, and `vfmv.s.f` should check for a NaN box when FLEN > SEW (and correspondingly `vfmv.f.s` should check for a NaN box when SEW > FLEN).  This is a semantic change to the move instructions, so the task group will have to weigh in.

Part of my rationale is that it's simpler for the hardware if all f-register operands to vector instructions are checked for a NaN box.  (Otherwise, the f-register check becomes opcode-dependent in addition to SEW-dependent, as e.g. `vfadd` and `vfmerge` would perform different checks.)  And it does provide a smidge more error checking, but I think this is a minor point.

In the presumably rare circumstance that checking for a NaN box is undesirable, one can first transfer the `f` register to an `x` register and use the integer merge instruction.

aswaterman, 2019-11-27:
BTW, thanks for making sure this issue didn't get lost!

JamesKenneyImperas, 2019-11-28:
Hello Andrew,

I agree with your logic. The implementation in riscvOVPsim is cleaner if the NaN-box test also applies to `vfmerge.vfm` and `vfmv.v.f` and I would expect that the same would be true in hardware. Assuming that this is agreed by the task group, it should then be explicitly clarified for these two instructions in the specification, since behavior is different to similar-looking instructions in the base floating point ISA.

James.

aswaterman, 2019-11-29:
Thanks for the feedback.  I've written this up in the spec, but am leaving the issue open until the task group reaches consensus.

kasanovic, 2019-12-07:
I believe vfmv.s.f and vfmv.f.s should check for NaN boxing, same as vfmv.v.f .

Because vector registers are unified float and integer, it is straightforward to use integer moves to copy bits around without interpretation if needed.

Will bring up with TG.

kasanovic, 2019-12-07:
I'm going ahead to merge this change for vfmv.f.s and vfmv.s.f as I believe it will be uncontroversial, as it removes an inconsistency where the odd behavior has no real purpose.

commit 86f2c472 (2019-12-07): Make floating-point values moved in vfmv.f.s and vfmv.s.f check
for NaN-boxing, as with other similar instructions.
Closes #339
closed 2019-12-07 by kasanovic