riscvarchive/riscv-v-spec #340: Constraints on vector reduction operations when VLMUL>1

JamesKenneyImperas, 2019-11-28:
Section 15 (Vector Reduction Operations) specifies no constraints on source and destination register values. Is it correct that there are indeed no such constraints, even when VLMUL>1?

For example it looks like the following is valid for VLMUL=2:
    `vredor.vs v10,v10,v12`
but what about this:
    `vredor.vs v11,v10,v12`

Does this not cause element 0 of v11 to be written before being read?

Can you please clarify what, if any, overlap constraints apply?

Thanks,

James.

JamesKenneyImperas, 2019-11-28:
I guess that actually this is probably ok. I assume that the result is accumulated in a temporary and committed only at the end of the instruction.

aswaterman, 2019-11-28:
It’s straightforward to implement reductions in such a way that the destination register is not written until the very end, so no overlap constraint is necessary.

This also relies on the fact that interrupts are always reported with vstart=0, i.e., no partial result is written on an interrupt.

JamesKenneyImperas, 2019-11-28:
Thanks Andrew,

I think I beat you by one minute. Closing this issue.

closed 2019-11-28 by JamesKenneyImperas