riscvarchive/riscv-v-spec #342: Can SEW ever be set greater than ELEN

MarkHillHuawei, 2019-11-29:
Can you confirm/clarify that if SEW is set to be greater than ELEN that vtype,vill be set. I think this is implied but it would be clearer to be more explicit about this.

JamesKenneyImperas, 2019-11-29:
ELEN is by definition the maximum size of a single vector element in bits (i.e. maximum possible SEW). Then, section 6.1 says:

_If the `vtype` setting is not supported by the implementation, then the `vill` bit is set in `vtype`, the remaining bits in `vtype` are set to zero, and the `vl` register is also set to zero._

An attempt to set `SEW` greater than `ELEN` falls into this category, because it is not supported by the implementation.

kasanovic, 2019-12-06:
I believe the spec is clear here so closing this issue.

closed 2019-12-06 by kasanovic