riscvarchive/riscv-v-spec #346: Masking format in vpopc.m and reduction instructions

HanKuanChen, 2019-12-06:
From https://github.com/riscv/riscv-v-spec/issues/189, https://github.com/riscv/riscv-v-spec/commit/95e10633b3fb0253959c030f7083936dbb17473e and https://github.com/riscv/riscv-v-spec/issues/301, I think the masking format in `vpopc.m` and reduction instructions should use `v0` instead of `v0.t`.

kasanovic, 2019-12-07:
For reductions and `vpopc.m`, the mask is including or excluding elements from the pop count, so I think `v0.t` is correct.

closed 2019-12-07 by kasanovic

HanKuanChen, 2019-12-09:
I think we are in a different point of view in mask.
From https://github.com/riscv/riscv-v-spec/issues/189,
`vmerge` use `v0` because no matter what mask is used, destination vector register always takes a value from source. It violates the rule in [Vector Masking](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#53-vector-masking),
> Element operations that are masked off do not modify the destination vector register element and never generate exceptions.

So you change the `v0.t` into `v0` (https://github.com/riscv/riscv-v-spec/commit/95e10633b3fb0253959c030f7083936dbb17473e). That makes sense because people also have same opinion in https://github.com/riscv/riscv-v-spec/issues/301#issuecomment-538627887.

But reduction instructions are special. **They always write the first element, no matter what mask is used.** Take VL=1 as an example, if mask is used, the destination vector register still updates the first element. It breaks the rule in [Vector Masking](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#53-vector-masking). Using `v0` is better than `v0.t` in these instructions.

`vpopc.m` and `vfirst.m` are also a special case. But the destination register is a scalar register. Not in the scope of [Vector Masking](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#53-vector-masking). But I prefer `v0` instead of `v0.t` since they always write new value into scalar register.