riscvarchive/riscv-v-spec #347: Special-casing x0-stride accesses as splats

aswaterman, 2019-12-07:
Since this ISA doesn't have the ability to broadcast scalars in element[0] of a vector register across an entire operation, it will sometimes be useful to splat an element from memory across an entire vector register group.  This can be done with a zero-stride load, save for two issues:

- As currently specified, if the access is to non-idempotent memory, the side effect must be actioned `vl` times, not just once.  (This means that implementations of the current spec that want to optimize zero-stride loads must first check the idempotency PMA; they can't do it based solely upon the opcode.)
- If another hart stores to the same memory location, some vector elements might receive the old value and some might receive the new value.

I am proposing that we add a special case for a stride of `x0`, which mandates that the load is only performed once (if any elements are active) or zero times (if no elements are active).  If the old behavior is ever desired (e.g. when deliberately accessing a non-idempotent device register), it can be recovered by loading 0 into a register and using that register as the stride.

kasanovic, 2019-12-07:
TBD in TG.

kasanovic, 2020-02-07:
Too difficult to maintain semantics of "single atomic memory access" in presence of interrupts or other pipeline events.   Can still perform optimization of only a single memory access if PMA says memory is idempotent, but not required to when rs2=x0.

closed 2020-02-07 by kasanovic