riscvarchive/riscv-v-spec #350: Move mstatus/sstatus VS to bits 10:9

jhauser-us, 2019-12-08:
The current draft proposes adding field VS, the vector context status, to bits 24:23 of `mstatus`/`sstatus`.  But there is an existing empty two-bit space in bits 10:9.  Unless bits 10:9 are being saved for some other purpose, placing field VS there would maximize the future options for allocating fields in the few remaining bits in RV32 `sstatus`.  For example, it is unknown (to me, at least) whether the next two "must have" allocations in `sstatus` will both be 4-bit fields.  If so, placing VS at 24:23 prevents those two fields from both being allocated, since there would no longer be two contiguous 4-bit spaces in RV32 `sstatus`.  For all I know, the addition of field VS may be the best opportunity we have to use bits 10:9.

aswaterman, 2019-12-08:
This is peak bikeshedding. Make a pull request if you want to change it.

jhauser-us, 2019-12-08:
Okay.  #351.

aswaterman, 2020-02-12:
#351 has been merged.

closed 2020-02-12 by aswaterman