riscvarchive/riscv-v-spec #352: vfcvt.x.f.rtz

aswaterman, 2019-12-12:
C conversions from float to int round towards zero.  Currently we can express that with

    csrrwi t0, fflags, 1
    vfcvt.x.f.v v0, v1
    csrw fflags, t0

Unfortunately, the CSR writes can't be hoisted out of a strip-mine loop when the source of the conversion is another floating-point operation that does not want to round towards zero.  This is not inherently problematic---implementations can optimize writes to fflags relatively easily, especially implementations that send commands to the vector unit at commit time.  But it's less than ideal.

Opcode space is not much of a concern for these unary operations; they'd fit into the VFUNARY0 opcode.

cc @jhauser-us

kasanovic, 2020-03-28:
Makes sense, can you put together a PR.

jhauser-us, 2020-03-28:
Ideally I would've brought this up sooner, but didn't notice it before now.

The draft has an existing round-to-odd conversion instruction named VFNCVT.ROD.F.F.W.  What is the logic that determines that the rounding mode gets inserted in the middle of VFNCVT.F.F.W but at the end of VFCVT.X.F?

aswaterman, 2020-03-28:
No logic at all. When I made the PR to add this instruction, I inserted rtz in the middle of the name, rather than at the end.

PR #403 (2020-04-03): Add FP->int truncating conversions
Resolves #352
commit 39bda682 (2020-04-03): Add FP->int truncating conversions

Resolves #352
closed 2020-04-03 by kasanovic