riscvarchive/riscv-v-spec #354: Remove byte, half and word vector segment loads and stores (including strided and indexed)

solomatnikov, 2019-12-18:
Vector segment loads and stores with element width should be sufficient for most use cases.

High performance implementation of byte, half and word vector segment loads and stores add a lot of extra complexity and wiring.

Low performance implementation of these instructions is not likely to be useful.

aswaterman, 2019-12-18:
Agreed on both counts.

kasanovic, 2019-12-18:
I agree this simplification should not be that impactful.

David-Horner, 2020-01-01:
Is the recommendation that they are not in the base 32 bit instructions?
Or is it further suggested that they are not in the non-compressed  instruction base as well and thus not in a likely compressed 32 bit extension?
I ask because 
1) this is an opportunity to start defining fundamental  characteristics of the non-compresses instructions.
2) further extensions to V should be "compressed versions" of the non-compressed base.

solomatnikov, 2020-01-14:
This is a suggestion to remove byte, half and word vector segment loads and stores completely because HW cost and complexity is very high and it's unlikely they are useful for SW

David-Horner, 2020-02-13:
This is a subset of the #362 is the more general case of 
Remove byte, half and word vector loads and stores in all its variants including segment.

Discussion is continuing  on 362 that likely will have bearing here.

commit f7cbeba5 (2020-04-24): Remove widening loads and narrowing stores.
Closes #354, #362.
closed 2020-04-24 by kasanovic