riscvarchive/riscv-v-spec #358: support .vv variants for masked slide1up/slide1down

David-Horner, 2020-01-02:
This variant is a simple, low cost and consistent extension of current instructions:

Description:
    This .vv variant uses the vs1 vector register group as the source for the non-masked elements, replacing rd for this purpose. Regardless of mask the replacement for the "shifted out element" is sourced from vs1. As with .vx variants destination cannot overlap either source. The ,vv non-masked variant should be reserved as it is equivalent to the .vx non-masked.

    It is a natural extension to the instruction and it parallels other instructions variants such as merge.

   It's cost then is comparable to other variants to base instructions.
    
Equivalences:

    Where low bit of mask is 0, vslide1up.vx vd, vs2, vs1, vm replaces the two instruction sequence:

        vmv vd,vs1

        slide1up.vv vd, vs2, x0, vm # x0 not used.

For vslide1down it is equivalent to the corresponding two instruction sequence when the vl-1 mask element is cleared.

Known application:

The DUPH operation in the FFTW3 library can be implemented in a single instruction with a mask of (0,1,0,1,....)

        vslide1up.vv  vd, vs1, vs1, vm

Similarly the DUPL operation in a single operation (with a complement mask)

Addresses design restriction:

To allow slide1up/slide1down to be restartable, thde destination cannot overlap source.
As a result without the .vv variant, a vmv or equivalent is required to address this restriction when it arises.

Alternatives:

Although the two instruction equivalence above could be optimized via fusion or chaining,   two benefits over this approach are provided by the instruction variant:

1) the instruction provides simplification, the requirement to clear the appropriate bit in the mask is removed.

2) for minimal additional gates this optimization is available to even simplest implementations.

Further, chaining and especially fusion (which in the scalar instruction set can be implemented with relatively simple designs) the RISCV synchronous interrupt approach imposes considerable constraints on designs in vector implementation.

David-Horner, 2020-01-02:
An alternative formulation of the .vv variant is to have a default value be used for the "shifted out element" , or perhaps it is better termed the "to be shifted in " element..

If this is the preferred approach then binary zero would be consistent with the clearing that is done with the trailing elements.

This alternative would be equivalent with the stated 2 instruction code regardless of mask value.
Perhaps this simplification and equivalence is preferred.

However, I do not see any other alternative defaults that make sense for most cases.
vs1[0] element for slide1down is a possibility that allows a rotation of elements when vs1 = vs2 and mask is all ones, but it is not as easily implemented for slide1up; picking the vs1[vl-1] element. 

Similarly rotate down is  easy to achieve with the current .vx with a simple sequence 
          vmv.x,s rd, rs1 
          slide1down vd, rs1, rd
with the scalar instruction being operationally quite inexpensive so no large incentive here., 

I suspect algorithms that require rotation will be reworked to rotate down and thus not provide much opportunity for the slide1up.vv instruction to facilitate this "rotation" operation.

I still think the best is as I proposed.

kasanovic, 2020-01-23:
I can see the appeal but OTOH, this requires mask register is setup and might only make sense for even-odd interleaving which could end up being handled with EDIV?

David-Horner, 2020-06-28:
This does not need to be decided before V1.0. 
Deferring also defers the consideration of reserving the non-mask variant as a duplicate of another instruction.