riscvarchive/riscv-v-spec #364: Make handling traps more precise.

David-Horner, 2020-01-24:
1) Describe base trap handling, Introduce trap-complete instructions as those that are bending completion, guaranteed to complete and results are not accessible until complete. Divide instruction good example.

Vector instructions can be complete in this manner.

2) Introduce Vstartable instructions - Precise Restartable 
a) Note that not only pre-start elements are completed but also related state (especially in vxsat and fflags in fcsr)
b) Define restartable precise in which elements vstart and beyond are undisturbed (contain the same element values as if all prior instructions were completed and before  instruction start.
c) no additional state is affected, specifically  vxsat and fflags in fcsr as a result of any internal procssing of vstart and on elements. 

3) introduce Vstartable instructions - Imprecise Restartable  
a) this is an imprecise trap 
b) however, a restart of the trapped instruction recovers the instruction as if it had not been trapped.

4) introduce rerunnable vector instructions that would recover imprecise results.
a) for the trapped instruction (as is the case when vstart is set on trap to zero).
  i) introduce concept that trapped instruction can have imprecise results even when set up for rerun. (vstart = 0)
b) recoverable means that upon complettion the results would be the same as if the trap did not occur.

b) for subsequent vector instructions to the one which claimed the trap.
   i) required concurrent execution of subsequent instruction to the trapped instruction.
   ii) note as above that non-vector instruction require precise trap-complete instructions
   iii) it therefore follows that this instruction is consecutive to the "trapping" instruction
   iv) further extend this to any number of consecutive vector instruction.

5) constraints on execution of a set of consecutive vector instructions following a trap
   i) the trapping instruction must be recoverable by restart or rerunable.
   ii) subsequent consecutive instructions must be recoverable by rerunning the instructions serially.
            
   iii) any or all of the consecutive vector instructions following the trapping instruction may have imprecise state.
   iv) during execution of the set of consecutive instruction,  consecutive vector instructions need not be executed in sequence. (As with 
        For a given instruction in the consecutive set there is no need for prior instruction to be trap-complete nor even to have started instruction specific processing if the instruction unde consideration has all necessary current data,.
  v) the exception to the above rules is that for vector instructions that write a scalar register (x or f) all previous instructions must be trap-complete.
 
From these rules it follows
6) Software is to consider a sequence of consecutive vector instructions as a single super vector instruction that is restartable.
   i) This instruction can be serial executed as individual instructions. On restart it must be execute in its entirety.
   ii) an equivalent instruction sequence  of alternating vector instructions and base instructions  is easily constructed and can be substituted by a debugger or jit runtime, This would allow break points. To single step the component vector instructions a conditional branch may be require to seperate the vector instructions.. 

Note: the assumption is that the OS or Debug environment does not attempt to inject into a super-vector instructions state that was not present at trap.
(eg. changing contents of scalar inputs (x and f) )

David-Horner, 2020-01-24:
requiring consecutive instructions and allowing imprecise results over two or more vector instructions facilitates opcode fusion and instruction chaining with minimal hardware support (eg. not requiring as register renaming.).

kasanovic, 2020-10-26:
Some earlier work along these lines:
Mark Hampton and Krste Asanović, "Implementing Virtual Memory in a Vector Processor with Software Restart Markers", 20th ACM International Conference on Supercomputing (ICS06), Cairns, Australia, June 2006. https://people.eecs.berkeley.edu/~krste/papers/restart-ics2006.pdf

Mark J. Hampton, "Reducing Exception Management Overhead with Software Restart Markers"
Ph.D. Thesis, Massachusetts Institute of Technology, February 2008.
https://people.eecs.berkeley.edu/~krste/papers/mhampton-phd.pdf

But for v1.0, I think we've reached agreement on supporting precise to the instruction, but with restartable within instruction if idempotent.  Or allowing imprecise with accurate report of fault location.

Swappable or more complex restart conditions are left for after v1.0.

commit 00185010 (2020-10-26): Clarified imprecise trap behavior.
Closes #364.
Decided to leave definition of state save/restore instructions for
swappable/resrtartble implementation until after v1.0.
closed 2020-10-26 by kasanovic