riscvarchive/riscv-v-spec #365: What is the case about vsetvl{i} x0, x0, {vtypei|rs2} without keeping VLMAX ?

lihanglin, 2020-01-25:
Section 6.1 says 
>When rs1=x0 and rd=x0, the current vector length in vl is used as the AVL, and the resulting value is only written to vl
>This form of the instruction allows the vtype register to be changed without changing vl, provided VLMAX is not reduced. The current vl value can be read from the vl CSR.

Suppose VLEN=512 and  the following code,

_vsetvli x3, x0, e8, m8
_vsetvli_ x0, x0, e32, m1
_vadd.vv_ v0, v8, v16_

The first vsetvli specified AVL=VL=VLMAX=512 and the second one just changed vtype without changing VL.
What is the value of CSR VL after second vsetvli was executed ? 16 or 512 ?
What is the effective VL for vadd.vv ? 16 ?

David-Horner, 2020-01-25:
What is the value of CSR VL after second vsetvli was executed ? 16.
What is the effective VL for vadd.vv ? yes,16.

To be more precise:

the second [vsetvli]  just changed vtype without **providing a new VL value.**

These all follow from the formula 
VLMAX = LMUL*VLEN/SEW (found in 3.2.2. Vector Register Grouping (vlmul))
that you applied correctly in the first vsetvli.

No real surprises. 
rs1 = x0  is only a shorthand for providing the current vl and doesn't change any of the formula.
And rd =x0 doesn't change any of the rules, only that the new vl value is "thrown away".

aswaterman, 2020-01-25:
@David-Horner rd=x0 does change the behavior, following a spec change a few months ago. rd=x0 & rs1=x0 behaves as you described. rd!=x0 & rs1=x0 means set vl to VLMAX.

lihanglin, 2020-01-26:
A sentence in Table 6. in section 6.1

rd | rs1 | AVL value | Description/Usage
-- | -- | -- | --
0 | 0 | Value in vl register | **Change vtype without changing vl**

To be more precise:
Should it be revised to **Change vtype and use VL as AVL** ?
The second [vsetvli x0, x0, e32, m1] did change VL.

David-Horner, 2020-01-26:
The intent is to retain original vl. 
So it is described as such in the "Usage" column.
However if new vtype reduced VLMAX below original vl, VLMAX is the new vl.
I created a PR stating intent and possible outcome:
           retain 'vl' on 'vtype' change ('vl' = min( new 'VLMAX', old 'vl')) 

#366

vbx-glemieux, 2020-01-26:
I think it should cause a trap instead of changing VL.

The code emitted is making an intentional effort to preserve VL, hence
rd=x0 and rain=x0. If it is changed silently, this will be killer to debug,
and not be congruent with programmer intent.

Guy

David-Horner, 2020-01-27:
vsetvl{i} does not directly trap but instead sets vill.
It would be possible to set vill in this case and then subsequent vector instructions that rely on it trap.

But do programmers really need this safety net. Is it reasonable to provide it. 

This "unintended" result can as readily occur if rd1 uses a register with vl value.
The disconnect is lack of awareness that VLMAX can change and affect vl.

There are many other situations that programmers may assume are less complicated because most time they do not occur.
(See note on Example of implementing isgreater() in 14.11. Vector Floating-Point Compare Instructions)
We don't provide anticipatory traps for these. 

The "expected" vl is deterministic. 
The programmer has to know the current vtype settings to ensure current instruction sequence is operating as intended. 
With this knowledge she knows whether a specific change in vtype can reduce VLMAX. 
In other words the programmer intent is not just to retain a value of vl but also to appropriately set SEW/LMUL and
  whatever other configuration fields might be added.  

In most cases the programmers intent is to retain current vl and current VLMAX by offsetting SEW and LMUL changes. 
If the result is an unintended increase in VLMAX because LMUL is larger than required, the programmers intent is not achieved. 
Granted the subsequent instructions will execute the vl length of vectors without incident (if LMUL aligned), 
and because elements beyond vl are left undisturbed in most instructions such an expanded register group 
may not affect overlapped register groups, until that is an instruction that affects all elements (such as vector integer merge instruction ) is used.
This would also be especially hard to debug.

However, with vsetvli instructions, static analysis can detect both potential bugs. 
And dynamic analysis can identify potential for this bug when vsetvl is used. 

I do think we should establish guidelines, at least document reasoning in situations like this.

I believe the current philosophy of simplicity argues against complicating physical designs to 
   1) provide this special case handling (different behavior than equivalent of rs1 with vl value)
   2) provide a deferred trap when non-trapping behavior can alternately be sufficiently documented.
   3) provide a symantic of programmer intent to instruction behavior

vbx-glemieux, 2020-01-27:
Note: my commentary below, and david horner's preceding commentary,
actually applies to issue #366.

In Issue#366, David proposes that "However if new vtype reduced VLMAX below
original vl, VLMAX is the new vl." when the rd=x0 && rs1=x0 variant of
vsetvl{i} is used.

I think changing vl to VLMAX is a bad idea, because it violates the
assumption that software wants to make that vl is preserved. David's
suggestion destroys a useful "loop invariant" that software wants to
exploit with an exception, which then forces software to always check the
new VL result, thus adding overhead and destroying the intent of having a
low-overhead way of changing vtype.

I agree with David that this variant should instead set vill=1 if the new
vl cannot be preserved.

More verbosity below.

--

vsetvl{i} instructions with rd=x0 and rs1=x0 are for changing vtype without
changing vl.

I expect this variant will be most useful within inner loops where the data
width is changed within the inner loop. if the arch spec allows vl to
change (to VLMAX) because the new vtype isn't compatible, then it will
violate an invariant inside the inner loop and lead to a silent data
corruption problem which is hard to debug.

the main advantage of keeping the current vl is that software doesn't need
to keep track of the current vl in a scalar register (adding to scalar
register pressure), and it doesn't need to query the VL CSR. this leads to
lower overhead when changing vtype (but keeping vl) within an innermost
loop.

when changing vtype (but keeping vl) in an outer loop or straight-line
code, overhead isn't an issue and this instruction variant has little
advantage except perhaps code density.

this variant shouldn't be used across function call boundaries etc, since
it would be better/safer to use the other variants and that check the
resulting AVL.

---

this instruction variant should trap somehow. the suggestion of setting
vill and letting the next compute instruction trap is perfectly fine and is
consistent with the rest of the spec.

without some kind of trap, the current spec allows vl to change to VLMAX.
this enable silent data corruption, and is harder to debug, which is why I
don't support the ability to change vl.

---

other variants of vsetvl{i} can change to a change vl, but software expect
vl to change, and in those cases rd can be specified to get the AVL, or
software can read the VL CSR.

guy

David-Horner, 2020-01-31:
I'm hoping this is a start to a formal doc on guidance for the usage of vsetvl variants,

In reviewing Guy's response I realized that there are two significant benefits from the x0,x0 formulation,
   one he mentioned - avoid obtaining or retaining vl in x[rs1] 
   but also - decouple instruction from integer processor and x register set.

Additionally, Guy highlights the ambiguity and complexity of the special meaning given to x0 usage.
This iteration of rd and rs1 with x0  is unusually complex.
 (only rivalled by JALR as it relates to the Return Address Stack with both rd and rs1 interacting for 5 distinct states).

Use of x0 for rd should be familiar to users of RISCV. It is used consistently here;  the return value is discarded.
However, in conjunction with rs1 x0 usage, the rd x0 usage becomes complex.  
We have four distinct cases, three  that have specific formulation that are non-obvious or precluded other expected formulations.

I propose that the 3 unusual cases be given pseudoinstruction names to help clarify their meanings and to help avoid "common" errors.

In addition to the base vsetvl{i} with rs1 !0  (with typearg either immediate or register parms)  :
  
 **vsetvlzero{i} rt, typearg** - returns 0;  generates sequence: li rt,0 ;vsetvl{i} x0, rt, typearg
         ( rt !0, prohibited due to special case below)
 **vsetvlmax{i} rd, typearg** - returns MAXVL - generates vsetvl{i} rd,x0, typearg (where  rd !0 )
 **vsetvlmin{i}  typearg** - set vl = min(MAXVL, vl)  - generates vsetvl{i} x0,x0, typearg

Providing these pseudoinstructions and limiting use of the variants where rs1 = 0, 
either by prohibition or warnings, would substantially aid in avoiding unintended outcomes.

**Traps for vsetvl{i}**

In RSICV there is a high bar for the inclusion of trapping behaviour.
That is, only when the situation leaves no choice, there is no other alternative to ensure correct hardware execution, 
does the ISA raise an exception.
I believe this is one of the design principles (a comprehensive list of which we should compile). 

This is seen in the vsetvl{i} instruction itself.
On an illegal configuration it does not trap but sets vill to allow software to determine hardware specific limitations/abilities.
Only when the attempt to execute a vector instructions (variant) which the hardware does not support does it raise an  illegal instruction trap.

> "Earlier drafts required a trap when setting vtype to an illegal value. However, this would have added the first data-dependent
> trap on a CSR write to the ISA"
> 

The principle is seen in RISCV the  IEEE 754 "exception" functionality; accomplished without traps nor additional instruction support:

> "As allowed by the standard, we do not support traps on floating-point exceptions in the base ISA, but instead require explicit checks of the flags in software. We ... ultimately
> chose to omit [additional] instructions to keep the ISA simple."
> 

This principle is seen in integer arithmetic instructions as well:

> "No integer computational instructions cause arithmetic exceptions."
> "We considered raising exceptions on integer divide by zero, ... however 
> .... only a single branch instruction needs to be added to
> each divide operation."
> 

> " We did not include special instruction-set support for overflow checks on integer arithmetic ...
> For general signed addition, three additional instructions after the addition [and two additional temporary registers] are required,
> ....."

**Need for EE support if trap supported.**

As mentioned in the integer divide:

> "We considered raising exceptions on integer divide by zero, with these exceptions causing a trap in most execution environments. However, this would be the only arithmetic trap in the standard ISA  and would require language implementors to interact with the execution environment’s trap handlers for this case. "

This is a substantial support burden and therefore will probably be get little support for vsetvl.

**The current vsetvl plan of record appears to parallel the approach for integer arithmetic:**

**Firstly** the "programmer" must anticipate the need to detect vl truncation [as with integer overflow in integer arithmetic].
[... 15 lines cut ...]
     trap on widening operations when LMUL already max value
     trap on incompatible operation and element length (e.g. no byte float yet)
These restrictions make dynamically configurable vector code difficult and/or inefficient 
compared to  selectable code segments targetting specific vtype sets. 

It should also be noted that widening/narrowing operations apply modified offsetting SEW/LMUL to operands  in the instruction allowing invariance in vl element processing.
As a result corresponding vsetvl{i} instructions are executed to allow non-widening/narrowing instructions to operated on the results.
In this common use case vl remains unchanged throughout the code and no runtime checking is required.

In summary there exists sufficient guidance to determine if a runtime check is needed and if checking can be avoided altogether.

**Secondly,** as with integer arithmetic, efficient code sequences exist to detect truncation at run time
       i) two temp registers required for the general case and additional instructions (two vs three) :
               cssr t0, vl                        ; read current vl
               vsetvl t1, t0, a0             ;  apply type parameter a0 
               blt  t1,t0                         ;  vl reduced
       ii)  no need to read vl if already in a register (one branch and one temp register)

Suggestions to mitigate unintentional truncation of vl.

1) warning or prohibition for x0 rs1 vsetvl{i} usage. 

2) specific pseudo instructions for 3 exotic variants

3) static and dynamic analysis of vl usage (with assertion directives?)

aswaterman, 2020-01-31:
Thanks for thinking this through.  It is definitely philosophically consistent to permit this sort of misuse without raising an exception, but it makes sense to help catch these bugs in software.  Using (requiring?) pseudo-ops like `vsetvtype[i]` will surely help.

David-Horner, 2020-01-31:
pseudoinstructions for vsetvl special cases ---  pull request

vbx-glemieux, 2020-01-31:
thank you for the thoughtful exposition. i’m still digesting it — i think
it is a very well constructed proposition.

i’m still not sure I agree that the x0,x0 variant should be allowed to
change vl to vlmax, so i disagree with calling the pseudo instruction
vsetvlmin{i}. instrad, i think it would be better named vsettype{i} rather
than “set vl”, because it’s intention is to avoid changing vl.

as for whether vill is set, i’m still debating it myself and would love to
see other perspectives. I did see Andrew’s response already and that still
has me leaning towards supporting the use of vill, but i could go either
way.

guy

aswaterman, 2020-01-31:
While I'm still very much in the "let the programmers hang themselves" camp, I agree that, if we did wish to add ISA support to catch this case, we would do so via `vill`.  That way, we wouldn't add any new data-dependent exceptions.

vbx-glemieux, 2020-01-31:
maybe I can argue it this way:

logic already exists to set vill for other instructions.

likewise, logic already exists to set vl to vlmax for other variants of
vsetvl.

ie, both are equal in implementation cost.

from this, i would conclude that the best option is to do what is better
for programmers. it is also more natural to provide a guarantee in the spec
that “this variant will not change vl” than to say “except when...”.

there are lots of things in the spec that apply to only one instruction or
apply differently to different instructions. the vector isa is far from
orthogonal! as a simple example, vcompress is currently the only
instruction that reads one source operand as elements of width sew, and the
other source operand as a mask (as a result, i may soon suggest that
vcompress should only be allowed to use vs0 as a mask source, but that’s
another issue i haven’t yet officially raised)

guy

aswaterman, 2020-01-31:
Yeah, AFAICS, the HW cost is neutral.  I just haven't convinced myself that illegalizing this behavior is actually more beneficial than leaving it alone.  Much of the discussion on this thread has presupposed that it's inherently erroneous for `vsetvtype` to be used in a way that reduces VL.  While it's clearly true that, in the "keep the SEW:LMUL ratio constant" regime, `vsetvtype` should never reduce VL, are we certain there aren't legitimate use cases that might do so?

(Re: vcompress: the vs1-as-register-group and vs2-as-mask thing is not actually a special case in my implementation.  For other reasons, I found it necessary for the vs1, vs2, and vd pointers to advance at independent rates.  But I wouldn't be upset if the TG chose to adopt your suggestion.)

David-Horner, 2020-01-31:
>I just haven't convinced myself that illegalizing this behavior is actually more beneficial than leaving it alone.

I agree. A trap (direct or indirect) is not an advantage to an operational program. If the program provides a recovery mechanism to be invoked when the exception arises it has already anticipated the error case and could as readily check for it explicitly with low overhead and substantially less impact when it does occur. If it does not or cannot provide a recovery mechanism, the program is as good as useless when the trap occurs. Are we to anticipate that the EE is going to be able to correct such a program deficiency? I think not. 
As a result, I believe the use case is almost exclusively to provide debug support. And nothing prohibits a debug environment from enabling a trap on vsetvlmin ( or whatever it might be called) if vl changes.

 >Much of the discussion on this thread has presupposed that it's inherently erroneous for vsetvtype to be used in a way that reduces VL. While it's clearly true that, in the "keep the SEW:LMUL ratio constant" regime, vsetvtype should never reduce VL, are we certain there aren't legitimate use cases that might do so?

The legitimate cases for a type change to reduce vl are readily supported by the "stripmining" varient, so there is no pressure to have vsetvtype support it. The discussion presuposing no change in vl, to me, recognizes an opportunity to enforce the SEW:LMUL constraint. If we desire a fail safe for programs that intentionally use this variant to maintain that constraint, then the trap should occur not when vl changes, but when MAXVL changes. And I believe it should in this case be an immediate trap rather than through vill. Similar as reasoned  earlier, if the program anticipated checking vill after this vsetvtype it could validate before the instruction (for that matter outside a inner loop) that the current vtype is MAXVL invariant with the about to be applied. And if the program does not anticipate the mismatch, then trapping on the instruction provides better debugging information than whatever vector instruction will later trap.

I prefer that we not mandate a trap for this scenario, and rather focus attention on support to ensure code is readily (simply and verifiably) written to specifications.

Any implementation can elect to provide run time support for immediate trap either through debug or EE without being in violation of the standard.

vbx-glemieux, 2020-01-31:
what about simple implementations without such debug support? i’ve done
many vector implementations on FPGAs and debug is the last thing I want to
add (too much work, too much extra area, compromises Fmax, etc).

the implementation cost for setting vill is nearly  zero. it costs the
exact same as changing vl to VLMAX, so the arguments here are about being
pedantic.

myself, i would much rather have a processor that behaves as expected and
detects incorrect program behaviour. it’s nice to get a page fault or
divide-by-zero fault etc.

i was the one to propose this vsetvtype variant, and i did it with the
complete intention that it should be used only when keeping the SEW:LMUL
ratio fixed. ie, vl should be unchanged. other uses of vsetvtype are
possible, but in all of those cases they should use the regular vsetvl and
check the result saved to rd — there is no need to check vill in the CSR or
the vl CSR. so, the only time this variant is used, it must not change vl.

along the way, someone thought to more fully describe the behaviour on the
corner case when VLMAX changes.

now, looking back without this history, a third party might stay “the
vsetvtype instruction lets me change SEW and LMUL, so i can set it to
anything i want. i wonder what it does when VLMAX changes? oh that’s good
it changed vl for me, how clever. i think i will use this instruction and
then add a bunch of code to test when vl changes because i might be an
idiot programmer. nah, i’m a supercoder and i’ll get this right the first
time.”

sorry, that last part was my sarcasm leaking through.

like i said before, i’m still digesting your earlier argument about the
merits of this. i do think the ISA should be as orthogonal and consistent
and predictable as possible.

that said, so far, i am convinced that the vill approach does zero harm and
saves from significant harm: it will totally eliminate a class of bugs that
may otherwise go by undetected — much like buffer overruns and other
security holes. what i haven’t made myself sure of are your points on
consistency etc.

TL;dr:  the “new” thing I am repeating here is the fact that this
instruction variant should ONLY be used when vl should not change. if VLMAX
does change, this was the wrong instruction to use, and the surrounding
code should behave differently. it is incorrect to assume the software
should (pre)verify that this was the correct instruction to use on the fly.

guy

David-Horner, 2020-01-31:
>what about simple implementations without such debug support? ....the implementation cost for setting vill is nearly zero. it costs the exact same as changing vl to VLMAX, so the arguments here are about being pedantic.

Agreed. First, that the variants we have considered so far have low implementation cost (over the base). Second	, that debug support should not be required to support trapping; and it is not. An illegal instruction exception is required by all most vector instructions (vsetvl and whole register moves excepted).  The cost to include it for vsetvtype must be very small. M-mode can simulate whatever behaviour it wants within the trap including forcing a compatible SEW:LMUL. To do that however, the trap should not modify vtype nor vl, which I believe is standard RISCV behaviour.  

>i would much rather have a processor that behaves as expected and detects incorrect program behaviour.

I agree it is desirable to have processor execution be intuitive (behave as expected). However expectations are varied. RVWMO is unexpected to those with sequential consistency mindset. 

Other than for the most limited definitions of errant behaviour, processor detecting incorrect program behaviour is currently unachievable. However, this instruction, detecting a change in MAXVL as the programmer requested, is indeed desirable.

 > i was the one to propose this vsetvtype variant, and i did it with the complete intention that it should be used only when keeping the SEW:LMUL ratio fixed.  vl should be unchanged. other uses of vsetvtype are possible, but in all of those cases they should use the regular vsetvl and check the result saved to rd — there is no need to check vill in the CSR or the vl CSR. so, the only time this variant is used, it must not change vl. 

That is helpful to know the original intent, concept and expectations,

>along the way, someone thought to more fully describe the behaviour on the corner case when VLMAX changes. now, looking back without this history, a third party might stay “the vsetvtype instruction lets me change SEW and LMUL ....

The chosen encoding (rd=rs1=0) is useful in other use cases and thus competes.
If the “trap on MAXVL change” were instead encoded as a flag in the vtype immediate field or the corresponding bit in the rs2 register the functionality could be applied to any vsetvl{i} instruction or variant. I mention this to help with some of the conflated issues at play. Envisioning a situation in which we can “have our cake and eat it too” should help to objectively look at alternatives with the intent of finding the best.

As intended the vsetvtype instruction only detects this limited “incorrect program behaviour”. It does not ensure nor assist in maintaining MAXVL. What if we provide an alternative that does its best to maintain MAXVL? Consider, vsetsew (or alternately vsetlmul) which would assist retaining MAXVL by generating the SEW value from the provided LMUL value (or visa-versa). There are still cases which cannot retain MAXVL with the provided SEW(LMUL) and so the exception behaviour still needs to be addressed. We needn’t consider the actual encoding at this point. But assessing the relative importance of whether SEW or LMUL should be derived should help with the general question of what functionality is required to develop “correct” programs. Is this assist in reality valuable in avoiding erroneous code? Is software solution as effective? So, reducing conflation here I hope will be helpful. 

> i’m still digesting your earlier argument about the merits of this. i do think the ISA should be as orthogonal and consistent and predictable as possible.

I appreciate you spending this effort. Sorry to add further points to consider. There may be merit in the “flag to enable exception” and the “set sew/mul” suggestions themselves. But the considerations you are making of overall benefits of various approaches are likely to give even greater insight on how to ensure RISCV vector programs are reliably constructed (even if we cannot guarantee error free).

>so far, i am convinced that the vill approach does zero harm and saves from significant harm: it will totally eliminate a class of bugs that may otherwise go by undetected 

I see direct (precise) trap  providing the same benefit with the same level of harm. With the additional benefit of determining unambiguously the vsetvtype that first raised vill. A program could branch from many locations to the trapping vector instruction, even when a vsetvtype immediately precedes it. (which would mislead many investigators to assume it was the preceding vsetvtype and not another).

I will also note at this point that vsetvtype cannot check invariance of MAXLV if type has vill set. It could still check for vl unchanged. I'm still undecided on what to do in this case. I lean towards exception when vill is set at invocation..

>what i haven’t made myself sure of are your points on consistency etc. 

Thank you again! For this response and your continued investigations.

>TL;dr: the “new” thing I am repeating here is the fact that this instruction variant should ONLY be used when vl should not change. if VLMAX does change, this was the wrong instruction to use, and the surrounding code should behave differently.

I was initially puzzled by this summation. I think I got it now, combining the two sentences: “The programmer should not use vsetvtype when they know VLMAX (and thus possibly vl) can change and the program can still execute correctly”. The corollary is that this instruction can be used in any other case to detect potential MAXVL mismatch. I hope I got this correct. 

>it is incorrect to assume the software should (pre)verify that this was the correct instruction to use on the fly. 

But this still puzzles me.

kasanovic, 2020-07-03:
I believe this issue has distilled down to whether the "vsetvli x0, x0, imm" form sets vill when vl is changed as a side effect of the new SEW/LMUL setting.

I appreciate the alternative position that if vl would be changed, vill is set instead, such that subsequent use causes a trap to 
help debugging.

But I am in favor of maintaining the current specification.  In many cases, an assembler could warn statically (not waiting until runtime) that this situation could arise.   I am also not 100% convinced that there are no actual uses for the current behavior.

kasanovic, 2020-07-03:
The consensus appears to be to keep current behavior for vsetvli x0, x0, imm.

However, it would be useful to define assembler pseudo-instructions that indicated intent when using variants of this instruction.  I'd propose that the full version "vsetvli x0, x0, immfields" implies that the user is fully aware of consequences, and that a pseudo-instruction with proposed syntax "vsetvtype immfields" is used with assembler warning when it notices that vl can also change.

vbx-glemieux, 2020-07-04:
sorry, I think it is too late as the decision has been made, but I think it
is a bad idea in general to silently accept a new vl.

(1) the assembler warning is a good idea. unfortunately, this only helps at
compile time — it cannot check intent at runtime, during debugging, or with
disassembly dumps (it’s the same opcode, so it can’t be distinguished). so,
the only time you get the warning is when you’ll never actually encounter
the bug.

(2) what about vsetvl (non-immediate version), because in many cases this
can’t be checked statically? what about when the former vtype cannot be
ascertained, so it’s not clear whether vl changes or not?

I’m still in favour of the trap when vl changes. I’ll re-explain this
below...

If the program had intended to accept a different/changed vl, they would
specify an rd other than x0. Any other interpretation is wrong, in my
opinion, and erring on program correctness seems better to me.

If we don’t trap, then software must technically check the new AVL in rd
after executing vsetvli, which means keeping the old AVL in a register and
doing a comparison. this adds register pressure. this could also be in a
loop, eg toggling between multiple dynamic vtypes using vsetvl, which adds
a lot of checking overhead. this is all cleared by having an exception to
assert programmer intent: that vl must not change here. note that such
vtype changes commonly occur with mixed-width data types.

we already have other such exceptions as assertions for programmers, such
as null pointer dereferences.

TL;dr:

— by not generating an exception on a changed vl, performance of some loops
will be negatively impacted by contributing to register pressure, forcing a
comparison, and forcing software to write a handler in the event of failure
— generating an exception allows programs to run faster and safer, with no
perceived downside
— there is no way to “mimic” the exception behaviour efficiently if it
isn’t provided; on the other hand, it is easy to have software ignore the
exception simply by specifying rd != x0.

Guy

vlsetvli can be statically checked by software, but what about vlset

kasanovic, 2020-07-19:
In the 2020/7/17 meeting we went over this issue again.  The consensus at the meeting was to stay with PoR, which might alter vl when SEW/LMUL ratio changes rather than set vill.  The  sentiment was that the debugging value was small, and there was some potential uses for the behavior.

Rereading Guy's comments above again, the register form of vsetvl case was not considered in TG discussion.  Though that is a relatively rare form of the instruction., and in general won't benefit from static checking (e.g., for unsupported SEW).

We did discuss making the behavior in these cases "reserved", so we could redefine those cases in future to perform some useful alternative vl setting function, but this would not enable forward compatibility via emulation (neither would setting vill), so the decision was that a different encoding should be used when a different function is required.  This makes me reconsider our PoR.

For a while, I've been trying to think of cases where a vsetvli x0,x0, with different SEW/LMUL would perform a useful function and I struggle with coming up with plausible use cases.  One example I had thought of was when taking a vector of say bytes and wanting to cast these into vector of halfwords.  The instruction could be used to change vl appropriately, though only in one direction and this only really works when vl=VLMAX initially, which we can accomplish more reliably using the rd!=x0, rs1=x0 variant.

More methodically, when new SEW'/LMUL' > existing SEW/LMUL, then vl might either stay the same or be clipped.  I struggle to see the utility there for portable software.

When SEW'/LMUL'<SEW/LMUL then vl will be unchanged, but upper portions of vector register groups at SEW' will be inaccessible.  There might be use there, but it  seems a bit esoteric.  While one could possibly use the lower portions of a vector register group to work with other vector register groups, the same effect would be had simply using a different LMUL to keep the upper portion undisturbed (and maybe set tail-undisturbed for fractional LMUL.

I am coming around to the idea the vsetvl{I} x0, x0 form should set vill if SEW/MUL ratio would change, leaving vl unchanged in all cases.  This might also simplify hardware as there is no potential change in vl with this instruction.

commit 579fef98 (2020-10-26): Make vsetvli x0, x0, imm form of instruction be reserved if it would cause vl
to change.
Closes #365.
closed 2020-10-26 by kasanovic