riscvarchive/riscv-v-spec #368: Z Sub-Extensions of Vector

Nelson1225, 2020-02-06:
Hi,

According to the vector spec, we have four sub-extensions now,
`Zvlsseg` (vector load-stroe segment), `Zvamo`, `Zvqmac`, and `Zvediv`

There are two questions about the z extensions of vector,

1. Should we allow the vector sub-extensions are used without the basic v-ext?
For example, if we want to use the `vamoaddw.v` instruction, then could we just enable the `Zvamo` extension without V-ext? Or should we have to enable both basic V-ext and `Zvamo`?  And if we allow the former, then should the basic V-ext is implied to the `Zvamo`?

I think set both basic V-ext and `Zvamo` is more reasonable to me.  But it seems that the B-ext allows that we can use the `Zbb` (`Zbe`, ...) instructions and just need to enable the `Zbb` (`Zbe`, ...) extension without the B-ext.

However, `Zvlsseg` and `Zvqmac` also meet the same problem.

2. If the `Zvediv` isn't set, then should we raise illegal instruction exceptions when the `vsetvli` instruction set the `vediv` field of `vtype` register?

Thanks :)
Nelson

kasanovic, 2020-02-07:
Assumption is vector base is present before adding any of current Zv extensions.  there could be future Zv subsets.

if Zvediv is not implemented, then vill will be set in vtype  after vsetvli and subsequent attempted use will trap.

closed 2020-02-07 by kasanovic

Nelson1225, 2020-02-10:
Got it, thank you very much :)