riscvarchive/riscv-v-spec #372: opcode order: M instructions in V-extension

Rolfy360, 2020-02-11:
Just a comparison, though I don't think it's a big problem.
V-ext 
finct6_inst
100000_vdivu
100001_vdiv
100010_vremu
100011_vrem
100100_vmulhu
100101_vmul
100110_vmulhsu
100111_vmulh

M-ext
funct3_inst
000_mul
001_mulh
010_mulhsu
011_mulhu
100_div
101_divu
110_rem
111_remu

aswaterman, 2020-02-11:
Shouldn’t translate into actual HW cost, since bits are coming from different positions in scalar vs. vector.

I didn’t review the rest of the encoding, but if the current encoding of vdiv etc. lines up better with other vector instructions, then forcing them to be consistent with the scalar instructions might actually increase cost.

aswaterman, 2020-02-13:
Current encoding matches up slightly better with signed/unsigned encoding for other vector instructions.  As you point out, it's not a big deal either way, but I think current design is slightly better than switching to M-extension encoding.

closed 2020-02-13 by aswaterman