riscvarchive/riscv-v-spec #373: reserve encoding where rs1=0 and .vx and .vi instructions reduce to simpler

David-Horner, 2020-02-12:
When rs1=0 each of the following op.vx and op.vi  are equivalent to vmv.v.v

> vadd, vsaddu, vsadd,    vor
>   vsll, vsrl, vsra, vssrl, vssra
>   vslideup, vslidedown

When rs1=0 each of the following op.vx and op.vi  are equivalent to vnsrl with rs1=0:

>  vnsra, vnclipu, vnclip

Note: the vxsat bit is not set for any of the above and regardless of vxrm encoding no rounding occurs.

When rs1=0 the op.vx and op.vi  

>of  vaadd  and vasub   are equivalent to vssra with shift count of 1.
>of  vaaddu and vasubu  are equivalent to vssrl with shift count of 1.

I propose we reserve these encodings.
The identification of these is very simple, rs1=0.
I will leave it to others to estimate the gate cost, but I expect it to be minimal and less than the circuitry required to identify unused opcode space.

As much as we anticipate the 64bit vector instruction formulation the likelihood is there will be motivation to pack additional operations into the 32bit allocated opcode space for some time to come.
The C-extension made substantial effort to remove redundant encoding and was successful in re-purposing what would have been lost opcode space. Even though its opcode space was very limited it still has available space for expansion.
The v-extension does not have as much pressure to eliminate waste but we need to act now if we hope to accommodate such pressure later. When the extension has matured near the tipping point to 64bit encoding there will be curses about design decisions that wasted space for some requested purpose or other. We can at least have eliminated this cause for denigration.

David-Horner, 2020-02-12:
**I made an attempt to itemize the remaining rs1=0 equivalent opcodes.**
The lists below are in addition to the ops that have equivalent instructions. 

For the following ops the op.vx and op.vi are equivalent when rs1=0:

>  vmseq, vmsne, vmsle, vmsgt, vmsleu and vmsgtu.

For the following ops the op.vxm and op.vim are equivalent when rs1=0:

>vadc, vsbc, vmadc and vmerge.

Note: vadc and vsbc are equivalent when rs1=0.
   As with vnsrl, one of them should become the stipulated (non-reserved) encoding.

**Following the C-extension decision and logic, make x0 the reserved encoding.**
This allows micro-architecture early in the decode pipeline to request rs1 from the integer register file without checking rs1 value. The attempted instruction can be aborted later by various mechanism once X0 is detected. The effects of an illegal instruction process far out way any delay in detection.
There is substantial benefit to avoiding functional X0 access as many vector implementation will loosely couple the integer register set and the vector processor. This is, therefore,  even more relevant than in the base instruction case which justified preferring imm over X0 ,
 
Further to benefit the micro-architecture , the imm values can be extracted from the instruction and sent to the actual processing units without checking for 0 values.

kasanovic, 2020-07-03:
Any encoding space savings has to be weighed against the implementation cost of trapping reserved opcodes.  Many implementers will be motivated to build non-conforming implementations that simply do the obvious with these encodings.

David-Horner, 2020-07-03:
if these are designated reserved in the V1.0 candidate review allows feedback for this and any other feature that is seen as arduous.
If not designated reserved, they will be unrecoverable in later spec versions.
Up until we have repurposed them, fallback is possible to make such previously non-compliant implementations accepted..
Previous implementations can readily be made compliant with the new standard by trapping and executing the equivalent instruction. 
By then software ecosystem will already avoid the trapping construct. 
Indeed, even without these redundant formulation being reserved it is highly unlikely that compilers will use and intentionally restrict  a more complex instruction rather than use the simpler form that meets the software requirement.

kasanovic, 2020-07-03:
Meeting agreed with reserving while acknowledging implementers might be tempted to treat as non-conforming variants of existing instructions.  Assemblers should check for reserved usages of these opcodes.

David-Horner, 2020-07-10:
Does this format work:

Instruction to reserve when V or I zero|funct6 encoding|V|I|replacement instruction
-|-|-|-|-
vadd|000000|Y|Y| vmv.v.v

if so I will continue to flesh out the table.

kasanovic, 2021-06-09:
I'm going to recommend we not mark these as reserved.

There is a some small amount of logic, and some additional documentation/verification complexity, in decoding which instructions are reserved when rs1=0.

The gain in usable opcode space is tiny relative to the entire 32b encoding space, and these would be in the least desirable portion of the opcode map, so I do not believe the tiny benefit outweighs the small cost.

David-Horner, 2021-06-09:
Still have reservations.
So what I propose is that we still provide the table as preferred
instruction formulation. That will provide some value and if we end up
needing more encoding space they can be actively depreciated.