riscvarchive/riscv-v-spec #374: Please clarify interaction of misa.V and mstatus.VS

JamesKenneyImperas, 2020-02-12:
Both `misa.V` and `mstatus.VS` are in some sense enables for the Vector Extension. In an implementation where `misa.V` is writable, what is the visible effect of clearing it to zero when `mstatus.VS` is non-zero (i.e. Vector Extension is currently enabled)? I can see at least three possibilities:

1. Changing `misa.V` to zero has no effect on `mstatus.VS` (which remains writable) but disables all vector features (in other words, Vector Extension is enabled only if both `misa.V` _and_ `mstatus.VS` indicate it). This is not consistent with the V extension being absent, because `mstatus.VS` remains writable.

2. Changing `misa.V` to zero also clears `mstatus.VS`, which then becomes WIRI. This is consistent with the V extension being absent. Changing `misa.V` to one later would cause the Vector Extension to be re-enabled in the Off state.

3. Changing `misa.V` to zero masks `mstatus.VS` so that its apparent value is zero and makes it WIRI. This is consistent with the V extension being absent. Changing `misa.V` to one later would cause the Vector Extension to be re-enabled in whatever state it had when previously disabled.

Depending on the answer to this, I think I will have some follow-up questions: my ultimate goal is to  understand exactly how access to the `fcsr`, vector CSRs and various classes of vector instructions is affected by all possible legal combinations of `misa.F`, `misa.D`, `misa.V`, `mstatus.FS` and `mstatus.VS ` in specification versions 0.8 and 0.9.

Thanks.

aswaterman, 2020-02-12:
The FS field is allowed to be present even if the F extension is not present; this supports transparent emulation of the floating-point extension so that even S-mode context-switch code doesn't know that it's missing.

Now, I don't think transparent emulation of the vector extension is nearly as useful.  I just mean to say that option 1 is not inconsistent with current practice.

kasanovic, 2020-07-03:
To be consistent with FPU, we'll go with option 1.  This also leaves most implementation freedom and has least implementation complexity.  If either misa.V or mstatus.VS are clear, vector instructions will trap.  xstatus.VS can exist without misa.V being set.  We could mstatus.VS bits as another extension, whose existence is implied by the V extension.

commit e8412137 (2020-07-03): Clarified interaction of misa.v and mstatus.vs bits.
Closes #374
closed 2020-07-03 by kasanovic