riscvarchive/riscv-v-spec #376: Special lmul code in vsetvl{i} to derive LMUL from existing SEW/LMUL and provided sew code.

David-Horner, 2020-02-13:
As mentioned in the TG we suggested widen LMUL to 3 bits with 7 (explicit) states.
I suggest we name them l1,l2,l4,l8 as before and lf2, lf4, lf8 the fractional 1/2, 1/4, 1/8 respectively.
lf16 may be useful in larger machines, but already lf8 cannot be used on minimal RV32 machines.
( I recommend we use lf2, lf4 and lf8 rather than f2, f4, f8 to avoid confusion with float and to associate with lmul as a fraction value)
**I propose the remaining value be used to derive LMUL from the existing SEW/LMUL and provided sew code.** 
Specifically, new LMUL = ( new sew / old SEW )  * old LMUL which retains the initial SEW/LMUL ratio. 

Tentatively, I use lx (lmul solve for x) as the mnemonic.
I recommend the immediate variant using lx have the pseudo-opcode  vsetlmuli. 
As a shorthand I refer to vsetlv that uses the lx code in rs1 as vsetlmul. 

Removing the fixed size load/stores motivates the increased lmul range and increases the need to issue additional instructions.
This facility compliments the increased lmul range and mitigate the need to explicitly specify the correct lmul in the additional associated vsetlmul{i} instructions.
As a result the first vsetvl{i} instruction establishes the SEW/LMUL ratio and subsequent vsetlmul{i} instructions maintain it. 

Notably, values of vl that  were legal in the original vsetvl{i} are legal in subsequent (legal) vsetlmul{i} instructions. 

The additional fractional lmul values provide a substantial dynamic range and reduces the possibility of an illegal calculated LMUL. 
Even so, the calculated LMUL can be out of range for vsetlmul{i} which will set vill.

Other proposals have the objective of changing SEW and LMUL and enforcing that vl and/or the SEW/LMUL ratio not change.
This proposal compliments those recommendations.
In particular it addresses issue #365 (What is the case about vsetvl{i} x0, x0, {vtypei|rs2} without keeping VLMAX the same)
when rs1 is x0 to use the current value of vl , the use of lx ensures vl, if possible, will remain the same as will VLMAX and SEW/LMUL ratio and if not possible vill will be generated.

**To have our lf16 and lx too.**
As I recommended initially the lmul code in vsetvl{i} that would otherwise be used for lf16 is used for lx.
That means that vtype lmul field cannot have a lf16 value set by an explicit lmul code.
However, we could allow lf16 to be calculated and stored by an instruction using lx.
A read of vtype will correctly show the lf16 code.
Restoring vtype from the value read from vtype is a multistep step process.
    If lmul code in not lf16 then issue vsetvl with save vtype in rs2
    if lmul code is lf16
        change sew code in rs2 to one larger than provided. (e.g. use e16 if e8 in save vtype)
        set lmul in rs2 to lf8
        issue vsetvl with revised sew and lmul vtype in rs2
        change sew code in rs2 back to original (current example from e16 back to e8)
        change lmul in rs2 to lx 
        issue vsetvl with original seq and lmul=lx vtype in rs2
        lmul will now be lf16 and sew code same as saved.
This approach will work for all values of sew and lf16 that are valid in vtype.
The question is if lf16 is worth the extra effort, or does it potentially cause confusion/failures for application code.

David-Horner, 2020-02-13:
I recommend the codes for the new LMUL be in order of increasing VLMAX value as so:

......vlmul...... | mnemonic | LMUL | VLMAX | #groups | Grouped registers
-- | -- | -- | -- | -- | --

0 | 0 | 0 | lf16 / lx | 1/16 | VLEN/SEW/16 | 32 | vn (single register, lower 1/16th)
-- | -- | -- | -- | -- | -- | -- | --
0 | 0 | 1 | lf8 | 1/8 | VLEN/SEW/8 | 32 | vn (single register, lower 1/8th)
0 | 1 | 0 | lf4 | 1/4 | VLEN/SEW/4 | 32 | vn (single register, lower 1/4th)
0 | 1 | 1 | lf2 | 1/2 | VLEN/SEW/2 | 32 | vn (single register, lower 1/2th)
1 | 0 | 0 | l1 | 1 | VLEN/SEW | 32 | vn (single register in group)
1 | 0 | 1 | l2 | 2 | 2*VLEN/SEW | 16 | vn, vn+1
1 | 1 | 0 | l4 | 4 | 4*VLEN/SEW | 8 | vn, …​, vn+3
1 | 1 | 1 | l8 | 8 | 8*VLEN/SEW | 4 | vn, …​, vn+7

This allows for a quick "andi and bz" check of vtype for the "special" LMUL 1/16 

It also allows the formula 
        new LMUL = ( new sew / old SEW )  * old LMUL 
to be implemented as
      vlmul = new vsew - old vsew + old vlmul

The first column could be inverted to allow the original code values with no appreciable hardware cost, but it complicates software for the two above cases.

kasanovic, 2020-05-03:
commit 9c7732b provides solution to this issue.

closed 2020-05-03 by kasanovic