riscvarchive/riscv-v-spec #377: potential problems with vcsr.fflags/frm scheme

aswaterman, 2020-02-14:
The current `vcsr` scheme says that, if FS=0 (i.e. if FP context is currently lazily swapped out), that `vcsr.fflags` and `vcsr.frm` read as zero.  I see a few issues with this scheme, though all have workarounds:

- User code that accidentally relies on the value of `vcsr.fflags` and `vcsr.frm` will behave unpredictably, since those fields' values depend on whether a context switch has recently occurred.
- For the above reason, there's a side channel that allows a process to determine if it has been context-switched.  (Not sure if this is any worse than the existing FP lazy save/restore side channel.)
- User-level context switch code can't take advantage of `fcsr` fields being packed into `vcsr`; it has to save both.

My guess is that, to avoid the first two issues, OSes will choose to always swap the FP state whenever they swap the vector state.  (The converse is unnecessary; it's OK to swap FP state without swapping vector state.)  The third issue could be avoided if the ABI mandated that the OS behaved that way.

Is that a good enough answer?

kasanovic, 2020-02-21:
Not echoing fflags and ffrm in vcsr would appear to be simple solution, just requiring one more CSR swap on every context swap.

aswaterman, 2020-02-21:
I'm starting to favor that simpler approach.

closed 2020-02-23 by kasanovic

aswaterman, 2020-02-23:
Do you think we still need language explaining that vector instructions that access floating-point state require FS != Off, and vector instructions that write FP state will set FS = Dirty? Or are you thinking that the existing definition of FS in the priv spec makes that sufficiently clear?

commit 951b64fb (2020-02-23): Removed mirrors of fcsr fields from vcsr.
Closes #377.
commit 4695f6ef (2020-02-26): Clarified need to have scalar floating-point unit enabled to execute
vector floating-point code.
Closes #377.