riscvarchive/riscv-v-spec #379: Revisit memory order of whole register load/store.

David-Horner, 2020-02-15:
TL;DR 
In that 

- hardware already supports striped load/store groups
- compiler register spills could benefit from equivalent behaviour of LMUL=nr, 
- the mapping in memory to known component registers LMUL is straightforward, 
- interrupt handlers etc. must consider the whole register load/store as an opaque object
- and future relaxation of nr to 3,5,6,7 are readily accommodated with strided interleaving

I propose that whole register load/store be equivalent to strided load/store of SEW=08, vl = VLEN/8 and LMUL=nr and we allow nr be 2,4 and 8.

Current POR in spec:

>  In a future extension, when multiple registers are
> transferred, the vector register contents are mapped to contiguous
> bytes in memory as if LMUL=1, with the lowest-numbered vector register
> held in the lowest-numbered memory addresses. 

This allows an equivalence of vl<nr>r.v/vs<nr>r.v with nr consecutive vl1r.v/vs1r.v .
This may be useful to OS and library functions as is the intent of the instruction:

> NOTE: These instructions are intended to be used to save and restore
> vector registers when the type and length of the current contents of
> the vector register is not known, or where modifying `vl` and `vtype`
> would be costly. Examples include compiler register spills, vector
> function calls where values are passed in vector registers, interrupt
> handlers, and OS context switches.

However, we do not know now if this equivalence will be of any value.
But we do know that different circuitry will be required to store and load whole register groups 
than is used to store and load the equivalent register groups where SEW=08, vl = VLEN/8 and LMUL=nr.

We noted that "interrupt handlers, and OS context switches" etc. may not have understanding
of the LMUL used to establish the data in these register. I think this lead to an assumption that we should then assume LMUL=1 for each of the component registers.
Further, I think we may have conflated that uncertainty with a perception that striped memory mapping will increase uncertainty about the memory contents of the whole register load/stores.

However, one of the primary use cases - compiler register spills - does know the LMUL of the register   
groups and could benefit from striped load/store that would match. 

To the other use cases "interrupt handlers, and OS context switches" etc it is sufficient that the instruction behave as a black-box with opaque memory encoding as long as store/load exactly recover original contents.

Further, nr values of 3,5,6, and 7 can validly use the same striped mappings and so stripping does not limit future relaxation of nr.

NOTE: If the stripping memory layout mismatches with the component register-group's LMUL, and those LMULs are known then 2 dimension arrays/structures can be used to aggregate the pieces appropriately.

kasanovic, 2020-05-03:
I believe with the new register layout for SLEN<VLEN, this point is now moot.  Successive vs1r/vl1r can save/restore a vector register group.
@David-Horner can you confirm and close this issue if so.

kasanovic, 2020-06-28:
I think this is moot now, closing.

closed 2020-06-28 by kasanovic