riscvarchive/riscv-v-spec #380: vslideup overlap constraints

JamesKenneyImperas, 2020-02-19:
There seems to have been confusion on destination/mask overlap constraints for vslideup and vslide1up instructions. I will describe the history below; can you please confirm what is the correct behavior?

Originally, the constraint was specified as (for example):

`The vslide1up instruction requires that the destination vector register group does not overlap the source vector register group or the mask register`

It was then pointed out that there is no WAR conflict for slide up instructions because destination element i is masked by mask element i. Andrew then modified this constraint on July 29th (commit b853d77) as follows:

`The vslide1up instruction requires that the destination vector register group does not overlap the source vector register group.`

This change then seems to have been reverted by Krste on November 24th (commit 01a1adc), so it is now once again:

`The vslide1up instruction requires that the destination vector register group does not overlap the source vector register group or the mask register.`

Is this an error or intentional? Why was this reversion made if there is in fact no WAR constraint?

chuanhua, 2020-02-26:
The original text before 2019/7/29 is as follows:

> The vslide1up instruction requires that the destination vector register group does not overlap the
> source vector register group and the mask register **if masked**.

The current text is as follows:

> The vslide1up instruction requires that the destination vector register group does not overlap the source vector register group or the mask register.

Is the current requirement too restrictive even when the mask register is not used?

JamesKenneyImperas, 2020-02-26:
I think Andrew's change was correct so there should be no restrictions on mask overlap (apart from the normal LMUL>1 restrictions he mentions in commit b853d77). I do not understand why Krste reverted the text; it would be very helpful to have either Andrew or Krste comment on this.

aswaterman, 2020-02-26:
The commit message I wrote on b853d77d9ca539f669960f35d93eed6da9068c1c still seems right to me.  @kasanovic?

chuanhua, 2020-02-27:
Krste reverted the change on November 24th and closed #214 on November 25th.

JamesKenneyImperas, 2020-02-27:
Two comments here:

1. Issue #214 is ancient history, predating issue #267 which prompted Andrew's change. At the point of #214, people were still laboring under the misapprehension that there was a non-existent WAR conflict with the mask.

2. I see nothing in #214 that requires Krste's change anyway.

To repeat, I think Andrew's change was correct. But until we get feedback from @kasanovic we can't know for sure.

eroom1966, 2020-04-01:
@kasanovic @aswaterman 
Hi Krste, Andrew,
Would it be possible to provide an update/clarification to this issue ?
We are currently blocked until we have a resolution
Thx
Lee

aswaterman, 2020-04-01:
I've made a PR that goes back to what I wrote (overlapping the mask is OK if LMUL=1).

kasanovic, 2020-04-03:
Was editing error.  Reverting.

commit 441cd001 (2020-04-01): slide[1]up can overlap the mask if LMUL=1

Resolves #380

I added this to the spec in b853d77d9ca539f669960f35d93eed6da9068c1c
but @kasanovic reverted it in 01a1adc5274238f103ee1f622d21e735762f916b
without explanation, which I think might have been an accident.

The rationale is that, since _output_ element i is masked by mask
element i, there is no intra-instruction WAR hazard.
PR #407 (2020-04-03): slide[1]up can overlap the mask if LMUL=1
Resolves #380

I added this to the spec in b853d77d9ca539f669960f35d93eed6da9068c1c but @kasanovic reverted it in 01a1adc5274238f103ee1f622d21e735762f916b without explanation, which I think might have been an accident.

The rationale is that, since _output_ element i is masked by mask element i, there is no intra-instruction WAR hazard.
closed 2020-04-03 by kasanovic