riscvarchive/riscv-v-spec #381: Supporting XLEN-bit Scatter/gather/AMO indices if sub-SEW memory accesses are removed

aswaterman, 2020-02-19:
See #306 for context on supporting indexed memory accesses with different index and data widths.

Under the no-sub-SEW-memory-access regime proposed #362, we'll need a new way to decouple the index width from the data width for indexed memory accesses.  In particular, we want to support XLEN-bit indices (e.g., pointers), regardless of data width.

We can continue supporting indices that are SEW bits wide, to optimize VRF usage when the indices happen to be the same size as the data.  It's not clear to me how important this really is, but it's not too hard to provide.

My proposal is:

When funct3=7, indexed memory accesses use SEW for both the data and index width.  (This is the status quo.)

When funct3=6, indexed memory accesses use SEW for the data width, but XLEN for the index width.  (This opcode is where the explicitly sized 32-bit accesses currently live.)  Assuming ELEN=XLEN, these instructions access the index vector as though LMUL were increased by a factor of (XLEN/SEW), so they have the constraint that (XLEN/SEW)*LMUL <= 8.  If ELEN > XLEN, then the data might be wider than the index, and the constraint (SEW/XLEN)*LMUL <= 8 also applies.

David-Horner, 2020-02-20:
>  (If ELEN > XLEN, then the data might be wider than the index, and so the constraint (SEW/XLEN)*LMUL <= 8 also applies.)

I don't see how the constraint applies.

What is significant is vl and that is determined on the underlying vtype for the data.
If XLEN is less than ELEN then the index register group multiplier is still (XLEN/SEW), constrained to max(smallestLMUL,  (XLEN/SEW)*LMU). It could be fractional if that proposal is accepted, but it is sufficient that smallestLMUL=1 as vl will work even when MAXVL is much greater.
 
> Vector AMO instructions are only supported for the memory element widths supported by AMOs in the implementation’s scalar architecture. Other widths raise an illegal instruction exception.

So are you proposing extending AMO*E to other SEW values than the currently support W and D?

You revised this issue to modify only VAMO*E instructions.
I believe there may still be merit in considering the same addressing for vlxe.v, vsxe.v and vsuxe.v.
Isn't the infrastructure in place for these when this version of VAMO*E is implemented? 

Although you propose index be XLEN so that it can be a full address value, are you still recommending that rs1 be the base address? I expect rs1=x0 will be an optimized case that does not need to interact with the scalar integer register set..

aswaterman, 2020-02-20:
I was thinking about AMOs when I wrote this, so accidentally focused the issue there.  What I wrote was meant to apply to indexed loads and stores, too.

I do not propose to extend AMOs to support other widths; they should continue to raise an exception unless 32 <= SEW <= XLEN.

You may be right about the LMUL constraint; I've edited my post to drop that subject for the time being.

knightsifive, 2020-04-03:
A followup to @aswaterman's comment from earlier in the thread:

>  By far the most important case to support is XLEN-bit indices (so they can hold general pointers) and SEW-bit data. While this isn't especially efficient for narrow data, at least it is complete.

I just want to offer a non-synthetic example of this inefficiency. Machine learning applications frequently evaluate componentwise operations (e.g., [softmax](https://en.wikipedia.org/wiki/Softmax_function)) via "narrow" lookups, e.g., gathering from a `int8_t[256]` with 8-bit indices. When XLEN=64 this'll need three widening adds.

aswaterman, 2020-04-04:
Yep.  Note that for VLEN >= 256, it might be better to use `vrgather` for that case anyway (even though that increases register pressure further).  It's not ideal to depend on VLEN, but you can branch and then move the hopefully uncommon case (reverting to `vlxe`) out-of-line.

knightsifive, 2020-04-04:
@aswaterman : I think you mean VLEN >= 256*8 = 2048? But yes, I really do hope for machines where this is a common case!

kasanovic, 2020-06-26:
Resolved in v0.9.

closed 2020-06-26 by kasanovic