riscvarchive/riscv-v-spec #382: the differing nature of LMUL > 1 and fractional LMUL

David-Horner, 2020-02-20:
LMUL > 1 has significant beneficial characteristics. 
    Each register group is directly addressable
    Stripping is applied to each register in the register group.
    VLEN bits of each component register is usable from within the register group. 

A downside of LMUL > 1 is less than 32 register groups are mapped - specifically 1/2 for each increase in LMUL.

Fractional LMUL 
    Each register group is directly addressable
    Stripping is NOT proposed to apply to each register in the sub-register.  (perhaps it can be).
    Each sub-register uses only a portion of the underlying register. This could lead to waste. 

The up side is 32 fractional register are addressable from the 32bit encoding.

David-Horner, 2020-02-20:
**TL;DR**

> Novel mapping of fractional LMUL registers can increase their usability.  
> The mapping proposed here is but one of many configurations that might be useful.
> 

**Mitigation: Downside of LMUL > 1 is less than 32 register groups mapped**

Suggestions for the 64 bit V model include increasing the number of LMUL=1  registers and the corresponding increase of addressing to 64 (or even 128).
The 32bit encoding could then address 32 LMUL>2 register groups. The reserved odd register group numbers would then map into the extended registers. Some functionality on these registers could not be supported by the current 32bit instruction encoding (namely Whole Register Moves and instructions with LMUL=1). 

**Analogous Fractional mapping mitigation.** 
 (I use fl2 to represent LMUL of 1/2, fl4 LMUL of 1/4 , etc.)
> Hypothetically:
> Consider 64bit V encoding with 32 LMUL=1 registers supporting fractional LMUL. It could increase addressing to support all registers in the fractional groups. It could address all 64 fl2 registers with a 6 bit register field, all 128 fl4 with a 7bit field, etc.
> The 32bit  compressed instruction mapping could map to any 5bit subset of those 64bit 6 bit and 7 bit register fields.

Currently we have considered the fractional registers mapped so that scalar registers are the first element of each. As with the C-extension the subset mapping could go anywhere it is most useful.
Specifically, avoiding V0 in the mapping is likely useful as its mask purpose can be used seperately to advantage.

**I propose all 32 lf2 registers be mapped over v16-v31.**

And there is no reason that lf4, lf8 etc have to map over the same registers.
Specifically, **I propose lf4 map over v8-v15, lf8 over V4-7 and lf16 (if we define it) map over V2 and V3.**
This will allow widening /narrowing operations to fully utilize the VLEN of the underlying registers. And leverage those instructions with less disruption to the LMUL=1 register files (and by extension LMUL>1).
This should reduce register pressure overall.

I apologize that this is last minute  before our meeting.

David-Horner, 2020-02-21:
I refer to the fractional LMUL registers as sub-registers analogously to register groups for LMUL>1.
I designate fractional LMUL as lf2 (for 1/2), lf4 (for 1/4) etc. 

**Further Recommendations under fractional LMUL:**

As with LMUL>=1, **Scalars are the first element of underlying vector register file and are always accessible in instructions that use them.**
.  	As opposed to scalars as first element of sub-registers, this increases the directly usable register space while in fractional LMUL.

**For mask operations implicit and explicit references to V0 map to underlying register V0.**
This recognizes the special nature and mapping of v0 in the 32bit encoding.
Note: This removes a hazard for many operations.

**Further  that V0 has an effective length matching the current fractional LMUL.**
This may appear to be at odds with LMUL>1 behaviour, but is similar to the special treatment that bit allocation within elements is used to accommodate the larger register groups.

Other Characteristics:

**Similarities to LMUL>=1**

Whole register operational are unaffected by fractional LMUL.

Widening and narrowing instructions continue to work exactly as described in POR.
E.g. when LMUL = lf2 widening operations specify the destination as any of the base registers, including those to which  the lf2 registers are mapped. 

Note: when LMUL = lf4 or smaller the destination widened register does not overlap any of the unwidened source registers nor the mask register. As a result they are not restricted as other operations are.

David-Horner, 2020-02-22:
**TL;DR**

> **Propose interleaving as resolution to lane alignment.**
> Note the limited benefit of the structure to common element manipulation operations.
> **Propose revised mapping to create two distinct groups of sub-registers to facilitate active LMUL=8 register groups when fractional sub-registers are also active.**

I refer to the fractional LMUL registers as sub-registers analogously to register groups for LMUL>1.
I designate fractional LMUL as lf2 (for 1/2), lf4 (for 1/4) etc. 
    The general for lfn for 1/n (and thus VLEN can accommodate n sub-registers)

**Proposal:**
**Interleaving is the complimentary structure to striping.**

Byte interleaving of the set of sub-registers would align the elements to the SLEN lanes.
But that would require reassembling and realigning bytes to match with the lane processing unit.

**As this appears optimal I recommend:** 

**For lfn:** 

- as already proposed, establish a register range per lfn setting
- allocate n consecutively numbered sub-registers per vector register.
- interleaving the n sub-registers jn order within the vector register.
- Interleave chunks are the LMUL fraction of SLEN.

This structure allows 

1.   the wiring for each lfn set of registers to be specific to that lfn settings
 		each vector register then has only two structures, 
			SLEN striping (trivial when LMUL=1) and
			SLEN/n interleave.
2. independence of each vector register as sub-register do not crosses over to other vector register.                       This greatly enhances the ability to have active  LMUL>=1 register groups within the  range of registers allocate for specific fractional LMUL.
3. in-order interleaving (of the chunks within the sub-register, and the sub-registers within the vector register) is not only natural and intuitive but more importantly aligns elements across LMUL values.

Note:  When SEW matches SLEN/n the n sub-register have their elements interleaved within a single LMUL=1 register. This can be a very valuable software visible feature. Not generally portable but inherent in the structure,  

**Revised Proposal:**

**I previously proposed mapping**  

>   all 32 lf2 registers be mapped over v16-v31.
>     lf4 map over v8-v15, lf8 over V4-7 and lf16 (if we define it) map over V2 and V3. 

My thinking was that the register sets would provide enough flexibility to allocate the sub-registers within the groups optimally, trading off complexity with efficiency.
If the interleave structure proposed above is optimal across all ranges of VLEN then my guess was correct.
However, I could be wrong and lf16 especially, mapped over only 2 registers, may not allow adequate flexibility.

Further, that mapping, although it creates 4 distinct "blobs" prohibits (in the general case) allowing at least one of each LMUL sub-register/register group to be active at the same time.

**I now propose an alternate mapping. Sub-registers 0 through 15 overlay LMUL=1 registers 1 through 15, and sub-registers 16-31 overlay LMUL=1 registers 17 through 31.**

This creates two blobs that are isolated and as each has 16 sub-registers of each fraction LMUL in many cases the register pressure would not be sufficient to require both to be active concurrently.

E.g. Only sub-registers 0 through 15 in each of the fractional LMUL are active. Then registers 16-31 are available for LMUL>=1. Thus any combination of non-overlapping register groups
LMUL=8  registers 2 and 3
LMUL=4 registers 4 through 7
LMUL=2 registers 8 through 15 and 
LMUL=1 registers 16 through 31 are available.

If the interleave structure proposed above is accepted then there are more opportunities to restrict  ranges of  active sub-registers to allow LMUL>1 register groups. 

**I this case I further propose that within each blob lf2 registers are allocated to 8 highest numbered registers, lf4 to next 4 highest, lf8 to next 2 highest and lf16 the lowest register in the blob.**
Specifically sub-register are allocated consecutively 
> fl2 sub-register 0 to   15 to registers 8  through  15
> fl2 sub-register 16 to 31 to registers 24 through 31 
> fl4 sub-register 0 to   15 to registers 4   through 7
> fl4 sub-register 16 to 31 to registers 20 through 23 
> fl8 sub-register 0 to   15 to registers 2   and 3
> fl8 sub-register 16 to 31 to registers 18 and 19 
> fl16 sub-register 0 to   15 to registers 1
> fl16 sub-register 16 to 31 to registers 17 

This proposed mapping increases the availability of active LMUL=8 register groups with various combinations of active fractional sub-registers even with optimal register number allocation using the previous mapping.

hanna-kruppe, 2020-02-23:
I'll restate some things in my own words and give some concrete examples to check my understanding, please let me know if I got something wrong. I will restrict myself to the basic 32 bit instruction encoding, leaving aside the parts relating to a potential 64bit encoding.

The semantics of vector register accesses with LMUL = 1, 2, 4, 8 are not affected at all by your proposal. With LMUL >= 1, a reference to vector register number *n*  in an instruction refers to a register group starting at the *n*-th architectural vector register, but with fractional LMUL under your proposal, the register numbers map to rather different parts of the vector register file (except for v0 used as mask, more on that below). For example, the instruction `vmv.v.x v2, t0` under LMUL = 1/2 writes to (part of) the same register as `vmv.v.x v9, t0` does under LMUL = 1.

To reduce ambiguity, I will refer to the 32 architectural registers of VLEN bits each as `VRF[i]` and only use `v0`...`v31` for the register numbers used in instructions. So the previous two examples both write to `VRF[9]`.

Let's consider this instruction under LMUL = 1/4:

    vwadd.vv, v0, v1, v2

Both source operands are read from (part of) `VRF[4]` and the result is written to (part of) `VRF[8]`.  The four LMUL=1/4 vectors (v0...v3) mapped into `VRF[4]` are interleaved in chunks of SLEN/4 bits each, while the two LMUL=1/2 vectors (v0 and v1) mapped into `VRF[8]` are interleaved at a granularity of SLEN/2 bits. Thus, under SLEN = 32, we interleave in 16 bit chunks at LMUL = 1/2, while at LMUL = 1/4 we'd interleave bytes. It won't show up in these examples but note that unless SLEN >= 128, smaller LMUL eventually lead to sub-byte chunks being interleaved.

Here's how the registers accessed by the above instruction are organized internally for VLEN = 128 and all the possible values of SLEN. In the tables below, each (register, byte) cell contains the (sub)register number that this byte of this register is associated with:

```
SLEN = 32:
 Byte     F E D C B A 9 8 7 6 5 4 3 2 1 0
 VRF[4]   3 2 1 0 3 2 1 0 3 2 1 0 3 2 1 0    (LMUL = 1/4, SLEN/4 = 8b)
 VRF[8]   1 1 0 0 1 1 0 0 1 1 0 0 1 1 0 0    (LMUL = 1/2, SLEN/2 = 16b)

SLEN = 64:
 Byte     F E D C B A 9 8 7 6 5 4 3 2 1 0
 VRF[4]   3 3 2 2 1 1 0 0 3 3 2 2 1 1 0 0    (LMUL = 1/4, SLEN/4 = 16b)
 VRF[8]   1 1 1 1 0 0 0 0 1 1 1 1 0 0 0 0    (LMUL = 1/2, SLEN/2 = 32b)

SLEN = 128:
 Byte     F E D C B A 9 8 7 6 5 4 3 2 1 0
 VRF[4]   3 3 3 3 2 2 2 2 1 1 1 1 0 0 0 0    (LMUL = 1/4, SLEN/4 = 32b)
 VRF[8]   1 1 1 1 1 1 1 1 0 0 0 0 0 0 0 0    (LMUL = 1/2, SLEN/2 = 64b) 
```

Continuing with the above example of `vwadd.vv, v0, v1, v2` under LMUL = 1/4, let's look at how each SEW=16 element is actually formed from the bytes of `VRF[4]` and how the (32b) results are scattered into `VRF[8]` in the SLEN = 32 case:

```
 Byte     F E D C B A 9 8 7 6 5 4 3 2 1 0
 VRF[4]   _ d b _ _ d b _ _ c a _ _ c a _
 VRF[8]   _ _ y y _ _ y y _ _ x x _ _ x x

    lf4 v1: [     b    |     a     ]
                  +          +
    lf4 v2: [     d    |     c     ]
    ==>           =          =
    lf2 v0: [     y    |     x     ]
```

As for masks, the register number v0 still refers to `VRF[0]`. You are *not* proposing this exception for any other register numbers, so e.g. `vmand.mm v0, v2, v3` under LMUL = 1/2 would read both operands from `VRF[9]` but write the result to `VRF[0]` instead of `VRF[8]` (to which v0 usually refers at LMUL = 1/2). As a corollary, mask operations (even `vmcpy.m`) would need to perform some packing/unpacking in those cases (instead of just performing bitwise operations on VLEN bits as they do for LMUL >= 1).

I don't see any explicit description of how multiple masks are packed into one VLEN-sized register, nor how masks with fewer entries than possible at LMUL >= 1 are stored in `VRF[0]`. I see an obvious way to extrapolate the current layout of mask registers: take the VLEN bits of `VRF[0]` or the VLEN/n bits of a subregister, divide them into as many equally-sized chunks as the corresponding vector has elements, and store the mask bits in the LSBs of those chunks. For masks packed into subregisters, use the same SLEN/n chunking as for vectors. That would mean a mask for the previous example (LMUL = 1/2, SEW = 32, VLEN = 128, SLEN = 32) would store the mask bits in:

- LSB of byte 0 and byte 8 for the "first" subregister mapped into a whole register
- LSB of byte 2 and byte 10 for the "second" subregister mapped into a whole register

which lines up with the LSB of the destination elements they apply to. More generally (for all fractional LMUL, all SEW, and regardless of VLEN and SLEN), each mask element in a subregister takes up SEW bits and lines up with the SEW-sized vector element it controls. This is also true for LMUL = 1 masks today. For LMUL = 1/n masks that occupy the whole v0 / `VRF[0]`, the mask elements are larger (n*SEW bits) such that that they still line up (as much as possible) with vector elements as they are stored in LMUL = 1/n subregisters.

---

Assuming all of the above is correct, some concerns and suggestions come to mind, but let's see if I misunderstood anything first.

David-Horner, 2020-02-23:
Thank you for your clear and illustrated presentation.

You are correct with the analysis up to the mask discussion. You lost me at
> As a corollary, mask operations (even vmcpy.m) would need to perform some packing/unpacking in those cases (instead of just performing bitwise operations on VLEN bits as they do for LMUL >= 1).

Are the cases you reference those that have mixed mask v0 and mask non-v0?
If so I believe it depends on the mapping in v0 which you rightly pointed out I have not defined yet.

> I don't see any explicit description of how multiple masks are packed into one VLEN-sized register, nor how masks with fewer entries than possible at LMUL >= 1 are stored in VRF[0]. 

I envisioned that non-v0 masks would be store as v1 through 31 sub-registers are stored with the same SEW and fractional LMUL
Also, I anticipated that only one mask would be stored in VRF[0]. It would be spred out as if interleaved. In other words the allocation of v0 for non-mask would exactly match the mask in VRF[0]. The zero entries of your diagram at bytes 0 and 8:
``
```
SLEN = 32:
 Byte     F E D C B A 9 8 7 6 5 4 3 2 1 0
 VRF[4]   3 2 1 0 3 2 1 0 3 2 1 0 3 2 1 0    (LMUL = 1/4, SLEN/4 = 8b)
 VRF[8]   1 1 0 0 1 1 0 0 1 1 0 0 1 1 0 0    (LMUL = 1/2, SLEN/2 = 16b)

```
> I see an obvious way to extrapolate the current layout of mask registers: take the VLEN bits of VRF[0] or the VLEN/n bits of a subregister, divide them into as many equally-sized chunks as the [...]
Yes, I believe that agrees with what I envisioned.

I agree with you that with SLEN <= 64 the interleaving chunks are sub-byte for LMUL 1/16.
It presents an architectural anomaly, with nibbles (and smaller) not bytes being smallest observable objects. 

I greatly anticipate hearing your concerns and suggestions.
Thank you again for your replies.

hanna-kruppe, 2020-02-26:
> Are the cases you reference those that have mixed mask v0 and mask non-v0?
> If so I believe it depends on the mapping in v0 which you rightly pointed out I have not defined yet.

Yes, that's the cases I refer to. It seems the mask register layouts I assumed match what you envisioned, so I believe what I had in mind when writing this is correct, but I expressed it rather poorly, so let me rephrase. When LMUL < 1, the number of bits used for each mask element (MLEN) is different both (1) between v0 and v1-v31, and (2) between v1-v31 at different LMUL. Consequently:

- mask operation that mix v0 with masks from v1-v31 are not reading and writing the same amount of bits in all mask registers involved
- Going through v0 is the (only) way to transfer a mask between subregisters of different fractional LMUL. In effect, moving a mask to v0 unpacks the "LMUL < 1 mask format" into the regular mask format used at LMUL > 1, and moving back packs them again (possibly for a different LMUL).

I'm not sure if this is an important aspect, it just seemed worth noting to me.

---

With that out of the way, my concerns and suggestions can be summarized as: your proposal supports packing larger elements into small fractions of registers than necessary for the intended use cases of fractional LMUL, and this has several undesirable consequences. We could avoid those problems by imposing some natural restrictions on how fractional LMUL is used. Intuitively, code should use LMUL >= 1 for the largest element size needed, while fractional LMUL should only be used for proportionally smaller element widths. For example, placing double-precision floats in an LMUL = 1 vector and using LMUL = 1/2 for single precision.

Some examples of undesirable behaviors that emerge otherwise (throughout, LMUL = 1/n):

- This is a general conceptual issue, not specific to your proposal, but: if VLEN / n < SEW then the concept of a fractional vector is nonsensical -- you can't have a vector of 0.5 elements. Thus, portable software (which assumes only VLEN % ELEN == 0) must avoid adhere to the aforementioned restriction anyway.
- Currently, each vector element is always contained within an SLEN-sized window (assuming SLEN <= ELEN, but even in the rare implementation where SLEN > ELEN you at least store each element contiguously). But with the SLEN/n interleaving scheme this is not guaranteed if both n and SEW are large and SLEN is small. For example, with SLEN = ELEN = 32, an LMUL = 1/8 vector of 32b elements breaks up each element into 8 chunks of 4b each and spreads them across 8*SLEN bits.
- Interleaving logically different registers at nibble (or even 2b, at LMUL = 1/16 and SLEN = 32) granularity seems problematic for some implementations. For example, without equally fine-grained write-enable signals (which might not even be available, e.g. in an FPGA implementation storing the VRF in BRAM) it seems to me that writes to registers interleaved at such a fine granularity need to be implemented as read-modify-write of the surrounding byte (or larger granularity) to preserve the contents of adjacent bits belonging to other subregisters. Some mask operations have a similar issue today (if MLEN < 8 and `vstart`/`vl` don't fall on a byte boundary), but I believe it would be a novel and more severe problem for vectors (e.g., affect far more regfile writes and affect every element written, not just the first and last byte written).
- Mask registers v1-v31 being completely segregated by LMUL leads to more LMUL changes, more mask copies, and a need for new instructions to reduce those problems. Today, if some code juggles multiple masks, it can just freely operate on them all and move each mask into v0 as necessary, independently of `vtype` (assuming constant VLMAX, of course). But under your proposal, mixed-width code may need to change LMUL back and forth and funnel masks through v0 just because it can't simultaneously access masks created at all different element widths. This could be mitigated with new instruction(s), but having "LMUL-crossing" variants of all operations on masks would be a lot of encoding space, so we'd likely only add a few (at least `vmcpy.m`) and still have extra mask copies for other operations.

---

I therefore suggest two (three) modifications:

1. Require that (legal) `vtype` settings have SEW <= ELEN / n when LMUL = 1/n.
2. Treat mask registers as you proposed to treat scalars operands: always taken from / placed in whole registers, never packed into fractional subregisters.
3. (Possibly, see discussion below: interleave subregisters in ELEN/n-bit chunks, rather than SLEN/n.)

The first point implies that (at least for implementations where SLEN >= ELEN), elements are never dispersed over a distance greater than SLEN and subregisters are not interleaved at sub-byte granularity (indeed, they're even interleaved at element granularity or greater).

Implementations where SLEN > ELEN still have the aforementioned problems (to a lesser extent, but still). It seems natural to define interleaving in terms of ELEN/n instead of SLEN/n, and I believe it solves those problems as well. However, I haven't thought about this in much detail, so I'm not 100% sure about this.

David-Horner, 2020-02-27:
Thank you once again for your thoughtful considerations.

> When LMUL < 1, the number of bits used for each mask element (MLEN) is different both (1) between v0 and v1-v31, and (2) between v1-v31 at different LMUL. Consequently:
>
> mask operation that mix v0 [...]

Although that would do, an appropriate widening/narrowing instruction would also work, just as it is required to move integer values. By definition the mask bit is the LSB of each mask element of the same SEW. So the standard integer transfers will always cover the mask use.

> With that out of the way, my concerns and suggestions can be summarized as: your proposal supports packing larger elements into small fractions of registers than necessary for the intended use cases of fractional LMUL, and this has several undesirable consequences. 

Other than anticipating that fractional LMUL will be useful in replacing fixed width load/stores, I haven’t yet specified use cases for fractional LMUL. I anticipate it will be useful in more situations than I envision. Are there specific use cases you have in mind? They may be useful in validating or refining the design.
> We could avoid those problems by imposing some natural restrictions on how fractional LMUL is used. Intuitively, code should use LMUL >= 1 for the largest element size needed, while fractional LMUL [...]

Agreed in the sense that if we impose such restrictions they would not be arduous in general practice.

> Some examples of undesirable behaviors that emerge otherwise (throughout, LMUL = 1/n):
>
> This is a general conceptual issue, not specific to your proposal, but: if VLEN / n < SEW then the concept of [...]

Agreed. This “restriction” is a natural consequence.
 
> Currently, each vector element is always contained within an SLEN-sized window (assuming SLEN <= ELEN, but even in the rare implementation where SLEN > ELEN you at least store each element [...]

Agreed. Fractional byte allocations are undesirable. Avoiding the spread would be valuable and restricting implementations to SEW <= SLEN / n might be appropriate. You later propose SEW <= ELEN /n which may be unnecessarily harsh. 
> Interleaving logically different registers at nibble (or even 2b, at LMUL = 1/16 and SLEN = 32) granularity seems problematic for some implementations. For example, without equally fine-grained [...]

Agreed. Fractional byte allocations are problematic. We need to either prohibit or avoid.

>Mask registers v1-v31 being completely segregated by LMUL leads to more LMUL changes, more mask copies, and a need for new instructions to reduce those problems. 

I am not convinced that creative software solutions cannot mitigate these concerns. 

>Today, if some code juggles multiple masks, it can just freely operate on them all and move each mask into v0 as necessary, independently of vtype (assuming constant VLMAX, of course). 
But under your proposal, mixed-width code **may need to change LMUL back and forth and funnel masks through v0** just because it can't simultaneously access masks created at all different element widths.

As mentioned above widening and narrowing operations can avoid shuffling through v0 and the related LMUL changes.

The Plan of Record,  widen to LMUL * n and narrow from LMUL * n may be biased to positive LMUL. There are many permutations to consider. Perhaps to adequately support smaller fixed load/stores the alternate approach is required.

The isolation of fractional LMUL sets is similarly “problematic” for non-mask element types.

 >This could be mitigated with new instruction(s), but having "LMUL-crossing" variants of all operations on masks would be a lot of encoding space, so we'd likely only add a few (at least vmcpy.m) and still have extra mask copies for other operations.

Additional variants of instructions may directly enhance fractional LMUL register set use.
However, other approaches, such as partial overlap of fractional LMUL sets may provide sufficient "LMUL-crossing" access.

> I therefore suggest two (three) modifications:
> 
> Require that (legal) vtype settings have SEW <= ELEN / n when LMUL = 1/n.

As I noted above, I believe SEW <= SLEN / n is sufficient. 
It does leave the problem of fractional byte allocations. 
Various approaches are possible.

I lean towards this byte allocation/mapping. 
For each set of registers that would otherwise be allocated a fractional byte, one of those registers is allocated the full byte. The simplest mapping is the byte is allocated round-robin to register-set starting with the lowest numbered register. 
Considering  your example above:
>With SLEN = ELEN = 32, an LMUL = 1/8 vector of 32b elements breaks up each element into 8 chunks of 4b each and spreads them across 8*SLEN bits.

And from your previous post
>To reduce ambiguity, I will refer to the 32 architectural registers of VLEN bits each as VRF[i]

At VRF bit 0   v0 is allocated its byte 0 as its byte 0
At VRF bit 8   v2 is allocated its byte 0
At VRF bit 16 v4 is allocated its byte 0
At VRF bit 24 v6 is allocated its byte 0

At VRF bit 32 v1 is allocated its byte 0
At VRF bit 40 v3 is allocated its byte 0
At VRF bit 48 v5 is allocated its byte 0
[... 7 lines cut ...]
I can see this work. 
I am reluctant to give up the benefits of reduced register pressure that the packing affords.
But your approach is consistent with mask operations working on LMUL=1 objects even when actual LMUL>1.
I’m willing to adopt this as the new working model.
[... cut ...]
I believe you meant ELEN > SLEN? 

I agree that ELEN/n is a more natural metric. It is not implementation dependent. However, LMUL>1 uses SLEN for its striping for [purely] performance reasons. Having to consider SLEN is already a given. And yes, I will think through this idea as well. I hadn’t considered it previously.

One less important but related consideration:
As the term “sub-registers” are often used in the context of register renaming, perhaps frac-regs , short for fractional registers would be a good handle?

David-Horner, 2020-02-27:
I promised examples of how this fraction LMUL proposal aids in the removal of fixed length load/stores.

I have only one ready at present.

The example of a minimal implementation with 32 LMUL=1 registers of VLEN=32 and therefore  SLEN=32. ELEN is also 32.

As proposed by hanna kruppe
>To reduce ambiguity, I will refer to the 32 architectural registers of VLEN bits each as VRF[i]

Case 1: load 4 bytes from memory and sign extend into 4 elements of LMUL=4 register set v12, VRF[12:15]
>Note: Although this use case is specific to minimal configuration the principles are applicable to LVEN > 32. However, in that case other approaches are also possible.

>Note: Widening operations restrict source and destination from overlapping.
>V12 is chosen to reduce that impact. 
>However, any of v16, v20, v24, v28 would also work with the same choice of fraction registers.
>And with fractional registers chosen from > 16, v0, v4 and v8 are supported. 

    
Example using POR operations plus fractional LMUL.

```
# a0 is memory address of bytes to load. 

vlseti t0,x0,e8,lf4  # 01 fractional LMUL of 1/4,  vl4with ed 
vlseg4 v0,a0         # 02 load lf4 registers v0 through v3, VRF[4] 
vwadd.vx  v0,v0,x0   # 03 into VRF[8]
vwadd.vx  v1,v1,x0   # 04
vwadd.vx  v2,v2,x0   # 05 into VRF[9]
vwadd.vx  v3,v3,x0   # 06
vlseti x0,x0,e16,lf2 # 07
vwadd.vx  v12,v0,x0  # 08 into VRF[12] 
vwadd.vx  v13,v1,x0  # 09
vwadd.vx  v14,v2,x0  # 10
vwadd.vx  v15,v3,x0  # 11
vlseti x0,x0,e32,l4  # 12 done. But only works with larger VLEN if SLEN=ELEN
```

Three VRF register are temporarily used in addition to the result LMUL=4 register set.
A total of 12 instructions including three vlseti instructions are used.
The greatest impact appears to be the 8 widening operations.

If quad widening add immediate were available:

``` 
vlseti t0,x0,e8,lf4  # 01 fractional LMUL of 1/4,  vl4with ed 
vlseg4 v0,a0         # 02 load lf4 registers v0 through v3, VRF[4] 
vw4add.vx v12,v0,x0  # 03 into VRF[12] 
vw4add.vx v13,v1,x0  # 04 into VRF[13]
vw4add.vx v14,v2,x0  # 05 into VRF[14]
vw4add.vx v15,v3,x0  # 06 into VRF[15], overwriting original 4 byte load.
vlseti x0,x0,e32,l4  # 07 done. But only works with larger VLEN if SLEN=ELEN
```
One VRF register is temporarily used in addition to the result LMUL=4 register set.
A total of 7 instructions including two vlseti instructions are used.
Four widening operations are still required, one for each fractional register.

Compared to POV without fractional LMUL:

Example using POR operations 

```
# a0 is memory address of bytes to load. 

vlseti t0,x0,e32    # 01 set vl for target
vlseti t0,t0,e8     # 02 use that vl
vlseg4 v12,a0       # 03 load registers v12 through v15, VRF[12:15] 
vwadd.vx v8,v12,x0  # 04 into VRF[8]
vwadd.vx v9,v13,x0  # 05 into VRF[9]
vwadd.vx v10,v14,x0 # 06 into VRF[10]
vwadd.vx v11,v15,x0 # 07 into VRF[11]
vlseti x0,x0,e16    # 08
vwadd.vx v12,v8,x0  # 09 into VRF[4] overwriting original 
vwadd.vx v13,v9,x0  # 10
vwadd.vx v14,v10,x0 # 11
vwadd.vx v15,v11,x0 # 12
vlseti x0,x0,e32,l4 # 13 done. But only works with larger VLEN if SLEN=ELEN
```
Only four VRF registers are temporarily used in addition to the result LMUL=4 register.

A total of 13 instructions including four vlseti instructions are used.
The greatest impact appears to be the 8 widening operations.

If quad widening add immediate were available:

``` 
# a0 is memory address of bytes to load. 

vlseti t0,x0,e32    # 01 set vl for target
vlseti t0,t0,e8     # 02 use that vl
vlseg4 v8,a0        # 03 load 4 register VRF[8:11] 
vw4add.vi v12,v8,0  # 04 into VRF[7] 
vw4add.vi v13,v9,0  # 05 into VRF[6]
vw4add.vi v14,v10,0 # 06 into VRF[5]
vw4add.vi v15,v11,0 # 07 into VRF[4], overwriting original 4 byte load.
vlseti x0,x0,e32,l4 # 08 done. But only works with larger VLEN if SLEN=ELEN
```
Four VRF registers are still temporarily used in addition to the result LMUL=4 register set.
A total of 8 instructions including two vlseti instructions are used.
Four widening operations are still required, one for each register.

> Note: If quad widening did not have the overlap restriction the code without fractional LMUL would not require any temporary registers.
> 
> However,  the fractional LMUL approach would still require one VRF except in two special cases.

solomatnikov, 2020-02-29:
> I promised examples of how this fraction LMUL proposal aids in the removal of fixed length load/stores.
>
> I have only one ready at present.
>
> The example of a minimal implementation with 32 LMUL=1 [...]

These code examples can be misleading to SW developers because of a couple of issues:

1. While segment loads and stores are very useful for array-of-structs data like complex numbers or RGB colors, they should not be used when simple unit-stride loads and stores can be used. The reason is that segment load writes several register, so in a typical implementation its rate of execution is a fraction of simple unit-stride load and arithmetic instruction, e.g. vlseg4 execution rate is 1/4. This means arithmetic instructions can not be effectively chained behind segment load due to execution rate mismatch. Also, chaining and dependency logic for segment loads is a lot more complicated, so an implementation might not even support chaining for segment loads. Effective usage of segment loads and stores requires SW pipelining and careful instruction scheduling.

1. Widening arithmetic instructions should be used to actual useful work, not just widening. For example, addition of 2 vectors with widening to prevent overflow.

David-Horner, 2020-03-01:
@solomatnikov I will be pedantic in the explanation below. I expect others less knowledgeable than yourself may be reading this. 
Thank you for your comments.

I definitely agree with an aspect of your first mentioned concern. Functional mismatch between SW use, architectural and implementation complexity. I will respond to that below. But I will deal with the second concern first.

> 2. Widening arithmetic instructions should be used to actual useful work, not just widening. For example, addition of 2 vectors with widening to prevent overflow.

The RISCV philosophy avoids multiple instructions that perform the same operation. 
RISCV explicitly architects special cases and leverages architectural features, such as x0 as read only zero, to do so.

This provides a reduction in opcode use. Integer Add Immediate but no Sub Immediate.
But also important is the reduction in alternate ways to perform the same operation. When more than one exists, software use of both may causes the micro-architecture to chose from conflicting optimizations. 
e.g. The unprivileged doc stipulates to use jump for unconditional branch rather that any condition branch that has the condition always met.

Thus, RISCV does not have a specific instruction to move the value from one register to another. Instead it uses add immediate zero to provide that function. On most implementations the instruction decoder will detect the intent and perform just the move and not use the ALU.   
Similarly, the current RVV spec recommends this use of vwadd.vx with x0 to perform the widening move. Use of x0 allows the instruction decoder to avoid the add. 

Now some background for the example provided.
As mentioned, the objective was to provided code samples  that replace the fixed size load/stores.
There was also a request to provide such examples for small implementations. It is certainly my concern that small (and minimal) implementations are not dis-proportionally penalized by that removal.

As a result I chose the minimal RVV implementation and although I did not name it, the lvlb.v when LMUL=4.
And I leveraged the characteristics of the minimal RVV.

There are many complaints that can be made about this example.:
- It is not representative of the general case in which VLEN > 1.
-  I did not use the optimal LMUL>=1 algorithm.
- The code is wrong for the non-fraction LMUL case. The vwadd targets odd register pairs and appears to assume the target register is not a register pair.
- I didn’t spell vlseg4e.v opcode correctly.
- It is not well explained, and the analysis is flawed.

I’m sure there are more, and I look forward to hearing about them.

But I did try not to bias the examples in favour of fractional LMUL (to the extent of miscoding!).
That is why I did not start with a vle.v  with LMUL=1 and SEW=32. That would legitimately leverage the underlying structure of the fractional LMUL mapping but it would special case it to word aligned loads. And I did not use vle.v with LMUL=1 and SEW=8 which could be seen to be a fortuitous coincidence with the mapping, but used the same vlseg4e.v that non-fractional would optimally use.

Now to your concern about misleading software developers:

> While segment loads and stores are very useful for array-of-structs data like complex numbers or RGB colors, they should not be used when simple unit-stride loads and stores can be used.

In the LMUL>=1 case, 4 loads would be required whereas vlseg4e.v does what is intended and loads the 1 byte into all 4 registers.

> The reason is that segment load writes several register, so in a typical implementation its rate of execution is a fraction of simple unit-stride load and arithmetic instruction, e.g. vlseg4 [...]

I will accept your general premise. However, like with addi, I expect the special case of adjacent fractional registers when <fl> of  vlseg<fl>e.v matches  LMUL 1/<fl> will be optimized in a typical implementation.

My concern, highlighted by your comment, is that **fractional LMUL may be as disruptive to implement as fixed size load/stores.**

**I would appreciate insights on the actual challenges posed by the fixed length loads so that similar overheads are not introduced by fractional LMUL.**

David-Horner, 2020-03-01:
In the code example above, I corrected the typos and the original LMUL>=1 examples.

However, I did not use the optimal algorithm for the LMUL>=1 case.
I suspect that for most implementation this code will be much more efficient:

```
# simulate vlb.v with LMUL=4
# a0 is memory address of bytes to load. 

vlseti t0,x0,e32    # 01 set vl for target
vlseti t0,t0,e8     # 02 use that vl
vlseg4e.v v12,a0    # 03 load registers v12 ; VRF[12:15] 
vlseti t0,x0,e32,l4 # 04 make v12 a register set ; LMUL=4
vsll v12,24         # 05 left justify byte.
vsra v12,24         # 06 shift back as sign extended
                    #    done !!

```
No VRF registers are temporarily used in addition to the result LMUL=4 register.
And this works as well for LMUL=8 (with v24 for example).

and vlbu.v with LMUL=4 (or LMUL-8) is simpler.

```
# simulate vlb.v with LMUL=4
# a0 is memory address of bytes to load. 

vlseti t0,x0,e32,l4 # 01 set vl for target
vmv.v.i v12, 0      # 02 Splat zero.
                    #    the above could typically be done outside a loop.
vlseti t0,x0,e32    # 03 set vl for target
vlseti t0,t0,e8     # 04 use that vl
vlseg4e.v v12,a0    # 05 load registers v12 ; VRF[12:15] 
vlseti t0,t0,e32,l4 # 06 make v12 a register set ; LMUL=4
                    #    done !!

```
**More work needs to be done to make fractional LMUL emulation of fixed size load/stores be more efficient than the LMUL>=1 approaches.** 
**But, do they need to be efficient if widening instructions will as @solomatnikov proposed actually do meaningful work?**

David-Horner, 2020-03-06:
I am proposing some further changes to what I previously proposed.

**SEW be limited to e32 and less in fractional levels of LMUL.**

This would be true for RV32 and RV64 regardless of the max SEW LMUL=1 supports.

For lf2 (LMUL=1/2), SEW =  e32, e16 and e8 only are supported.
For lf4 (LMUL=1/4), SEW =  e16 and e8 only are supported.
For lf8 (LMUL=1/8), SEW =  e8 only is supported.

> Note: lf16 is reserved.

In fractional LMUL, this ensures that any SEW element is contained within one SLEN chunk (lane).
This reduces the wiring (mux) and/or shifts required to align source and especially to align result to output register.

> Note: For lf8 single width operations, only a single byte result is revised in every consecutive 32 bytes in the VRF register. An in-place register implementation can skip over unchanged SLEN chunks. However, a typical register rename implementation would need to copy 31 bytes of each.

----------------------------
   
 **Revised mapping to allow overlap of adjacent fractional LMUL register sets.**
  
Fractional LMUL register sets are overlayed on two sets of consecutive LMUL=1 registers.

I propose that in one half of the lower LMUL  overlap one quarter of the next higher. Note: 1/8 is lower than 1/4.

This approach allows register assignment to minimize LMUL=1 register usage.
e.g. in lf4 widened register w16 as the sum of v26 and v27 all co-exist in VRF[21]

In addition the interleaved structure of the two different register type will allow for horizontal SEW modifications (which are more useful than LMUL>1 vertical mods.)

As before, there are 2 sets of register allocation to consecutive VRF registers.
Specifically:
 
lf2 sub-register 0 to 15 to registers 4 through 11
lf2 sub-register 16 to 31 to registers 20 through 27
lf4 sub-register 0 to 15 to registers 2 through 5
lf4 sub-register 16 to 31 to registers 18 through 21
lf8 sub-register 0 to 15 to registers 1and 2
lf8 sub-register 16 to 31 to registers 17 and 18

Thus 
lf4 sub-register 8 to 15 overlap lf2 sub-registers 0 through 3
lf4 sub-register 24 to 31 overlap lf2 sub-registers 16 through 19
and
lf8 sub-register 8 to 15 overlap lf4 sub-registers 0 through 3
lf8 sub-register 24 to 31 overlap lf4 sub-registers 16 through 19

> Note: For this and the previous allocation, the constraint of 2* widening and 2* narrowing register overlap was dependent upon the specific mapping of the sub-registers in the lf2 case.
> This allocation also complicates the overlap determination for lf4 and lf8.

David-Horner, 2020-03-18:
Created by widening instructions at LMUL level
LMUL | 1/8  |  1/4  |  1/2 | 1 | 2 | 4 | 8|
-- | -- | -- | -- | -- | -- | -- | --
1:8| | | | | | x | |
1:4| | | | | x | | |
1:2 | | | | x | | | |
1:1  | | | x | | | |
4:4 
4:2 | | x | | | |
4:2 | | x | | | |

commit 9c7732b6 (2020-04-25): Added fractional LMUL, including modifying vector data register and
vector mask register layours for SLEN<VLEN implementations.

Closes #382, #393, #399.

Please reopen any unclosed fractional LMUL issues on top of this concrete proposal.
closed 2020-04-25 by kasanovic