riscvarchive/riscv-v-spec #386: vfredsum should grant more implementation flexibility

aswaterman, 2020-02-25:
The current draft says: "Each operator can either perform an exact addition, or a floating-point addition according to the RISC-V IEEE scalar floating-point specification and with the currently active floating-point dynamic rounding mode."

@jhauser-us points out that "Implementations would have much more flexibility if we allowed each operator to return not just one of those two results (exact or IEEE-rounded) but also any real number in between."

I'm inclined to agree.  This proposal does not reduce the accuracy of the instruction, and it allows the implementation to play more tricks.  I've reasoned that the difficulty of testing these instructions would not be materially increased, because the current definition basically mandates that any testing strategy either compute the error bounds and admit a range of real numbers, or must use inputs for which any partial sum is exact.  In either case, this proposal doesn't exacerbate the testing effort.

jhauser-us, 2020-02-27:
On second thought, I believe we need a different version than I first suggested.  Among the many other possible implementations of VFREDSUM, I believe we should want to allow one that performs an ordinary floating-point summation tree in wider precision, e.g. summing single-precision values in double-precision, before rounding the final result back to single-precision.  Neither the current draft text nor my original suggestion above allows this.  (Although my original suggestion provides a lot of freedom, it doesn't permit a higher-precision rounding to go in the opposite direction, up versus down, than would occur when rounding at the nominal precision, a situation that occurs quite easily when the rounding mode is to-nearest.)

I now propose to replace this sentence in the draft:
> Each operator can either perform an exact addition, or a floating-point addition according to the RISC-V IEEE scalar floating-point specification and with the currently active floating-point dynamic rounding mode.

With this (or equivalent text):
> Each operator first computes an exact sum as a RISC-V scalar floating-point addition with infinite exponent range and precision, then converts (rounds) this exact sum to a floating-point format with range and precision each at least as great as the element floating-point format indicated by SEW.  A different floating-point range and precision may be chosen for the result of each operator.

And replace this sentence:
> The root node in the tree must produce an IEEE result of the appropriate SEW.

with
> The rounded result from the root node in the tree is converted (rounded again) to the standard floating-point format indicated by SEW.

With this, I'm proposing to explicitly allow a double rounding of the final result.  In order for a simple double-precision summation tree to be valid for summing single-precision, the double rounding has to be accepted.  It's useful also for other possible implementations, and adds almost nothing to the worst-case error, especially when summing more than a couple of elements.

With my proposed new text, it's no longer necessary to explicitly allow exact addition for the summation operators, because you get exact addition simply by making the floating-point range and precision of an operator result sufficiently large.

aswaterman, 2020-02-28:
This further relaxation seems beneficial to me, since it allows the reuse of a DP reduction tree on systems that want to optimize primarily for DP.

And once again, I don't think this relaxation makes portable testing of these instructions any harder than it already is.

kasanovic, 2020-05-05:
Can you prepare a PR?

PR #447 (2020-05-05): Provide more flexibility to vfredsum implementations
Allows nodes in the reduction tree to be more precise than required.

Resolves #386.
commit 7f8fde29 (2020-05-05): Provide more flexibility to vfredsum implementations

Allows nodes in the reduction tree to be more precise than required.

Resolves #386.
closed 2020-05-05 by kasanovic