riscvarchive/riscv-v-spec #389: Add vlast.m mask instruction?

gsauthof, 2020-02-29:
The "V" instruction set provides the `vfirst.m` instruction for getting the index of the first mask bit.

But there is no instruction for getting the index of the last mask bit. That means there is no `vlast.m`.

Example use case: `memrchr()`

Similarly, the mask modification instructions (such as `vmsbf.m`) are left biased. Perhaps it also makes sense to add variants that work from the right.

kasanovic, 2020-03-20:
Mask variants counting "from the right" expose VLEN differences.

David-Horner, 2020-03-20:
On 2020-03-20 10:57 a.m., Krste Asanovic wrote:
>
> Mask variants counting "from the right" expose VLEN differences.
>
However:

> The vpopc.m instruction counts the number of mask elements *of the 
> active elements* of the vector source
> mask register

This implies vl applies to vpopc.m.

Similarly, does vfirst.m not use vl to determine the point that the scan 
for a set bit finishes?

In any case, there is no reason that  vlast.m could not use vl to 
establish the  last element, and then it is does not expose VLEN any 
more than any other instruction.

daweili1226, 2020-06-15:
The following loop lives-out "c" variable, and now that we have vfirst.m to find the first lane set, why don't we have vlast.m to find the last lane set for this case? Here "c" expect the last lane because only the newest update is valid.

```
    for (int i = 0; i < num; i++) {
        if (a[i] == b) {
            c = a[i];
        }
    }
```

kasanovic, 2020-07-03:
This could be added, but it does not really work well with length-agnostic stripmining when loops are stripmined in the forward direction.   Consider the example loop, which does not have a "break".  Each strip could have one or more matches with b, and we can find the last match in that strip and set a mask bit, but that would not really help with the overall loop. For the example loop, you would simply stripmine with a VLMAX vector of partial matches recording any match index and corresponding value of c.  At the end of the stripmine loop, you would max-reduce the index vector then read out the c value at the corresponding lane index.  I don't see how using flags would make this go faster, unless you knew the whole vector fit in VLMAX.

The existing "set X first" flag instructions are used for while loops that need to exit, but don't want to wait for scalar code to see and handle exit.  The "find last" loop structure shown has to execute all loop iterations anyway.

daweili1226, 2020-07-10:
I tried to write assembly as below based on the idea you just presented:

```  
    la          a0, num
    ld          s1, a                # a[]
    la          a1, b                # b
    vmv.s.x     v3, c                # initial c value to v3[0]
loop:
    vsetvli     t0, a0, e32          # t0 = vl, a0 = N, 32-bit
    vle32.v     v1, s1               # v1 = v_a
    vmseq.vx    v0, v1, a1           # a[i] = b?
    li          t1, 0                # initialize slidedown value to be zero
    vlast.m     t1, v0, v0.t         # last index of a[i] = b lane, t1 remains zero if v0 is none
    vslidedown  v3, v1, t1           # slide v_a down until v[0] is the last (no change if t1 = 0)
    sll         t2, t0, 2            # Multiply
    add         s1, s1, t0           # Bump Pointer
    sub         a0, a0, t0           # Decrement count by vl
    bnez        a0, loop             # Any more?

    vmv.x.s     a2, v3               # live-out c from v3[0]
```
I could not make this go faster but it looks like we just need to add on the two `li` and `vslidedown` instructions, and this kind of loop has to execute all loop iterations anyway.

kasanovic, 2020-07-19:
The C code you gave:
`for (int i=0; i<num; i++)
{
    if (a[i]==b)
    {
        c = a[i];
    }
}
`
reduces to "if any a[i]==b, c=b", which does not actually
need the index, so I assumed a more complex case where index of last
match is returned:

`for (int i=0; i<num; i++)
{
    if (a[i]==b)
    {
        c = i;
    }
}
`

The assembly code for the method I described looks like this:
```
    # a0 = num
    # s1 = a
    # a1 = b
    # a2 = c    
    
    vsetvli t0, a0, e32, ta,ma  # Agnostic on first iteration
    mv a3, t0    # Remember largest vector length used
    vid v2       # Put element index into v2, assume 32b for indices
    vmclr.m v3   # Clear "Found" flag

  loop:
    vle32.v v1, (s1)
    vmseq.vx v0, v1, a1
    vmerge.vvm v4, v4, v2, v0.t # Remember new found index
    vmor.mm v3, v3, v0          # Remember this element position has found match
    vadd.vx v2, v2, t0          # Bump indices
    sub a0, a0, t0
    slli t2, t0, 2
    add  s1, s1, t2
    vsetvli t0, a0, e32, tu,mu # retain tail and masked-off elements
    bnez a0, loop

    vsetvli x0, a3, e32 # Recover largest vector length used
    vfirst.m t0, v3  # Any matches?
    bltz t0, exit   # If not, skip assignment to c

    vmv.s.x v1, x0               # Initialize scalar for reductions
    vmv.vv v0, v3                # Move "found" mask to v0
    vredmaxu.vs v1, v4, v1, v0.t # Find max index over matches.
    vmv.x.s a2, v1               # Get index in scalar

   exit:
```

Note the loop runs at full vector speed with no cross-lane operations or reductions.  It's only the final reduction step that gets the max final index.

The `vfirst` will be faster than `vpopc` on most implementations when determining "any match?" as it can early out.

The second and later iterations use a "mu,tu" setting because you need to retain the partial reduction values in v4.  Some implementations might benefit from adding "ta" to the other vector instructions (this would be an example needing fine-grained switching of agnostic setting).  The other vector instructions are umasked so "ma/mu" choice is moot.

daweili1226, 2020-07-22:
Hi Krste,

It looks like you are assuming "last" index being "max" index. 
However, if the loop is reversed:
for (int i=num - 1; i >= 0; i--) { if (a[i]==b) {return i; } }
which is quite similar to "memrchr" Gsauthof mentioned, vlast can be same functional as vfirst in forward looping.

kasanovic, 2020-07-22:
The natural way to implement the backwards search is with a negative stride and vfirst.m.
I agree that might be less likely to run well on some implementations than using positive unit-stride memory loads, but some implementations might optimize negative unit-stride, and if the memory accesses in loop were not unit-stride then probably equal throughput regardless of sign of stride.

For the case where the while loop (i.e., data-dependent exit) only has negative unit stride, I could see that vlast.m might be helpful but would also need matching vms*l  functions for general while loop case (and possibly some different form of fault-only-first instructions).  When the loop also has strided or scatter-gather accesses, it will be difficult to map the loop, and it would probably only really work to implement as negative stride code.

I don't feel this is a strong case to provide vlast.m (only useful for while loops with only negative unit-stride memory accesses in loop), but am happy to look at other example uses.

kasanovic, 2020-08-07:
I'm going to close for v1.0 as no real use case presented.

closed 2020-08-07 by kasanovic