riscvarchive/riscv-v-spec #390: Disallow VL=0?

JamesKenneyImperas, 2020-03-02:
The current specification allows the vector length (`vl`) to be zero. I am wondering whether this is ever useful, or whether it would be simpler to just disallow it.

Allowing `vl=0` produces anomalies. The effect is to turn most but not all vector instructions that are not otherwise illegal into NOPs. However, if the instruction targets a scalar register then `vl` is ignored. This causes some odd asymmetries; for example:

```
    vmv.x.s rd, vs2        # extracts element 0 of vs2 when vl=0
    vmv.s.x vd, rs1        # NOP when vl=0
```

And similarly:

```
    vfmv.f.s rd, vs2      # extracts element 0 of vs2 when vl=0, unless Illegal
    vfmv.s.f vd, rs1      # NOP when vl=0, unless Illegal
```

Other instructions that still have some effect when `vl=0` are `vpopc.m` (sets `rd` to all-zeros) and `vfirst.m` (sets `rd` to all-ones).

Similar controls in other architectures (e.g. `ZCR_EL1.LEN` in the ARM SVE extension) avoid this problem by having the length field be active_elements-1. So setting `ZCR_EL1.LEN=0` specifies one active set of elements, `ZCR_EL1.LEN=1` specifies two active sets of elements, and so on. I think it would probably be too disruptive to change things in this way, but perhaps any attempt to set `vl` to 0 could instead cause it to be set to 1?

See also issue #103.

jnk0le, 2020-03-02:
This is mostly useful to avoid extra branch to handle corner case, like memset called with zero length.

JamesKenneyImperas, 2020-03-02:
Ok, that answers the question I think.

closed 2020-03-02 by JamesKenneyImperas