riscvarchive/riscv-v-spec #391: Allow register renaming as an implementation option?

JamesKenneyImperas, 2020-03-02:
The current Vector Specification assumes that implementations will not support register renaming and that data hazards must be avoided by explicit overlap constraints on various kinds of instruction. These constraints are numerous and sometimes not obvious (see the ongoing discussions on issue #380, for example). We also see the classic RISC issue by which simplifying the hardware architecture increases complexity downstream (in the tool chains, and in terms of usability and flexibility).

In future, I would expect that some implementations will be created that _do_ support register renaming and do not suffer from such overlap constraints. Should this not be given as an implementation option? The impact on the specification would be to state that this option is available, and to prefix any affected overlap restriction with text such as _In implementations that do not support register renaming_, or similar.

David-Horner, 2020-03-02:
The source / destination constraints are for vrestart, with mask also affected when LMUL>1 . Micro-architect design that avoid restart can relax such constraint independant of register rename. It is this restart feature that raises these inherant restrictions and not that this is a RISC design.

JamesKenneyImperas, 2020-03-02:
Is it therefore worth allowing a microarchitecture that _avoids restart_ as an implementation option as you describe?

I am trying to clarify whether there are restrictions in the specification simply because assumptions have been made about the implementation that may not apply in a more general case.

hanna-kruppe, 2020-03-02:
Well, what advantages do you see in allowing such variation between implementations? In general implementation-defined behavior has significant downsides, such as risk of software being unintentionally non-portable, more complexity in golden simulators and formal specifications, debugging/reproducing issues sometimes being harder, etc. and the benefits have to outweigh those concerns. In this case, non-portable software could potentially get slightly better register allocation, but it's not obvious to me that this is significant.

JamesKenneyImperas, 2020-03-02:
There are already considerable variations allowed between implementations, so the valid points you raise about the downsides of non-portability apply right now.

I am viewing things from a user standpoint, not a hardware standpoint. From this perspective, regularity and simplicity are desirable. Currently, there are restrictions in the specification that reduce regularity and simplicity because a particular implementation has been assumed. I was trying to understand whether these were considered truly fundamental or not.

closed 2020-03-02 by JamesKenneyImperas

David-Horner, 2020-03-02:
@JamesKenneyImperas 

Restart allows for exceptions and interrupts to be handled "immediately" and without undoing already processed useful work.  It greatly simplifies exception handling if vrestart precisely identifies the element that raised the exception. Forward progress is also enhanced (consider a heavy interrupt environment or situation). 

The POR allows for an imprecise restart. Intriguingly, in section 18.1, "**Precise** vector traps" is defined to accommodate this situation. #364 recommends abandoning the unique "precise" definition and embrace a forward recoverable definition that allows various imprecise states within a set of consecutive instructions. So rather than eliminate restart I propose we expand its usefulness.

hanna-kruppe, 2020-03-02:
> There are already considerable variations allowed between implementations, so the valid points you raise about the downsides of non-portability apply right now.

Yes, there are already many implementation defined aspects in the current spec and these all need to carry their own weight (if some don't, they should be removed), but adding even more exacerbates the downsides further.

> I am viewing things from a user standpoint, not a hardware standpoint. From this perspective, regularity and simplicity are desirable. Currently, there are restrictions in the specification that [...]

As a software/toolchain engineer, I agree it would be great if these irregular constraints could be avoided. However, since we can at best make them an implementation choice rather than removing them entirely, programmers and tools need to be aware of them in any case, even if they could sometimes ignore them in non-portable software. Thus, an implementation option *increases* the complexity and irregularity.