riscvarchive/riscv-v-spec #392: vslidedown behavior when vl<vlmax

JamesKenneyImperas, 2020-03-06:
For vslidedown, section 17.3.2 says:

```
vslidedown behavior for source elements for element i in slide
        0 <= i+OFFSET < VLMAX   Read vs2[i+OFFSET]
        VLMAX <= i+OFFSET       Read as 0
```

When VL<VLMAX, this has the effect of copying **tail** elements from vs2 into vd, which does not seem right, and also conflicts with the defined behavior of tail elements in section 5.4:

```
The tail elements during a vector instruction’s execution are the elements past the current
vector length setting. The tail elements do not raise exceptions, and do not update any
destination vector register group.
```

I would instead have expected the specification to be:

```
vslidedown behavior for source elements for element i in slide
        0 <= i+OFFSET < VL   Read vs2[i+OFFSET]
        VL <= i+OFFSET       Read as 0
```

Can you please confirm that the specified behavior is correct? If it is correct, can you state the best way to perform a slidedown that guarantees that all elements that would be sourced from otherwise inaccessible tail elements are zeroed?

JamesKenneyImperas, 2020-03-06:
I see that this is a duplicate of #261, but that issue was created at a time when tail elements were zeroed, whereas now they are left untouched and likely to be garbage.

aswaterman, 2020-03-06:
The current definition is deliberate (which is not to say your suggestion is off-base). It’s slightly simpler to implement as-is, and it seems to me to work fine in practice. Bear in mind that tail elements aren’t inherently garbage; you can contrive scenarios where shifting down from elements >= vl is actually the desired behavior. And you can often emulate your alternative formulation with low overhead by setting vl to ~0 and zeroing the source vector register before entering the strip-mine loop.

Note also that the current formulation is consistent with vrgather. In that case, vl indicates how many elements are gathered, and it’s quite deliberate that it’s possible to gather elements whose indices lie between vl and VLMAX.

The Section 5.4 rule doesn’t seem contradictory to me, since it doesn’t claim tail elements are never read, just that they aren’t written and don’t raise exceptions. Perhaps the language could be clarified a bit.

JamesKenneyImperas, 2020-03-06:
Ok, understood. I think I would recommend two clarifications to the specification:

1. In section 5.4, specifically state instructions that are exceptions to the tail element rule, and for which the final `and do not update any destination vector register group` statement does not apply; and:
2. In section 17.3.2, perhaps add a code fragment that shows the best way to perform a slidedown that guarantees that all elements that would be sourced from otherwise inaccessible tail elements are zeroed.

I wonder whether a new kind of `vmv.v.v` that zeroes the tail would be a useful thing to have?

colinschmidt, 2020-03-06:
Reading the spec it seems `vslidedown` still only processes `VL` elements so it doesn't touch the destination tail elements and therefore is not an exception to 5.4, which is why Andrew brought up `vrgather` which also gathers `VL` elements.

JamesKenneyImperas, 2020-03-09:
Colin, I think your comment would be correct if it was the case that section 5.4 was intended to apply only to _destination_ elements but this is not stated, and I suspect that it was not the case when it was written. I cannot see that it would do any harm to be more specific here.

I think `vrgather` should be considered a separate case from `vslidedown`. For `vrgather`, the application has complete control over the source elements: if it wants to pick elements from the tail region for some reason, then that is entirely fine. I can't immediately think of a use for this myself, but there is no reason to prevent it. The case for `vslidedown` is different. Here, the application has no choice: except in the degenerate case of a zero slide, the result _will_ be affected by tail elements in the source (if unmasked).

Suppose that a library routine wants to slide down a vector value by a large or variable amount and fill values sourced from beyond the current `vl` with zero (or something else).  When sliding _up_, it looks easy: because destination elements before `OFFSET` are _unchanged_, the application can splat the required default over the entire destination and then use `vslideup`. What is the best way to do something equivalent when sliding down? Perhaps this could be added as an example in 17.3.2.

chuanhua, 2020-03-27:
As Andrew suggested that you can prepare a vector register with all elements being 0 to start with. And when you want to do the slidedown operation, all tail elements are 0 already.

Assuming `v8` will be used later for slidedown operation,

```
vsetvli t0, x0, e32     ; set VL to MAXVL
vxor.vv v8, v8, v8      ; zero all elements
vsetvli t0, a0, e32     ; a0 has the true VL for the algorithm
```

If this is not possible for your algorithm and the tail elements have values that you want to use later, then 

Assuming `v8` is the vector register with non-zero tail elements, and `v9` will have the slidedown value of `v8` with 0 on `v8`’s shifted down tail elements. `a0` has the slidedown `OFFSET`.

```
csrr t0, vl             ; done once outside the loop, t0=VL
vxor.vv v7, v7, v7      ; done once outside the loop
…
vslidedown.vx  v9, v8, a0   ; shift down tail elements
sub t1, t0, a0              ; calculate the start offset for zeroing
vslideup.vx  v9, v7, t1     ; zeroing shifted down tail elements

```

chuanhua, 2020-03-27:
Here is another example using mask registers for `LMUL=1`. Assuming `v8` is the vector register with non-zero tail elements, and `v9` will have the slidedown value of v8 with 0 on v8’s shifted down tail elements. a0 has the slidedown `OFFSET`. 

```
csrr t1, vl            ; t1=VL
setvli t0, x0, e32     ; set VL to MAXVL
vmset.m v1             ; set all MAXVL mask to 1
setvli t0, t1, e32     ; set original VL
vmclr.m v1             ; clear VL mask to 0, all tail element mask remains as 1
…
vslidedown.vx  v9, v8, a0   ; shift down tail elements
vlidedown.vx  v0, v1, a0    ; shift down mask tail elements
vxor.vv v9, v9, v9, v0.t    ; zeroing all shifted down v8 tail elements

```

The last zeroing instruction may be done once outside the loop if no cross-element computations, such as reduction/slide, are performed on the slided-down vector values.

HanKuanChen, 2020-06-08:
I prefer to change VLMAX to vl.

> I think `vrgather` should be considered a separate case from `vslidedown`. For `vrgather`, the application has complete control over the source elements: if it wants to pick elements from the tail [...]

Totally agree.

I don't know whether we can find a scenario that `vslidedown` uses tail part. But based on current [intrinsic](https://github.com/riscv/rvv-intrinsic-doc) design, it only allowes software programmers use [0, vl) elements.
In addition, software programmers cannot specify register usage.
That means for @chuanhua 's [case](https://github.com/riscv/riscv-v-spec/issues/392#issuecomment-604801974)

> As Andrew suggested that you can prepare a vector register with all elements being 0 to start with. And when you want to do the slidedown operation, all tail elements are 0 already.
>
> Assuming `v8` [...]

is not true for current intrinsic. See the following code.
```
vsetvlmax_e32m8();
vint32m8_t input = vzero_i32m8();
vsetvl_e32m8(true_vl);
input = vlw_v_i32m8(data);
vint32m8_t result = vslidedown_vx_i8m1(input, 1);
```
The `result[vl - 1]` is unknown, because compilers do not guarantee `input` use the same vector register as previous `input`.

kasanovic, 2020-07-03:
I believe the remaining issue here is in intrinsic specification rather than in the ISA?

JamesKenneyImperas, 2020-07-03:
I made a couple of documentation change suggestions (March 6, above) which would help clarify things, I think.

kasanovic, 2020-07-19:
I'm adding some clarification text to close the spec issue.

I believe there still might be some intrinsic issue on how to access the functionality, but that is probably best addressed in intrinsics forum, so I'm going to close here with the edit.

commit 393dd59c (2020-07-19): Clarified that prestart, body, tail specifications are for destination
elements. Added note that some instructions access source elements
from outside vl range.
Closes #392.
closed 2020-07-19 by kasanovic