riscvarchive/riscv-v-spec #394: vmerge for more than two source vectors?

NilsGraf, 2020-03-14:
I like the `vmerge` instruction, but it only has two source vectors. May I suggest a more generic `vmerge` that can operate on up to 31 vector registers simultaneously? Element `[i]` of the mask register `v0` would specify the source vector element (`v1[i] .. v31[i]`) for `vd[i]` as follows:
`vmerge.vvm vd, v0   # vd[i] = v[ v0[i] ][i]`

For reference, `vmerge` is currently defined as follows: 
`vmerge.vvm vd, vs2, vs1, v0    # vd[i] = v0[i].LSB ? vs1[i] : vs2[i]`

**Example with 4 elements:**
Mask register `v0 = {13, 12, 11, 10}`.  The resulting vector `vd` will contain the first element of `v10`, the second element of `v11`, the 3rd element of `v12`, and the 4th element of `v13` as follows:
`vd = { v13[3], v12[2], v11[1], v10[0] }`. 

Such a merge instruction could be useful for small matrix transpositions and permutations (e.g. for ML applications). 

I believe the hardware cost is fairly low as this could be implemented by modifying the existing 32-to-1 vector read-MUX so that each element has its own 5-bit select signal driven by the mask register `v0`.

solomatnikov, 2020-03-20:
This is extremely expensive from HW implementation point of view.

It would require 2nd pipeline stage that reads vector register file and this read stage must have register file read port with byte granularity - very expensive.

You have to show that this feature improve performance significantly across wide variety of vector workloads to be considered seriously.

kasanovic, 2020-04-03:
Too expensive, and limited advantage.

closed 2020-04-03 by kasanovic