riscvarchive/riscv-v-spec #396: illegal instruction exception on illegal frm value for FP instructions that do not use rounding mode

chuanhua, 2020-03-19:
Section 11.1 says that 

> all vector floating-point operations use the dynamic rounding mode in the frm register.

However, there are vector FP instructions that do not sue the rounding mode in the frm register.
Min/max:     vfmin, vfmax
Sign-inject: vfsgnj, vfsgnjn, vfsgnjx
Compare:     vmfeq, vmfne, vmflt, vmfle, vmfgt, vmfge
Classify:    vfclass
Merge:       vfmerge.vfm
Move:        vfmv.v.f
Reduction:   vfredmin, vfredmax
Permute:     vfmv.s.f, vfmv.f.s

If frm contains an illegal value, will these instructions generate "illegal instruction exception"?

aswaterman, 2020-03-19:
Widening conversions and widening multiplies are also independent of the rounding mode.

It’s simpler for the HW implementation if all FP instructions check the rounding mode, even if their result is unaffected by rounding. It also doesn’t seem to matter much for software, since the execution environment really should initialize frm anyway. So I’d advocate for the simple rule that all vector FP instructions check frm’s validity.

This is somewhat consistent with the scalar FP extension, which provides the static rounding mode opcode field even for instructions that don’t round.

chuanhua, 2020-03-19:
This makes sense, especially for simpler HW implementation. Could you clarify this in the spec?

JamesKenneyImperas, 2020-03-19:
Andrew, just to clarify - are you saying that _scalar_ FP instructions such as `fsgnj.s` should also cause Illegal Instruction exceptions if `frm` is invalid, even though they are really just bit manipulation instructions? (I didn't think that this was the case). Or are we saying that _vector_ instruction behavior will differ from _scalar_ instruction behavior in this respect?

My instinct would be to make equivalent scalar and vector instructions behave in the same way unless there is a very good reason not to. I was under the impression that in the scalar extension an Illegal Instruction was generated only if the rounding mode was actually used.

chuanhua, 2020-03-19:
fsgnj.s does not have rm field in its encoding. Most of FP instructions that does not use rm field has used fixed encoding in the rm field. "fcvt.d.s" instruction is the exception which has the rm field, but does not use the rounding mode.

JamesKenneyImperas, 2020-03-19:
Yes, I know that `fsgnj.s` does not have the `rm` field in its encoding, because rounding mode is not used by this instruction. But the rounding mode is not used by `vfsgnj` either, so it seems inconsistent to say that one instruction causes an Illegal Instruction exception when the other does not.

chuanhua, 2020-03-20:
Scalar FP computation instructions has static rounding mode encodings. So SW may not use and initialize the dynamic rounding mode field in fcsr.frm which can contain an invalid value. Vector FP computation instructions always use the dynamic rounding mode field in fcsr.frm. So SW has to initialize frm to do most of vector FP computation instructions. This expected SW behavior can be used architecturally to simplify the vector FP instruction decoding logic for illegal instruction exception generation. And the generation of illegal instruction exception is just used to help fixing SW bugs. Is it necessary to make this behavior consistent across scalar and vector FP instructions?

aswaterman, 2020-03-20:
I'm thinking along the same lines as @chuanhua.

JamesKenneyImperas, 2020-03-20:
If the definition of `fcsr.frm` was changed so that it was WARL then this would be a non-issue, and vector and scalar operations would be consistent (and simpler). Are the illegal values intended to enable virtualization of user-defined rounding modes? I'm not aware of the history.

aswaterman, 2020-03-20:
My recollection of the history is that we chose not to make it WARL so that new rounding-mode support could be added over time via trap-and-emulate.  (Aeons ago, this facilitated the addition of the round-to-nearest/ties-to-max-magnitude mode when the UCB implementations didn't support that mode in HW.)  I think we made the right decision, but it's admittedly not a slam-dunk.

JamesKenneyImperas, 2020-03-20:
If vector and scalar operations are not required to be consistent, then another alternative could be to have vector instructions treat illegal `frm` values are RNE (or any other mode of your choice). I'm not saying this is a good idea, just raising it in case it makes implementation simpler.

aswaterman, 2020-03-20:
It is another valid option, but trapping unsupported rounding modes favors extensibility.  I don't think adding new rounding modes is at all hypothetical, given recent trends elsewhere in the industry.

Anyway, I don't think muxing in a different rounding mode instead of trapping saves anything, given the need to trap on `FS` != 0.  (I am going out on a limb and assuming we will eventually agree that we will trap on `FS` for the same set of instructions for which we'll trap on `frm`, which is what my implementation currently does.)

JamesKenneyImperas, 2020-03-20:
Apologies for bleating on about this, but one more point: if the intention is that illegal `frm` with all vector floating point instructions causes an Illegal Instruction, then presumably this check will be applied even if `VL=0` (like the `FS` check)? If you are clarifing the specification then I think the `VL=0` behavior should be explicitly stated too.

aswaterman, 2020-03-20:
@JamesKenneyImperas if it can be described as bleating, it's certainly useful bleating :)

All these checks should be independent of the setting of `vl`, akin to how performing a widening operation when LMUL=8 should raise an exception even when `vl`=0.  Let's try to nail all this down before closing this issue.

chuanhua, 2020-03-27:
This vector instruction, `vfncvt.rod.f.f.w`, uses the rounding mode of "rounding towards odd", not the dynamic rounding mode from `frm`. However, from the above discussion, like the vector instructions that do not use any rounding mode, I assume that this instruction will also generate an "Illegal instruction exception" on an illegal `frm` value.

aswaterman, 2020-03-27:
It would be consistent with my recommendation for vfncvt.rod.f.f.w to raise
an exception if rm is invalid.

kasanovic, 2020-04-03:
I'm in agreement with above.

commit 7c63e970 (2020-07-03): Clarified vector FP illegal instruction exceptions for invalid rounding mode.
Closes #396.
closed 2020-07-03 by kasanovic