riscvarchive/riscv-v-spec #398: Pseudo instruction for integer truncation / narrowing convert

kito-cheng, 2020-03-24:
In vector ISA, we have `vwcvtu.x.x.v` and `vwcvt.x.x.v` for widening integer convert, which is pseudo instruction of `vwadd.vx` and `vwaddu.vx`, should we add similar one for narrowing convert, `vncvt.x.x.v`, it can be done by `vnsrl.wi`.

```
 vncvt.x.x.v vd, vs2, vm
 ->
 vnsrl.wi vd, vs2, 0, vm
```

Ref:
### 14.17. Narrowing Floating-Point/Integer Type-Convert Instructions

> An integer value can be halved in width using the narrowing integer shift instructions with a shift amount of 0.

kito-cheng, 2020-05-19:
ping

aswaterman, 2020-05-19:
I support having a canonical pseudoinstruction for integer truncation.

In case some implementations choose to optimize these instructions, it would be good to choose an encoding more similar to `vwcvt[u]`.

`vnsrl.wx` has an encoding more similar to `vwadd.vx`, so perhaps we should use `vnsrl.wx` instead of `vnsrl.wi`, in case implementations choose to optimize.

kasanovic, 2020-07-03:
PR request  https://github.com/kito-cheng/riscv-v-spec/commit/baf2ceadc5b5459fe7e54a9f3432e710521ce210 merged to fix this

closed 2020-07-03 by kasanovic