riscvarchive/riscv-v-spec #400: mstatus.VS field location in version 0.8

JamesKenneyImperas, 2020-03-25:
I'm confused about the sequence of events regarding the changed location of the `mstatus.VS` field. Could you please confirm whether this field is at position 10:9 or 24:23 in release 0.8?

Reading backwards, the changes seem to be these:

**February 9th 2020**, commit bdb8b55 (Merge pull request #351 from jhauser-us/vs-field-10-9): seems to modify field location from 24:23 to 10:9. This postdates version 0.8.

**December 14th 2019**, commit 9a65519 (This version is official v0.8 release of vector spec): this defines version 0.8.

**December 8th 2019**, commit d8fbe85 (Move mstatus/sstatus VS field to bits 10:9): also seems to modify field location from 24:23 to 10:9. This predates version 0.8.

I'm no GitHub expert, but how can the same change seem to happen both before and after version 0.8, and which change wins?

hanna-kruppe, 2020-03-25:
On December 8th, John Hauser authored the commit d8fbe859002979aedf9d34d5be0a57262c45f20f and opened #351 to propose this change for inclusion in the spec. At this point the commit was not yet incorporated into this repository: if you had downloaded the spec or visited the URL https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc, you would have seen `mstatus.VS` still in position 24:23. The pull request sat untouched for a long time, and during this time release 0.8 was made.

Then, on February 9th, Krste accepted/merged #351 and thereby incorporated these changes into the official version of the specification. Although this was just an administrative act, it is a separate commit in git (bdb8b5574ead5e8036d5f79902d4911a24d8c862), and GitHub displays what effectively changed as a consequence. In this case, no merge conflicts had to be resolved, so the change authored by John Hauser applied unmodified. Indeed, if you ask git or GitHub who authored the line "A vector context status field, `VS`, is added to `mstatus[10:9]` and shadowed" in the current version of the spec, it points to John Hauser's original commit.

In conclusion: this change was first proposed before release 0.8, but only incorporated into the specification some time after the release. In the official release 0.8, the VS field is in position 24:23.

JamesKenneyImperas, 2020-03-25:
Thanks Hanna.

closed 2020-03-25 by JamesKenneyImperas