riscvarchive/riscv-v-spec #404: relationship with ELEN and FLEN, XLEN.

dangerblue, 2020-04-01:
In the spec，it seems the ELEN is a individual parameter that is not limited by the FLEN and XLEN，but in the description of  SEW,  it says " In the base vector extension, only SEW up to max(XLEN,FLEN) are required to be supported."  what if the SEW is larger than max(XLEN,FLEN), what's the correct behavior ？

knightsifive, 2020-04-01:
For vector instructions that involve X- or F-registers, the spec gives details. For example, [Section 11.1](https://riscv.github.io/documents/riscv-v-spec/#_vector_arithmetic_instruction_encoding) says

> For integer operations, the scalar can be taken from the scalar x register specified by rs1. If XLEN>SEW, the least-significant bits of the x register are used. If XLEN<SEW, the value from the x register is sign-extended to SEW bits.
> 
> For floating-point operations, the scalar can be taken from a scalar f register. If FLEN>SEW, the value in the f registers is checked for a valid NaN-boxed value, in which case the least-significant bits of the `f`register are used, else the canonical NaN value is used. If FLEN<SEW, the value is NaN-boxed (one-extended) to SEW.

dangerblue, 2020-04-01:
ok,thanks, another question, the ELEN is not required to be the same with max(XLEN,FLEN), right? in another word, the implementation can support ELEN less than XLEN or FLEN

knightsifive, 2020-04-01:
Yes. As far as I'm aware, the spec only requires that ELEN >= SEW >= 8.

aswaterman, 2020-04-01:
I think we have mistakenly introduced a platform mandate into the ISA spec.  General-purpose platforms are surely going to require ELEN >= max(XLEN, FLEN), but I can also see the use case for embedded cores with XLEN=64 and FLEN=ELEN=32.

knightsifive, 2020-04-01:
Without this constraint, there is only ELEN >= SEW >= 8. (Allowing SEW < 8 would cause problems elsewhere.) However, I don't see the argument for mandating a stricter lower bound than ELEN >= 8.

aswaterman, 2020-04-01:
It might mess up scatter/gather depending on resolution to #401, but otherwise I agree that the ISA itself doesn't need to impose an ELEN constraint beyond ELEN >= 8, and it's up to platforms to do the sensible thing and put a stronger constraint on ELEN.

(Similarly, we have spec'd the scalar Zfh extension in a way that implies FLEN is always >= 32, but there might be no good reason to prohibit FLEN=16 implementations.)

knightsifive, 2020-04-01:
Yeah, "Zfh" is currently spec'd as if it were "Zhinf". I don't see a good reason to prohibit FLEN = 16 implementations. Some minor changes to Section 12.1 would be required.

JamesKenneyImperas, 2020-04-01:
I'll just mention here (I've come from #401) that enquiries about `XLEN=64` and `ELEN=32` is one of the most common we see and that we know of implementations under development using this combination.

dennisyyz, 2020-04-01:
> I'll just mention here (I've come from #401) that enquiries about `XLEN=64` and `ELEN=32` is one of the most common we see and that we know of implementations under development using this combination.

That's true. For low-cost embedded processors, the cost of parallel 64 bits (double FP or int64) arithmetic units cannot be ignored. Although these processors may support scalar 64-bit operations, the demand for vector 64-bit operations is not strong in embedded applications.
I think it should be clear whether ELEN < XLEN is permitted by the Spec.

David-Horner, 2020-04-01:
There is a related question: **If we allow 2 * ELEN for 2 * SEW in widening / narrowing instructions.**
If allowed, is it mandatory? Is it otherwise platform profile dependent?

2 * ELEN widths are transparently supported by unit-stride load/store at all LMUL>=1 levels as element pairs, as long as SLEN >= 2 * ELEN.

Allowing 2 * ELEN is low cost for integer widening operations add/subtract, somewhat more so for multiply/multiply-add. The tie-in to #401 is that for XLEN = 2 * ELEN, these 2 * ELEN widening operations allow building complex XLEN offsets based on successive multiplication and additions of matrix indices.

 2 * ELEN for widening float is substantially more expensive and less likely to be an implementation  design point.

For quad widening only allowing 2 * ELEN seems a reasonable limitation. The cost appears prohibitive to support a very few but expensive quad widening at 4 * ELEN into a LMUL level at which it cannot use narrowing ops.  (assuming quad narrowing is not implemented)  

It is worth noting that stipulating restrictions to these cases still leaves open  relaxing them later. So far I have not seen any relaxation suggestions that require modifications to the existing restricted behaviour.

**This discussion as lead into ELEN support for various data types:**

Perhaps it is warranted to have distinct ELEN for classes of datatypes. We anticipate crypto data types and complex which may benefit from seperate ELEN naming.

@aswaterman   -  no good reason to prohibit **FLEN=16** implementations.

Meaning, of course, FLEN=16 **only, vector** implementations. 

Have we reached a point of crossover in nomenclature that vector parameters should be prefixed, with say V. VFLEN, VELEN etc.  Future extensions such as Packed-SIMD are likely to have similar attributes to name. Such prefix allows for each extension to advance without confusion during cross domain discussions. In this case float extensions and vector.

kasanovic, 2020-04-03:
The intent was to reserve the single "V" letter for application processors (e.g. Linux) and that will indeed require ELEN >=max(XLEN,FLEN).  Embedded systems will want to support many different subsets, e.g., without floating-point. These subsets/supersets do need Zv* names defined.

@David-Horner ELEN is the maximum supported width, so a widening operation is illegal if result would be 2*ELEN.

David-Horner, 2020-04-03:
@kasanovic 
Defining ELEN as maximum SEW length resolves the semantic problem.
It doesn't resolve the concern of an anomalous element length.
However, @aswaterman argues exactly for this in #401.
This revised ELEN definition accommodates both XLEN and 2 * ELEN in respective use cases.

I appreciate the need to limit scope and functionality creep and certain things should not be in the v1 release candidate. This suggestion could be easily added to v1.1. The risk of subsequent fragmentation and ratification delay are low with this case. But better such be considered on merit and excluded for a substantive reason, even if that is avoiding bloat.

Hsiangkai, 2020-05-06:
Back to the original problem, I am also a little bit confused about “In the base vector extension, only SEW up to max(XLEN,FLEN) are required to be supported.” Is it possible we only support SEW up to max(XLEN, FLEN) in some platforms? If we only support SEW <= max(XLEN, FLEN), does it mean ELEN also be bounded by max(XLEN, FLEN) in such platforms?

kasanovic, 2020-06-28:
Spec needs clarification.

kasanovic, 2020-07-03:
I believe this has been clarified. Please open up separate specific issues if there are parts of the spec that are still confusing.

closed 2020-07-03 by kasanovic