riscvarchive/riscv-v-spec #405: How to deal with the pseudo instruction vmsltu.vi and vmsgeu.vi if the immediate is zero in assembler??

Nelson1225, 2020-04-01:
Hi Team,

The original discussion is in the following link,
https://github.com/riscv/riscv-binutils-gdb/issues/206

The current gnu assembler handles the pseudo `vmsltu.vi v25, v14, 0` and `vmsgeu.vi v25, v14, 0` to `vmsltu.vi v25, v14, -1` and `vmsltu.vi v25, v14, -1`, which are caused strange result.  As @aswaterman mentioned, we have two good solutions for now,

* Regard them as error pseudo instructions when imm=0.
* Convert the pseudo instruction to vmsne.vv/vmsgt.vv v25, v14, v14 when imm=0, which has the correct effect of setting all the elements to 0 (and also it maintains any illegal-instruction trapping behavior because of the register numbers).

So which solution do you prefer?

Thanks
Nelson

knightsifive, 2020-04-06:
I don't have a clear preference. Here are three comments.

1. In a perfect world, I'd like the vector instructions to extend the semantics of their scalar analogues. But my wish seems misguided in the case of integer comparisons because:
   - the immediates have different widths (12 vs. 5 bits), 
   - the immediates are sign-extended to different widths (XLEN vs. SEW bits), and 
   - many of the vector integer comparisons don't have scalar analogues anyway.

2. In your second option, there are three choices for `vmsltu.vi` with zero immediate: `vmsne.vv`, `vmsgt.vv`, and `vmsgtu.vv` (input arguments equal). And there are three choices for  `vmsgeu.vi` with zero operand: `vmseq.vv`, `vmsle.vv`, and `vmsleu.vv` (input arguments equal). There are other ways of zero- and unit-initializing a vector: these are just the choices from the cornucopia of comparisons, albeit perhaps the only choices in V-extension which preserve the illegal-instruction behavior you mentioned.

3. A third solution is to leave the behavior as-is, i.e., I don't think this "unexpected" behavior is necessarily a bug.

aswaterman, 2020-04-06:
Since no strong opinion has been expressed in a week, I'll put my finger on the scale so we can make progress on this:

These instructions are useless in optimized code, but they might show up occasionally in sloppy macro-ized assembly (test codes and the like).  So I propose we do the thing that will be most useful in that scenario, which is to convert them to an instruction that does what the programmer intends (vmsne.vv/vmseq.vv).

Nelson1225, 2020-04-07:
@aswaterman @knightsifive Thank you very much, I know how to deal with this now :)

closed 2020-04-07 by Nelson1225