riscvarchive/riscv-v-spec #416: vslidedown/vslide1down destination register and mask register overlap constraint

andes-arch, 2020-04-08:
At RVV spec 5.3:

> The destination vector register group **for a masked vector instruction** can only overlap the source mask register (v0) when LMUL=1. Otherwise, an illegal instruction exception is raised.

At RVV spec 17.3.2 (vslidedown):

> The destination vector register group cannot overlap the mask register if LMUL>1, otherwise an illegal instruction exception is raised.

At RVV spec 17.3.4 (vslide1down):

> The destination vector register group cannot overlap the source mask register if LMUL>1, otherwise an illegal instruction exception is raised.

For vslidedown and vslide1down, the "**for a masked vector instruction**" is missing. Does this mean that the 17.3.2 and 17.3.4 overlap rules also apply to a non-masked instruction? Or they only apply to a masked instruction? If they only apply to a masked instruction, can we add back the missing "for a masked vector instruction" phrase or just refer these rules back to the general constraint in 5.3?

JamesKenneyImperas, 2020-04-08:
I think perhaps that the offending text in 17.3.2 and 17.3.4 should just be deleted (assuming that `v0` is indeed a legal destination when unmasked). It is adding nothing but confusion.

(This is a duplicate of #415, so I will close that issue.)

aswaterman, 2020-04-09:
Agreed with @JamesKenneyImperas - the text in Sec. 5.3 is sufficient, so this is redundant anyway.

commit 43a69a4e (2020-04-09): Remove redundant mask register constraint for slide[1]down

Closes #416
closed 2020-04-09 by aswaterman