riscvarchive/riscv-v-spec #417: proposal for constant time execution as opt-in

jnk0le, 2020-04-14:
Because unlike SIMD, we are targeting temporal execution on a wide range of microarchitectures, we are about to face microarchitecture optimizations or their side effects resulting in data dependent behaviour exploitable via eg. tcp sockets.

Lets consider most obvious skips that could happen due to 1R1W SRAMs, or just a very sophisticated designs.

When eg. masked vadd.vv is executed:
- mask beat is fetched first
- if all elements are ones then we don't need to fetch destination register
- there can be also a tail to preserve
- if all are zeros then we don't need to fetch vs1/vs2, issue it to ALU and writeback (InO, OoO might want to copy it anyway or sub-rename beat)
- if mixed then it executes as regular 4R1W instruction

vfirst, vmsbf.m, vmsif.m, vmsof.m
- they can finish reading from source operand in first cycle (good luck with constant time comparisons)

vcompress
- it can write to variable number of beats, upper part of a last destination beat may need to be also read-preserved without agnostic tail

vredmaxu.vs, vredmax.vs, vredminu.vs, vredmin.vs, vredand.vs, vredor.vs
- they can finish after approaching: UINT{SEW}_MAX, INT{SEW}_MAX, UINT{SEW}_MIN, INT{SEW}_MIN, 0, 2^SEW - 1

Early out result from multipliers/dividers etc.

Renaming multiple sub-registers into one and executing them as a single beat (both operands, if only one then we save only on read ports)

Scatter/gather ommited due to reasons given in following part.

(of course I'm not HDL guy so someone might know better how things are actually done)

Preventing optimizations and artificially slowing down everything may be bad but letting manufacturers to get benchmarks few % above competition at the cost of silicon CVEs is even worser.
Therefore, I propose to add `vconst` field to vtype csr (candidate for stabler field #410).
When set, it will prevent any data dependent leakage (due to difference in masking and contention of read, write, execution ports etc.) from a data being put into vector registers after setting this bit, that will be visible in the same or other SMT domains.
(should fit 0.9 calling conventions and doesn't force mass copying on config change in casse of OoO)

The exception from this rule are things that are very expensive or nearly impossible to mitigate, like:
- caching/actual memory traffic
- scatter/gather/vrgather indices (eg. subbytes in AES, as this leads to classic cache timming attacks. ShiftRows and MixColumns can still use vrgather as the permutation pattern here is not secret)
- leaks from other SMT domains into current one

Simplest implementations will be constant time by their simplicity, so they can simply ignore this bit.

Optionally clearing `vconst` bit can trigger a zeroing of whole register file (plus some other ooo related stuff). The "power gating vdisable" can be reduced to simple hint like imm[0] bit in return jalr or resulting address. (You are free to not like this optional)

other aspects:

- It will allow designers to spend at least some time on security subject
- On rushed designs a "more than necessary" slowdown would be acceptable when limited to only crypto code
- It could serve as a hint for transparent constant power designs which might have 64bits of expensive constant power multipliers + 192bits of regular multipliers that could be used outside crypto code

### Why should this be in base rather then specialized crypto extension?
- When specialized extension is not available, we need to fallback to base where we will return to starting point of this issue. The cost on simplest implementations is just one extra bit to ignore.

kasanovic, 2020-04-19:
Since scalar instructions are needed to run vector code and can have data-dependent timing, wouldn't this proposal be subsumed under a single switch for whole hart?

jnk0le, 2020-04-21:
I aim for vector-only as there is way more possible performance trickery, that can be invisible in simplified unit tests.
There could be another CSR to control scalar multi cycle execution (anything else?), but I don't think that the regular supporting code would be a problem.

The other question is, if we want any "obvious variable time" vector execution at all.
At least it have to be made clear what can and what cannot be safely done. Otherwise there will be a lot of CVEs.

One more item for excepion list is `vl` dependent behaviour, as the executed data length will be exposed by `AVL`/`VLMAX` anyway and short vectors would suffer.

kasanovic, 2020-06-28:
This should be discussed as part of broader "constant-time execution" support profile discussion.  Can bring up with tech committee.

jnk0le, 2021-01-12:
Crypto group decided to use rs1/rs2 ordering hint in mul/clmul, so universal constant time csr is less likely.

CAFxX, 2021-01-12:
> use rs1/rs2 ordering hint 

Aren't implementations supposed to be allowed to ignore hints?

knightsifive, 2021-01-12:
> Aren't implementations supposed to be allowed to ignore hints?

I imagine a profile could mandate otherwise. (My hazy understanding is that a "profile" means a subfamily of (conforming) implementations with certain architectural parameters and implementation-defined behaviors in common.)

jnk0le, 2021-01-12:
> Aren't implementations supposed to be allowed to ignore hints?

The scalar crypto spec (section 3.6) explicitly calls it "hint"

aswaterman, 2021-01-12:
The spec should be changed to avoid the word "hint" for mandatory behavior.

ben-marshall, 2021-03-06:
> The spec should be changed to avoid the word "hint" for mandatory behavior.
@aswaterman - I think we were using "hint" less rigorously than is usual in RISC-V terminology. I'll change the language appropriately. Indeed, in the sense used in this thread, it is absolutely not an ignorable "hint".

jnk0le, 2021-03-09:
One more addition is to fine grain what is constant'd as not everything needs secret masks etc.

jnk0le, 2023-02-23:
zvkt

closed 2023-02-23 by jnk0le